srsran/srsRAN_Project #217: Getting Segmentation fault

KeyIrf, 2023-09-20:
## Issue Description ##
Getting a segmentation fault when running the gNB application in ORAN mode. Below is the stack trace reported
==== gNodeB started ===
Type <t> to view trace
Stack trace (most recent call last) in thread 68935:
#10   Object "", at 0xffffffffffffffff, in
#9    Source "../sysdeps/unix/sysv/linux/x86_64/clone.S", line 95, in __clone [0x7f7b0f721132]
#8    Source "/build/glibc-SzIz7B/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0x7f7b0fb48608]
#7    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7f7b0fa34df3, in
#6    Object "/usr/local/bin/gnb", at 0x559512d872eb, in std::thread::_State_impl<std::thread::_Invoker<std::tuple<srsran::unique_thread::make_thread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, srsran::unique_function<void (), 32ul, false>, srsran::os_thread_realtime_priority, srsran::os_sched_affinity_bitmask const&)::{lambda()#1}> > >::_M_run()
#5    Object "/usr/local/bin/gnb", at 0x559512d8cb09, in srsran::task_details::smallbuffer_table_t<srsran::general_task_worker<(srsran::concurrent_queue_policy)0, (srsran::concurrent_queue_wait_policy)1>::make_blocking_pop_task()::{lambda()#1}, void>::call(void*) const
#4    Object "/usr/local/bin/gnb", at 0x55951365e50f, in srsran::task_details::smallbuffer_table_t<srsran::ofh::realtime_timing_worker::start()::{lambda()#1}, void>::call(void*) const
#3    Object "/usr/local/bin/gnb", at 0x55951365e3c5, in srsran::ofh::realtime_timing_worker::timing_loop()
#2    Object "/usr/local/bin/gnb", at 0x55951365dfd1, in srsran::ofh::realtime_timing_worker::poll()
#1    Object "/usr/local/bin/gnb", at 0x55951365d646, in srsran::ofh::ota_symbol_dispatcher::on_new_symbol(srsran::ofh::slot_symbol_point)
#0    Object "/usr/local/bin/gnb", at 0x559513665df7, in srsran::ofh::transmitter_ota_symbol_task_dispatcher::handle_new_ota_symbol(srsran::ofh::slot_symbol_point)
Segmentation fault (Address not mapped to object [0x1])

## Setup Details ##
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04

gNB is operated in ORAN (split 7.2x) mode. gNB is in SA mode

Tried with the latest installation package and also built from source 

## Expected Behavior ##
gNB comes up establishes core connectivity and start sending Ethernet packets on FH interface

## Actual Behaviour ##
gNB establishes core connectivity, but just in few seconds it reports seg fault (as part of transmitter_ota_symbol_task_dispatcher)

## Steps to reproduce the problem ##
gNB installed and run the gNB with attached config

sudo gnb -c gnb_ru_ran550_tdd_n78_20mhz_mod.yml

## Additional Information ##

[gnb_ru_ran550_tdd_n78_20mhz_mod.zip](https://github.com/srsran/srsRAN_Project/files/12672961/gnb_ru_ran550_tdd_n78_20mhz_mod.zip)

andrepuschmann, 2023-09-21:
Hey - can you please make sure you use the latest code drop, use commit 5e6f50a202c6efa671d5b231d7c911dc6c3d86ed.
If you still see the issue, could you perhaps share logs (with OFH log to debug `log --ofh_level=info/debug`) output. Or even a PCAP of the link to the RU. Also what RU model do you use?

KeyIrf, 2023-09-21:
hi, 
As a initial step i have not connected to a O-RU. I am just trying to bring up the srsRAN gNB on Intel Xeon with 10 Gbps NIC. I dont think there is any FH packets that is generated yet. The application crashes at the begining - i have confirmed with tcpdump.
I tried the log --ofh_level but there were no logs. Attaching logs with --all_level=debug

[gnb_debug.log](https://github.com/srsran/srsRAN_Project/files/12684119/gnb_debug.log)

KeyIrf, 2023-09-21:
Also i moved to the recommended Ubuntu 22.04 and still see the issue

Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy

AlaiaL, 2023-09-22:
Hey, as you reported there was a bug with that specific configuration. It is already fixed in our internal `dev` branch, but it will take a while until that reaches the public repository.

If you want to use the OFH library, I suggest you to remove this workaround from your gNB configuration, as it is only valid for a Benetel RU in SISO mode (just removing the line or setting it to false will work):
`  is_dl_broadcast_enabled: true                                   # Set to true for a workaround over a firmware bug in the RAN550 when operating in SISO mode.
` 

Also, I'd suggest to remove this line from your configuration:
`   ru_bandwidth_MHz: 100                                           # RU instantaneous bandwidth.` 

Removing this line will make the RU and DU use the same bandwidth and makes it easier to configure.

KeyIrf, 2023-09-22:
Thank you this worked. At least dont get seg fault any more

closed 2023-09-22 by KeyIrf