srsran/srsRAN_Project #221: Unable to run srsRAN gNB with srsUE

rohitdwivedula, 2023-09-22:
## Issue Description ##
Attempting to run this codebase with srsUE and ZeroMQ following the instructions [on this page](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup)

## Setup Details ##
Operation System: Ubuntu 22.04.3 LTS

## Expected Behavior ##
In the user equipment `stdout`, I want to see the IP that is being assigned to the UE. For example, on the tutorial page linked above, this is what they gave as an example of these logs:

```
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=2094
Random Access Complete.     c-rnti=0x4602, ta=0
RRC Connected
PDU Session Establishment successful. IP: 10.45.10.2
RRC NR reconfiguration successful.
```

## Actual Behaviour ##
The user equipment script prints nothing to `stdout` after the RRC connected line:
```
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file zmq.conf...

Built in Release mode using commit fa56836b1 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=11.52e6
Supported RF device list: zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://127.0.0.1:2000
CH0 tx_port=tcp://127.0.0.1:2001
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=974
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
```
Looking at the UE logs at `/tmp/ue.log`, I see the following error line: 

```
2023-09-22T21:05:35.006981 [NAS5G  ] [I] DL PDU (length 11)
    0000: 7e 02 a6 39 cf fd 03 7e 00 64 5b
2023-09-22T21:05:35.006987 [NAS5G  ] [I] Integrity check ok. Local: count=3, Received: count=3  [a6 39 cf fd]
2023-09-22T21:05:35.006988 [NAS5G  ] [I] Decrypted DL PDU (length 11)
    0000: 7e 02 a6 39 cf fd 03 7e 00 64 5b
2023-09-22T21:05:35.006995 [NAS5G  ] [E] Not handling NAS message type: Status 5GMM (0x64)
```

It looks like some PDU message received by the UE is not being handled, causing it to not be assigned an IP. This error is coming from line#234 in [`srsue/src/stack/upper/nas_5g.cc`](https://github.com/srsran/srsRAN_4G/blob/master/srsue/src/stack/upper/nas_5g.cc) file in the `srsRAN_4G` project. 

## Steps to reproduce the problem ##
Find attached my config files gNB and srsUE. 
[gnb.txt](https://github.com/srsran/srsRAN_Project/files/12704792/gnb.txt)
[ue.txt](https://github.com/srsran/srsRAN_Project/files/12704793/ue.txt)

Commands that I run: 
```bash
# start gNB
sudo rm -rfv /tmp/*.log && sudo ip netns delete myue && sudo ip netns add myue && sudo ip netns list && cd ~/srsRAN_Project/build/apps/gnb && sudo ./gnb -c ./gnb.yaml

# start srsUE
cd ~/srsRAN_4G/build/srsue/src && sudo ./srsue ue.conf
```

## Additional Information ##
- I am able to run the [ZeroMQ based setup of 4G](https://docs.srsran.com/projects/4g/en/latest/app_notes/source/zeromq/source/index.html) perfectly - am able to ping/iperf3 and see changes in the UL/DL logs of the eNB. 
- commit 5e6f50a202c6efa671d5b231d7c911dc6c3d86ed on srsRAN_Project
- commit fa56836b14dc6ad7ce0c3484a1944ebe2cdbe63b on srsRAN_4G
- Installed Open5Gs via package manager on Ubuntu: 

```
open5gs-amf/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-ausf/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-bsf/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-common/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-dbg/jammy 2.6.4~jammy amd64
open5gs-hss/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-mme/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-nrf/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-nssf/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-pcf/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-pcrf/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-scp/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-sgwc/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-sgwu/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-smf/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-udm/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-udr/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs-upf/jammy,now 2.6.4~jammy amd64 [installed,automatic]
open5gs/jammy,now 2.6.4~jammy amd64 [installed]
```
- Linux 5.15.0-69-generic

brendan-mcauliffe, 2023-09-27:
Have you enabled IP forwarding? See[ this section](https://open5gs.org/open5gs/docs/guide/01-quickstart/#:~:text=Adding%20a%20route%20for%20the%20UE%20to%20have%20WAN%20connectivity) of the Open5GS docs.

If the UE is not being assigned a PDU address, despite being configured correctly in the core and connecting to the network, this is likely the issue.

rohitdwivedula, 2023-10-03:
Hi Brenden - thanks for responding. I was executing all the commands listed in the subsection you pointed out. Quick question - do I need to change the IPs present in these commands: 

```bash
sudo iptables -t nat -A POSTROUTING -s 10.45.0.0/16 ! -o ogstun -j MASQUERADE
sudo ip6tables -t nat -A POSTROUTING -s 2001:db8:cafe::/48 ! -o ogstun -j MASQUERADE
```

My understanding was that all the (default) instructions in this guide were targetted at people running everything on only node. If I do need to modify the addresses in the command here, (1) what would I have to change them to? (2) I assume I would have to update `/etc/open5gs/upf.yaml` as well - the below lines are present in this config file:
```
     subnet:
       - addr: 10.45.0.1/16
       - addr: 2001:db8:cafe::1/48
```

brendan-mcauliffe, 2023-10-04:
Hi @rohitdwivedula, if you have open5gs running in a 'vanilla' set-up then the above commands should be fine. 

On taking another look at your config file, there is a mismatch between the `netns` you create and the one you assign to the ue in the config file. You should fix this. I would also recommend not running all of your commands in one input to the command line, this may be more efficient, but it will probably lead to some errors as you can easily forget something. 

You should also double check that the credentials of the UE in the core match those in the config file.

Aniket621, 2023-10-09:
Hi all i am also facing the same issue with srsUE with srsGNB and ZMQ Radio.

s5uishida, 2023-10-09:
Hi,

I couldn't make the latest `main` branch work as [this issue](https://github.com/srsran/srsRAN_Project/issues/241), so I built it based on srsRAN Project 23.5 (`release` version) as follows.

https://github.com/s5uishida/build_srsran_5g_zmq

And I made it work with Open5GS as follows.

https://github.com/s5uishida/open5gs_5gc_srsran_sample_config

FYI.

Aniket621, 2023-10-09:
Hi The same part i have follow the same process

s5uishida, 2023-10-10:
Hi,

In my environment, the latest `main` branch worked. It seems I misunderstood.

Sorry.

Aniket621, 2023-10-11:
Hi s5uishida, 

Can u please share your main branch version.

Aniket621, 2023-10-13:
Hi all, 

After change in configuration and rebuild the gnb and ue both then now srsUE is able to attached with srsgnb and Open5gs. I have some query regarding this setup. 

[image]

a) After UE attached turn_srsue interface is not coming although it is configured.  
b) I am able to ping the IP in both direction like UL and DL. 
c) How can i enable the ue inactivity timer and configuration in gnb.

Aniket621, 2023-10-18:
Hi all Can u please update above my query.

AzeezEbrahim, 2023-11-01:
Hi everyone,
It is not i problem you just need to wait for the gnb to load the zmq stuff before running the UE command ...

First run the gnb with its configuration normally,
`sudo gnb -c ~/gnb_zmq.yaml`
 then wait until the zmq configuration is finished you can see it by 
watching the gnb.log file at `/tmp` folder
`watch -n 1 tail gnb.log`
you can run the ue command after seeing this lines which may appear after ~6 seconds (depends on you processor)
[image]

then run UE
`sudo srsRAN_4G/build/srsue/src/srsue ue_zmq.conf`

[image]

ismagom, 2023-11-08:
It seems the original issue has been fixed. Please open new issues if needed.

closed 2023-11-08 by ismagom