srsran/srsRAN_Project #227: ORAN CU & DU Separate software package

Manimaranchandran, 2023-09-29:
I would like to deploy ORAN CU & DU using seperate software whereas currently available srsRAN project has whole package for gNB. Could you support for CU & DU software if available. This is to ensure the internal communication of these elements works as expected along with RU. Thanks

yc541, 2023-09-29:
CU/DU split is on the roadmap, have a look at the link below
[https://docs.srsran.com/projects/project/en/latest/general/source/2_features_and_roadmap.html](url)

Manimaranchandran, 2023-09-29:
@yc541 - As of now we can use srsRAN project gNB which has combined with CU&DU correct?

andrepuschmann, 2023-09-29:
>  As of now we can use srsRAN project gNB which has combined with CU&DU correct?

That's right. Note that you can still observe the communication between CU and DU over F1AP and E1AP (log and PCAP) as well as to the RU over OFH.

closed 2023-09-29 by andrepuschmann

Meamarian, 2023-10-11:
@andrepuschmann  Hi, is there a way to see the f1-c and f1-u traffic too? i think you don't send it over the network.

andrepuschmann, 2023-10-11:
F1c over the above mentioned pcap. F1u you can't see as it's not packed and logged.