srsran/srsRAN_Project #241: gnb aborted (core dumped)

s5uishida, 2023-10-08:
## Issue Description ##

When build the latest main branch and run gnb, it aborted.

## Setup Details ##

```
# cd ~/srsRAN_Project
# mkdir build
# cd build
# cmake ../ -DENABLE_EXPORT=ON -DENABLE_ZEROMQ=ON -DAUTO_DETECT_ISA=OFF
# make -j`nproc`
```

## Expected Behavior ##

gnb should work without abnormal termination.

## Actual Behaviour ##

```
# cd ~/srsRAN_Project/build/apps/gnb
# ./gnb
/root/srsRAN_Project/include/srsran/adt/interval.h:53: constexpr srsran::interval<T, RightClosed>::interval(U, V) [with U = int; V = unsigned int; T = unsigned int; bool RightClosed = true]: 
Assertion `start_ <= stop_' failed - Invalid interval [2, 1).
Aborted (core dumped)
```

## Steps to reproduce the problem ##

## Additional Information ##

AlaiaL, 2023-10-10:
Hi Shigeru,

while I am trying to reproduce it locally would you be able to rebuild it in debug and run it again?
You can do it with this cmake command:
`
cmake ../ -DENABLE_EXPORT=ON -DENABLE_ZEROMQ=ON -DAUTO_DETECT_ISA=OFF -DCMAKE_BUILD_TYPE=Debug
`

Then the gNB would produce a backtrace and you can share it here. 

Regards.

closed 2023-10-10 by s5uishida

s5uishida, 2023-10-11:
Hi @AlaiaL 

I got the cause.

When the number of CPU cores in the VM was 1, an error occurred at starting gnb, but when the number of CPU cores in the VM was 2 or more, the error did not occur and it worked.

Previously, it worked even with 1 core, but that seems to have changed recently.

AlaiaL, 2023-10-11:
Hi @s5uishida,
 
Thanks for the info. I will look at it.

Regards,