srsran/srsRAN_Project #244: E2 connection with FlexRIC does not work

jsalescpqd, 2023-10-09:
## Issue Description ##
The E2 interface connection with srsRAN 5G and FlexRIC does not work as expected. We followed the tutorial from https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html

## Setup Details ##
Server hosts srsRAN 5G: Intel(R) Xeon(R) Gold 6348 CPU @ 2.60GHz, 56 cores, 112 threads, 256 GB RAM, 256 Gi SSD
VM hosts FlexRIC: 50 GB HD, 8GB RAM, 8 vCPUs

## Expected Behavior ##
E2 Setup connection between srsRAN and FlexRIC, exchanging E2 related messages (setup request/response, compatible service models)

## Actual Behaviour ##
Only INIT messages are exchanged, then ABORT message is seen.

## Steps to reproduce the problem ##
Just run srsRAN and FlexRIC.
srsRAN commit 5e6f50a202c6efa671d5b231d7c911dc6c3d86ed
FlexRIC version specified in the tutorial (with patch already applied)
[e2.zip](https://github.com/srsran/srsRAN_Project/files/12849867/e2.zip)

## Additional Information ##
None.

yagoda, 2023-10-09:
Can you share the gnb.log?

Ensure log level is set to into beforehand.

Thanks,
Justin

dmd17, 2023-10-10:
we found that we had to set the local bind addr in order for SCTP connection to be correctly established.
https://docs.srsran.com/projects/project/en/latest/user_manuals/source/config_ref.html

e2:
  bind_addr:

jsalescpqd, 2023-10-10:
Hello,

Thanks for the quick answer. Actually, we just made it work. It was a problem regarding the timeout for the SCTP connection. When we commented those lines in our config .yaml file, it worked.

Thanks!

pgawlowicz, 2023-10-23:
Could you share the SCTP connection timeout that worked for you?

closed 2023-11-14 by yagoda