srsran/srsRAN_Project #248: "Network is unreachable" when connecting to ORAN-SC near-RT RIC

hoangtuansu, 2023-10-10:
## Issue Description ##
srsRAN gNB failed to connect SCTP socket of ORAN-SC nearRT-RIC . I followed  the tutorial from https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html and only replace FlexRIC by ORAN-SC nearRT-RIC.

[image]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
Server host srsRAN gNB and Open5GS: Intel(R) Xeon(R) Gold 6348 CPU @ 2.60GHz, 96 cores, 192 threads, 256 GB RAM, 500 Gi SSD.

## Expected Behavior ##
The connection should be successfully created as it did with commit c54bcfe

[image]

## Actual Behaviour ##
Only NGAP messages are exchanged. 2 other files (gnb_e2ap, gnb_mac) are empty.

## Steps to reproduce the problem ##
srsRAN commit 5e6f50a20
ORAN-SC nearRT-RIC H version, deployed on K8s and exposed its pod, i.e. e2term, via a NodePort service.

[srs.zip](https://github.com/srsran/srsRAN_Project/files/12859508/srs.zip)
[gnb.log](https://github.com/srsran/srsRAN_Project/files/12859513/gnb.log)

hoangtuansu, 2023-10-11:
Close. Fix by adding bind_addr to config file.

closed 2023-10-11 by hoangtuansu

faratell, 2024-07-13:
Hello. I was installed near RT RIC from bellow link and all pods are ok running:
https://gerrit.o-ran-sc.org/r/gitweb?p=ric-plt/ric-dep.git;a=blob;f=new-installer/README.md;h=a4cb085c12a61bfd7421c8b04aee147232688c42;hb=HEAD

kubectl get pods -n ricplt
NAME READY STATUS RESTARTS AGE
deployment-ricplt-a1mediator-7fdbb756c5-2qrx7 1/1 Running 1 (15h ago) 15h
deployment-ricplt-appmgr-6f445568d9-8xtpd 1/1 Running 0 15h
deployment-ricplt-e2mgr-647999fcbc-zl755 1/1 Running 3 (15h ago) 15h
deployment-ricplt-e2term-alpha-5cdf68cb54-kcjlx 1/1 Running 0 15h
deployment-ricplt-rtmgr-86c96d6b6-mjwrv 1/1 Running 0 15h
deployment-ricplt-submgr-64c9b8cf89-kwd9h 1/1 Running 0 15h
ricplt-influxdb-0 1/1 Running 0 15h
statefulset-ricplt-dbaas-server-0 1/1 Running 0 15h

Also i set my mcc and mnc in near RT RIC yaml file. ric ip an aux ip are also set as ip of my near RT RIC machine.

extsvcplt:
ricip: "192.168.75.154"
auxip: "192.168.75.154"
%%
e2mgr:
e2mgr:
image:
registry: "nexus3.o-ran-sc.org:10002/o-ran-sc"
name: ric-plt-e2mgr
tag: 6.0.1
privilegedmode: false
globalRicId:
ricId: "AACCE"
mcc: "999"
mnc: "70"
rnibWriter:
stateChangeMessageChannel: RAN_CONNECTION_STATUS_CHANGE
ranManipulationMessageChannel: RAN_MANIPULATION
%%

i run open5gs and then gnb with e2 config as bellow :
e2:
enable_du_e2: true # Enable DU E2 agent (one for each DU insta>
e2sm_kpm_enabled: true # Enable KPM service module
addr: 192.168.75.154 # RIC IP address
bind_addr: 192.168.75.151 # A local IP that the E2 agent binds to fo>
port: 36422 # RIC port

I also installed srsRAN-project in another machine. But have this error:
./gnb -c gnb_zmq_e2.yaml

--== srsRAN gNB (commit 40b17b429) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in executor blocking mode.
N2: Connection to AMF on 192.168.75.150:38412 completed
Cell pci=1, bw=10 MHz, 1T1R, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz

Connecting to NearRT-RIC on 192.168.75.154:36422
Failed to connect SCTP socket to 192.168.75.154:36422. error="Connection refused" timeout=0ms
srsRAN ERROR: Failed to create SCTP gateway.

Can you help me how can solve it?

pgawlowicz, 2024-07-16:
can you ping RIC PC from the srsGNB pc?

faratell, 2024-07-24:
> can you ping RIC PC from the srsGNB pc?

yes srsGNB can ping RIC

faratell, 2024-08-04:
Can any one help me?