srsran/srsRAN_Project #250: UE  attacching fails

Zplusless, 2023-10-11:
## Issue Description ##
I have successully setup ZMQ-based system for srsRAN and srsUE.  When I switch to USRP-b210, the srsUE stucks after "Attaching UE..." appears on the terminal.  I also tried COTS UE(mi 12 and mi 11pro), but the cell phone cannot find the network named 00101 at all. 

# Setup Details ##

#### gnb
Intel(R) Xeon(R) D-1747NTE CPU @ 2.50GHz  which has 10 cores 20 threads
Ubuntu 20.04.6 LTS (GNU/Linux 5.4.0-81-lowlatency x86_64)
UHD version: both v4.5.0.0 and v4.0.0.0 are tested

#### UE
Intel(R) Core(TM) i5-10500 CPU @ 3.10GHz  which has 6 cores 12 threads
Ubuntu 22.04.1 LTS (GNU/Linux 6.2.0-33-generic x86_64)
UHD version:  v4.5.0.0

### for both
on both server, `uhd_find_devices` and `uhd_usrp_probe` works fine
`srsran_performance.sh` are executed on both server.

Both usrp b210 are not equipped with GPSDO device

# Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

open5gs runs on the same server as gnb. AMF and UPF are configured as follows:
[amf.yaml.txt](https://github.com/srsran/srsRAN_Project/files/12868777/amf.yaml.txt)
[upf.yaml.txt](https://github.com/srsran/srsRAN_Project/files/12868779/upf.yaml.txt)

## config with band 3

#### gnb
[image]

[gnb_local_cn.yml.txt](https://github.com/srsran/srsRAN_Project/files/12868801/gnb_local_cn.yml.txt)
[gnb.log](https://github.com/srsran/srsRAN_Project/files/12868804/gnb.log)

the only warning in the gnb.log is
```bash
2023-10-11T07:30:48.612836 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 336.1 while message corresponds to 336.0
2023-10-11T07:30:48.612843 [FAPI    ] [W] [     0.0] Real-time failure in FAPI: Received UL_TTI.request message out of time. Current slot is 336.1 while message corresponds to 336.0
```

#### UE
After launching UE, the program outputs the following and gets stuck at attaching UE
[image]

[ue_lab.cn.conf.txt](https://github.com/srsran/srsRAN_Project/files/12868833/ue_lab.cn.conf.txt)
[ue.log](https://github.com/srsran/srsRAN_Project/files/12868834/ue.log)

the warning in ue.log are
```bash
2023-10-11T07:30:58.344461 [RRC-NR ] [I] Proc "Cell Selection" - Completed with failure.
2023-10-11T07:30:58.344465 [RRC-NR ] [W] Could not finish setup request. Deallocating dedicatedInfoNAS PDU
```

#### cell_search
Because the COTS UE cannot find the target 00101 network (commercial networks can be found), I use the `cell_search` tool in srsRAN_4G to search cell with band 3, with following outputs

```bash

Found 10 cells
Found CELL 1820.0 MHz, EARFCN=1350, PHYID=354, 6 PRB, 4 ports, PSS power=-33.3 dBm
Found CELL 1835.6 MHz, EARFCN=1506, PHYID=323, 50 PRB, 2 ports, PSS power=-29.0 dBm
Found CELL 1849.2 MHz, EARFCN=1642, PHYID=1, 125 PRB, 1 ports, PSS power=-22.6 dBm
Found CELL 1849.7 MHz, EARFCN=1647, PHYID=2, 150 PRB, 2 ports, PSS power=-13.6 dBm
Found CELL 1849.8 MHz, EARFCN=1648, PHYID=2, 125 PRB, 1 ports, PSS power=-12.2 dBm
Found CELL 1850.0 MHz, EARFCN=1650, PHYID=491, 100 PRB, 2 ports, PSS power=-9.0 dBm
Found CELL 1850.2 MHz, EARFCN=1652, PHYID=2, 15 PRB, 4 ports, PSS power=-11.8 dBm
Found CELL 1850.3 MHz, EARFCN=1653, PHYID=2, 6 PRB, 4 ports, PSS power=-14.2 dBm
Found CELL 1867.3 MHz, EARFCN=1823, PHYID=468, 15 PRB, 2 ports, PSS power=-26.6 dBm
Found CELL 1867.5 MHz, EARFCN=1825, PHYID=39, 150 PRB, 2 ports, PSS power=-25.0 dBm
```

but according the output of gnb, the gnb parameters are
```bash
Cell pci=1, bw=10 MHz, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz
```

## config with band 78
I tried another config from the tutorial with band 78. 
Good news, the gnb seems working better than config with band 3. But the attaching process still fails.

[image]
[image]

[gnb_git.yml.txt](https://github.com/srsran/srsRAN_Project/files/12869252/gnb_git.yml.txt)
[gnb.log](https://github.com/srsran/srsRAN_Project/files/12869363/gnb.log)
[ue_lab.github.conf.txt](https://github.com/srsran/srsRAN_Project/files/12869257/ue_lab.github.conf.txt)
[ue.log](https://github.com/srsran/srsRAN_Project/files/12869255/ue.log)

in gnb.log, the following warning appears only once. 
```
2023-10-11T08:14:52.537488 [FAPI    ] [W] [   236.1] Real-time failure in FAPI: Received late DL_TTI.request. Current slot is 236.1 while message corresponds to 236.0
```

still, in ue.log, 
```
2023-10-11T08:15:33.195195 [RRC-NR ] [I] Proc "Cell Selection" - Completed with failure.
2023-10-11T08:15:33.195199 [RRC-NR ] [W] Could not finish setup request. Deallocating dedicatedInfoNAS PDU
```

I tried to run `cell_search` with band 78, but it crashed
[image]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

### benchmark_rate
````bash
sudo ./benchmark_rate --rx_rate 11.52e6 --tx_rate 11.52e6
```
#### gnb
![image](https://github.com/srsran/srsRAN_Project/assets/10062702/2652ba92-5fe4-4903-a0dd-6a689331595f)

#### ue
![image](https://github.com/srsran/srsRAN_Project/assets/10062702/55b01776-2132-4577-95be-e19bcce80a19)

[gnb-benchmark_rate.txt](https://github.com/srsran/srsRAN_Project/files/12869462/gnb-benchmark_rate.txt)

[ue-benchmark_rate .txt](https://github.com/srsran/srsRAN_Project/files/12869467/ue-benchmark_rate.txt)

Zplusless, 2023-10-11:
Should I add GPSDO to the system? But the current issue seems to be unrelated to the clock.

brendan-mcauliffe, 2023-10-12:
Hi @Zplusless, 

First thing to note is that srsUE has some limitations that means it will only work in FDD bands with 10 MHz BW and 15 kHz SCS. See the "Limitations" of [this tutorial](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#limitations).  So only your first configuration will work - as outlined in the docs. 

Have you checked that the UE is registered correctly in the core? I looks like your issue is likely coming from there. I would also suggest that you run the `srsran_performance` script. You can find it in the "scripts" folder in the srsRAN Project main folder.

aris002, 2023-10-12:
I also have the same issue since yesterday but on the ZMQ based installation. Aside from the guided set-up from doc, I have run the srsran_performance and checked the subscriber info in the webui of Open5gs.  Both configuration files are as downloaded from the srsRAN doc. Running on 10 MHz BW and 15 kHz SCS.

Zplusless, 2023-10-13:
Thanks for your reply. 
1. I will focus on the first config, thanks.
2. I have run srsran_performance script on both server in the setup stage.
3. UE register seems good, because I have successfully run ZMQ based installation, which share the same UE registration and config parameter.

By the way, is "Real-time failure in FAPI" in the ran.log a serious error? 

> Hi @Zplusless,
>
> First thing to note is that srsUE has some limitations that means it will only work in FDD bands with 10 MHz BW and 15 kHz SCS. See the "Limitations" of [this [...]

Zplusless, 2023-10-13:
Hi @aris002, did your system work well before the rebuilding?

> I also have the same issue since rebuilding both projects yesterday but on the ZMQ based installation. Aside from the guided set-up from doc, I have run the srsran_performance and checked the subscriber info in the webui of Open5gs. Both configuration files are as downloaded from the srsRAN doc. Running on 10 MHz BW and 15 kHz SCS.

Zplusless, 2023-10-14:
Thanks，I will try the release version of srsRAN Project. Currently I'm using the latest master branch code.

> > Hi @aris002, did your system work well before the rebuilding?
> 
> > 
> 
> > > I also have the same issue since rebuilding both projects yesterday but on the ZMQ based installation. Aside from the guided set-up from doc, I have run the srsran_performance and checked the subscriber info in the webui of Open5gs. Both configuration files are as downloaded from the srsRAN doc. Running on 10 MHz BW and 15 kHz SCS.
> 
> 
> 
> Hi, I just made it run on my system. I had installed the 4G project in my first installation for the UE and then didn't uninstall from the .local directory before rebuild. I just cleared all the instances of srsRAN both Project and 4G from my system (just to be sure), reinstalled it all from scratch and everything works like before. Also, I built the srsRAN_Project from the release zip from May and it works just fine.

Zplusless, 2023-10-16:
I have tried release_23_5 and release_23_3, both failed to run with following message:

```bash
INI was not able to parse ru_sdr.++
Run with --help for more information.
```

Zplusless, 2023-10-16:
> Hi @Zplusless,
>
> First thing to note is that srsUE has some limitations that means it will only work in FDD bands with 10 MHz BW and 15 kHz SCS. See the "Limitations" of [this [...]

[image]
[image]

[UE]
[image]

tilldroemmer, 2023-10-16:
Hi @Zplusless you can also try your configuration with UERANSIM. 
We used it to confirm, that our Core was working as intended, sadly it did not help us to get srsRANProject running.
We are currently stuck at the state RRC Connection as in #199. Maybe the config files will help

Zplusless, 2023-10-17:
> Hi @Zplusless you can also try your configuration with UERANSIM. We used it to confirm, that our Core was working as intended, sadly it did not help us to get srsRANProject running. We are currently stuck at the state RRC Connection as in #199. Maybe the config files will help

Hi @tilldroemmer! Thanks for your reply. I have tried srsRAN-ZMQ setup and it works well. So the core network is not bo be blamed. Our problem is the UE(both srsUE and COTS UE) cannot find the network broadcasted by the gnb. As a result, the UE accachment fails. 

By the way, can you see the gnb network(00101 or any other plmn) on your COTS UE when you turn off the automaticly selecting ISP?

tilldroemmer, 2023-10-17:
@Zplusless we are not using COTS, therefore I cant answer that. In our case the Core gets the IMSI of the UE but does not hold on to the connection.
I will install everything on a new set of computers and will try with two b210 instead of x310.

Zplusless, 2023-10-18:
> @Zplusless we are not using COTS, therefore I cant answer that. In our case the Core gets the IMSI of the UE but does not hold on to the connection. I will install everything on a new set of computers and will try with two b210 instead of x310.

@tilldroemmer Thanks!!  I'm trying to setup connection both on the two b210 config and b210+COTS UE config. Can't wait to hear the good news!

Zplusless, 2023-10-24:
Connecting two USRP-B210 with SMA cable instead of using antenna solves the problem.   Although the performance is not satisfying, but it's another problem. I'll close this issue.

closed 2023-10-24 by Zplusless