srsran/srsRAN_Project #254: QoS Bearer Setup fails

hajoha, 2023-10-13:
## Issue Description ##
We've got a working default bearer (5QI 9) and in addition to it we tried to setup a GBR bearer with (5QI 1) and after the PDU Session Resource Modify Request is accepted by the gNB, the gNB crashes on a failed assertion: 
```
/opt/srsRAN_Project/include/srsran/support/async/async_event_source.h:45: srsran::async_event_source<T>::~async_event_source() [with T = mpark::variant<
asn1::elementary_procedure_option<asn1::f1ap::ue_context_release_complete_ies_container>, srsran::no_fail_response_path, srsran::transaction_timeout>]:            
Assertion `not has_subscriber()' failed - Observers must not outlive event sources.
```

gdb bt: 
```
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140736515257920) at ./nptl/pthread_kill.c:44                                                                           [0/1979]
#1  __pthread_kill_internal (signo=6, threadid=140736515257920) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140736515257920, signo=signo@entry=6) at ./nptl/pthread_kill.c:89                                                                                           
#3  0x00007ffff6d4e476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff6d347f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x000055555568564a in srsran::detail::print_and_abort(char const*, int, char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) 
    ()                                                                                                                                                                                        
#6  0x0000555555cc959c in srsran::srs_cu_cp::f1ap_cu_impl::handle_dl_rrc_message_transfer(srsran::srs_cu_cp::f1ap_dl_rrc_message const&) ()                                                   
#7  0x00005555557cf522 in srsran::srs_cu_cp::rrc_ue_f1ap_pdu_adapter::on_new_rrc_pdu(srsran::srb_id_t, srsran::byte_buffer const&, srsran::srs_cu_cp::ue_index_t) ()                          
#8  0x0000555555d579ad in srsran::srs_cu_cp::rrc_ue_impl::send_dl_dcch(srsran::srb_id_t, asn1::rrc_nr::dl_dcch_msg_s const&, srsran::srs_cu_cp::ue_index_t) ()                                
#9  0x0000555555d7648a in srsran::srs_cu_cp::rrc_reestablishment_procedure::send_rrc_reestablishment() ()                                                                                     
#10 0x0000555555d7ef54 in srsran::srs_cu_cp::rrc_reestablishment_procedure::operator()(srsran::detail::base_coro_frame<srsran::async_task<void>::promise_type>&) ()
#11 0x0000555555d241ac in srsran::srs_cu_cp::rrc_ue_impl::handle_rrc_reest_request(asn1::rrc_nr::rrc_reest_request_s const&) ()                                                               
#12 0x0000555555d28508 in srsran::srs_cu_cp::rrc_ue_impl::handle_ul_ccch_pdu(srsran::byte_buffer) ()                                                                                          
#13 0x00005555557d05b6 in srsran::srs_cu_cp::f1ap_rrc_ue_adapter::on_ul_ccch_pdu(srsran::byte_buffer) ()                                                                                      
#14 0x0000555555ccafe7 in srsran::srs_cu_cp::f1ap_cu_impl::handle_initial_ul_rrc_message(asn1::elementary_procedure_option<asn1::f1ap::init_ul_rrc_msg_transfer_ies_container> const&) ()     
#15 0x0000555555bc17ce in srsran::priority_task_worker_executor<(srsran::task_queue_priority)1, (srsran::concurrent_queue_policy)0, (srsran::concurrent_queue_policy)2>::execute(srsran::unique_function<void (), 64ul, false>) ()                                                           
#16 0x0000555555cc7c0b in srsran::srs_cu_cp::f1ap_cu_impl::handle_message(srsran::f1ap_message const&) ()
#17 0x000055555625ecef in srsran::srs_du::f1c_srb0_du_bearer::handle_sdu(srsran::byte_buffer_chain)::{lambda()#1}::operator()() const ()                                                      
#18 0x0000555555baf8fd in srsran::task_details::smallbuffer_table_t<srsran::priority_multiqueue_task_worker<(srsran::concurrent_queue_policy)0, (srsran::concurrent_queue_policy)2>::priority_multiqueue_task_worker(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::array<unsigned int, 2ul> const&, std::chrono::duration<long, std::ratio<1l, 1000000l> >, srsran::os_thread_realtime_priority, srsran::os_sched_affinity_bitmask const&)::{lambda()#1}, void>::call(void*) const ()                                                             
#19 0x0000555555b7f85c in std::thread::_State_impl<std::thread::_Invoker<std::tuple<srsran::unique_thread::make_thread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator
```

## Setup Details ##
- USRP B210
- srsRAN config (inside the zip)

## Expected Behavior ##
- Getting a second bearer with GBR 5QI 1

## Actual Behaviour ##
- gNB crashes

## Steps to reproduce the problem ##
- register the UE
- request 5QI 1 GBR bearer over NG Application Protocol (PDUSessionResourceModifyRequest)

## Additional Information ##
- we got also a 4GB coredump, if interested (not inside the zip)

Inside zip
- gnb_b200_5g_native.yml
- gnb_e1ap.pcap
- gnb_f1ap.pcap
- gnb.log
- gnb_mac.pcap
- srsran_crash_qos_specific_4.pcapng

[srsRAN_debug.zip](https://github.com/srsran/srsRAN_Project/files/12892803/srsRAN_debug.zip)

andrepuschmann, 2023-10-13:
Hey, 
thanks for the feedback. Looks like the UE does not like the Reconfiguration and then attempts to Reestablish. We'll look into it. Meanwhile you can try to start with `gnb cu_cp rrc --force_reestablishment_fallback=true`.
It wont fix the Reconfig issue but it should at least not crash the app.

Also, can you confirm the commit hash of your build? The info seems to be cut from the log file.

Thanks

andrepuschmann, 2023-10-13:
Ok, then leave the command away. I see there is an issue with the RRC Reconfiguration that we send. Could you please collect full logs logs in debug mode, specifically `mac_level=debug`, and share them please?

hajoha, 2023-10-13:
Is this fine? Run the gnb without the command.
[gnb.log](https://github.com/srsran/srsRAN_Project/files/12895590/gnb.log)

andrepuschmann, 2023-10-18:
This is fine, thanks. What is the phone you're using?

hajoha, 2023-10-18:
I tested it on these devices, but I haven't had the opportunity to try it with srsUE.

| Device Name | Android Version | Manufacturer | Model     | Radio Version                                   | SDK Version | Security Patch | SoC Model |
|------------|-----------------|-------------|-----------|-----------------------------------------------|------------|----------------|-----------|
| S23+       | 13              | Samsung     | SM-S916B  | S916BXXS3AWIG, S916BXXS3AWIG                  | 33         | 2023-10-01     | SM8550    |
| X20        | 13              | HMD Global  | Nokia X20 | -01086.1-MANNAR_GEN_PACK-1.501374.2.520129.12, -01086.1-MANNAR_GEN_PACK-1.501374.2.520129.12 | 33         | 2023-09-05     | SM4350    |

andrepuschmann, 2023-10-18:
Ok thanks - we were able to reproduce and are working on a fix.

andrepuschmann, 2023-10-19:
Hey @hajoha - we've pushed a possible fix to your issue to https://github.com/srsran/srsRAN_Project/tree/test

Could you please give this a try when you get a chance and report back?

Thanks

andrepuschmann, 2023-10-20:
Could you also provide the MAC PCAP, please? Best is to upload a fresh set of logs from the same run. There is a problem with the RRC Reconfiguration that the phone rejects.

herlesupreeth, 2023-10-20:
@hajoha One thing odd I noticed in your trace is that the UE is attempting to attach to "ims" APN first and then to "srsapn". In theory, the order should have be the opposite (having seen in 4G VoLTE scenario). Can you please go to APN settings and manually select "srsapn" APN (I believe "ims" would be selected) before attaching the UE?

The above is not a solution but just wondering whether thats the issue.

herlesupreeth, 2023-10-20:
Also, I see that 5GC is pushing packet filters for the PDU session 2 (which is `default` APN), which is weird. Have you configured the "default" APN to have packet filters in 5GC??

[image]

andrepuschmann, 2023-10-20:
> One thing odd I noticed in your trace is that the UE is attempting to attach to ims APN first and then to default. In theory, the order should have be the opposite (having seen in 4G VoLTE scenario).

It's weird I agree but I observed the same with my Moto. I guess the VoNR stack runs in parallel and just does it's own messaging.

hajoha, 2023-10-26:
Hi sorry for my late response.

> @hajoha One thing odd I noticed in your trace is that the UE is attempting to attach to `ims` APN first and then to `default`. In theory, the order should have be the opposite (having seen in 4G VoLTE scenario). Can you please go to APN settings and manually select `default` APN (the radio button on the right) before attaching the UE?
> 
> The above is not a solution but just wondering whether thats the issue.

Already did this, we are unable to disable IMS on the phones. 

> Also, I see that 5GC is pushing packet filters for the PDU session 2 (which is `default` APN), which is weird. Have you configured the `default` APN to have packet filters in 5GC??
> 
> [image]

As far as I'm aware, the IP corresponds to the iPerf3 Server IP that we set in the QoS Request in order to achieve a 5QI of 2.

[Here](https://webspace.fokus.fraunhofer.de/index.php/s/sH7j8eqjB4cyqwi) are the new logs.

herlesupreeth, 2023-10-27:
> As far as I am aware, yes, the 5GC is configured for filters on the default APN. But can you clarify what you mean by that question?'

If its configured for `default` APN then all good. Usually the scenario I have seen is that there is a signalling bearer for IMS and when a IMS call is placed then a dedicated bearer is used on top of that signalling bearer. And, the dedicated bearer is usually defined by a certain 5QI/QCI, packet filters and guaranteed bitrate. Since I had not seen a packet filters being defined for anything other than `ims` APN it made me a bit suspicious. Thanks for clarifying it.

andrepuschmann, 2023-10-27:
Hey @hajoha - there is something the phone(s) don't like about the RRC Reconfig, there are no major differences to other RRC Reconfigs so it's a matter of playing with the messages and see which field is the culprit. Can you confirm that exactly the same issues happens with both the above mentioned phones? Also, can you share your core network setup and how you trigger the pdu session addition/modification so we can reproduce that here?

hajoha, 2023-10-27:
sure we can manage to do this, i will send you an email

andrepuschmann, 2023-11-15:
The issue has been fixed in https://github.com/srsran/srsRAN_Project/tree/test . Thanks @hajoha for confirming.

closed 2023-11-15 by andrepuschmann

AEncinas19, 2023-11-23:
> The issue has been fixed in https://github.com/srsran/srsRAN_Project/tree/test . Thanks @hajoha for confirming.

[logsandconfs.zip](https://github.com/srsran/srsRAN_Project/files/13450623/logsandconfs.zip)
What is the srsue configuration file? I'm using the new test commit and gnb configuration allocated in /gnb_rf_b210_fdd_srsUE.yml and my srsUE after PDU Session Establishment receives an RRC Release due to LCID 1 doesn't exist. 

The .zip file contains:
- gNB configuration
- srsue configuration
- gNB logs
- srsue logs

Thanks in advance.