srsran/srsRAN_Project #258: gNB `bind_addr` issue with separate N2 and N3 networks

niloysh, 2023-10-16:
## Issue Description ##
When N2 and N3 are on different subnets, PDU session is established, but packets don't go through.

`bind_addr` in gNB config file refers to the N2 bind_addr on the machine where the gNB is running. This works, and the NGAP connection is established with AMF, and subsequently PDU session is established but packets don't go through.

We think there should be an additional parameter in the gNB config file for n3_bind_addr e.g., something like
```
upf:
    bind_addr: <bind_addr>  # N3 IP addr on gNB machine 
```

This is because if we have the same subnet for N2 and N3, then everything works correctly, and packets go through. 

It may be possible that the gNB is trying to use the N2 bind_addr for binding to the N3 connection right now.

We tried setting the `bind_addr` to N3 IP on the gNB machine, but that caused the SCTP connection with AMF to fail.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

We are using Open5GS v2.6.6 and latest release of srsRAN. We are running Open5GS in Kubernetes, where each NF has its own container. We have separate networks for N2 and N3. N2 network is 10.10.2.0/24 and N3 network is 10.10.3.0/24. 

## Expected Behavior ##
When N2 and N3 are on separate subnets, packets should go through.

## Actual Behaviour ##
PDU session established, but no packets go through.

## Steps to reproduce the problem ##
In the gNB config file,

```
amf:
  addr: 10.10.2.1       # N2 IP address of AMF
  bind_addr: 10.10.2.22    # N2 IP address on gNB machine
```

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2023-10-19:
yes, It's a known limitation. We already have this on the todo list.

wmwwimor, 2023-10-22:
> yes, It's a known limitation. We already have this on the todo list.

I have the same issue.When gnb and open5gs are both in the same subnet on a single virtual computer,especially according to the officail tutorial ,everying is ok when using loopback address.However, when I change the addr of amf or upf to the Virtual machine's static IP addr  which matches what is in the gnb.yml,there will be an erro said the SCTP connection issue.

ismagom, 2023-12-15:
This feature has been added in the last release, we now have the options: 

```
amf: 
  n2_bind_addr:
  n3_bind_addr:
```

closed 2023-12-15 by ismagom