srsran/srsRAN_Project #263: UE(ue_zmq.conf) failed to connect to Open5GS via gNodeB of srsRAN Project 23.10

s5uishida, 2023-10-21:
## Issue Description ##
UE(ue_zmq.conf) failed to connect to Open5GS via gNodeB of srsRAN Project 23.10.

## Setup Details ##
I built srsRAN_Project as below. (Virtualbox VM Ubuntu 22.04)
```
cd srsRAN_Project
mkdir build
cd build
cmake ../ -DENABLE_EXPORT=ON -DENABLE_ZEROMQ=ON -DAUTO_DETECT_ISA=OFF
make -j`nproc`
```

It was successful with the following versions of srsRAN_Project. The log is below.

srsRAN_Project `commit:5e6f50a202c6efa671d5b231d7c911dc6c3d86ed` on 2023.09.20
```
~/srsRAN_4G/build/srsue# src/srsue ue_zmq.conf
Reading configuration file ue_zmq.conf...

Built in Release mode using commit fa56836b1 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://192.168.0.122:2001,rx_port=tcp://192.168.0.121:2000,base_srate=11.52e6
Supported RF device list: zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://192.168.0.121:2000
CH0 tx_port=tcp://192.168.0.122:2001
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=334
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
PDU Session Establishment successful. IP: 10.45.0.2
RRC NR reconfiguration successful.
```
But it failed with the latest srsRAN Project 23.10. The log is below.

srsRAN_Project `commit:e38e418bda8432397b2fa7dc399cb7afde3c3b95` on 2023.10.20

```
~/srsRAN_4G/build/srsue# src/srsue ue_zmq.conf
Reading configuration file ue_zmq.conf...

Built in Release mode using commit fa56836b1 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://192.168.0.122:2001,rx_port=tcp://192.168.0.121:2000,base_srate=11.52e6
Supported RF device list: zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://192.168.0.121:2000
CH0 tx_port=tcp://192.168.0.122:2001
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=331
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=361
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=391
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=421
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=451
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=481
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=511
```
gnb_zmq.yaml)
```
# This configuration file example shows how to configure the srsRAN Project gNB to allow srsUE to connect to it. 
# This specific example uses ZMQ in place of a USRP for the RF-frontend, and creates an FDD cell with 10 MHz bandwidth. 
# To run the srsRAN Project gNB with this config, use the following command: 
#   sudo ./gnb -c gnb_zmq.yaml

gnb_id: 0x19B

slicing:
  - sst: 1
    sd: 1

amf:
  addr: 192.168.0.111                  # The address or hostname of the AMF.
  bind_addr: 192.168.0.121             # A local IP that the gNB binds to for traffic from the AMF.

ru_sdr:
  device_driver: zmq                # The RF driver name.
  device_args: tx_port=tcp://192.168.0.121:2000,rx_port=tcp://192.168.0.122:2001,base_srate=11.52e6 # Optionally pass arguments to the selected RF driver.
  srate: 11.52                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 10         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "00101"                     # PLMN broadcasted by the gNB.
  tac: 1                            # Tracking area code (needs to match the core configuration).
  pdcch:
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
```
ue_zmq.conf)
```
[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 11.52e6
nof_antennas = 1

device_name = zmq
device_args = tx_port=tcp://192.168.0.122:2001,rx_port=tcp://192.168.0.121:2000,base_srate=11.52e6

[rat.eutra]
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 3
nof_carriers = 1

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
opc  = E8ED289DEBA952E4283B54E88E6183CA
k    = 465B5CE8B199B49FAA5F0A2EE238A6BC
imsi = 001010000000000
imei = 353490069873319

[rrc]
release = 15
ue_category = 4

[nas]
apn = internet
apn_protocol = ipv4

[slicing]
enable = true
nssai-sst = 1
nssai-sd = 1

[gw]
#netns = ue1
ip_devname = tun_srsue
ip_netmask = 255.255.255.0

[gui]
enable = false
```
## Expected Behavior ##
UE (ue_zmq.conf) can connect to Open5GS via gNodeB of the latest srsRAN_Project.

## Actual Behaviour ##
UE (ue_zmq.conf) failed to connect to Open5GS via gNodeB of the latest srsRAN_Project.

## Steps to reproduce the problem ##

## Additional Information ##

herlesupreeth, 2023-10-21:
In order to fix this you can add the following to gnb_zmq.yaml under `cell_cfg`

```
  prach:
    prach_config_index: 1

```

herlesupreeth, 2023-10-21:
@ismagom It seems srsUE does not like `prach_config_index: 16`. We may have to revert that change or update the srsUE tutorial.

s5uishida, 2023-10-21:
Hi @herlesupreeth 

I solved this issue by your setting.

Thank you very much!

closed 2023-10-22 by s5uishida