srsran/srsRAN_Project #265: Unable to detect cell with Foxconn RPQN-7800E

xfoukas, 2023-10-22:
## Issue Description ##
I am trying to test srsRAN with a Foxconn RPQN 7800E RU. The RU, core and UEs have been tested and works with several other RAN stacks, so synchronization,the RU HW or network connectivity are not issues. However, all the UEs that I tested are unable to detect the cell.

## Setup Details ##

Stack: srsRAN (commit https://github.com/srsran/srsRAN_Project/commit/e38e418bda8432397b2fa7dc399cb7afde3c3b95);
CPU: Intel Xeon Gold 6338N CPU @ 2.20GHz
RAM: 196 GB;
OS: Mariner Linux
PTP Grandmaster: Qulsar Qg2. 
RU: Foxconn RPQN 7800E (firmware v1.6 or v3.1.15q)
The RU, DU server and PTP grandmaster are all connected to an Arista 7050CX3 switch. 
UE: OnePlus Nord N10 5G  and [Raspberry Pi with Quectel RM502Q-AE](https://sixfab.com/product/raspberry-pi-5g-development-kit-5g-hat/)

## Expected Behavior ##
The UEs should detect the cell and try to attach to it. 

## Actual Behaviour ##
The RAN comes up and attaches to the core and I can also see traffic to/from the RU on the switch (4.53Gbps towards the RU and 23.7Mbps towards the DU), as well as in the logs of the RUs that I tested. However, no UE actually detects the cell. 

## Steps to reproduce the problem ##
Here are the configuration files that I have used for this setup:

My gnb config file:

```
amf:
  addr: x.x.x.x                                              # The address or hostname of the AMF.
  bind_addr: x.x.x.x                                            # A local IP that the gNB binds to for traffic from the AMF.

ru_ofh:
  ru_bandwidth_MHz: 100                                           # RU instantaneous bandwidth.
  t1a_max_cp_dl: 470                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 258                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 429                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 285                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 196                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 50                                                  # Minimum T1a on User-Plane in microseconds.
  ta4_max: 75                                                    # Maximum Ta4 on User-Plane in microseconds.
  ta4_min: 0                                                      # Minimum Ta4 on User-Plane in microseconds.
  is_prach_cp_enabled: true                                       # Configures if Control-Plane messages should be used to receive PRACH messages.
  ignore_ecpri_payload_size: true                                 # Configures if eCPRI payload size field should be ignored by the eCPRI packet decoder.
  compr_method_ul: bfp                                            # Uplink compression method.
  compr_bitwidth_ul: 9                                            # Uplink IQ samples bitwidth after compression.
  compr_method_dl: bfp                                            # Downlink compression method.
  compr_bitwidth_dl: 9                                            # Downlink IQ samples bitwidth after compression.
  compr_method_prach: bfp                                         # PRACH compression method.
  compr_bitwidth_prach: 9                                         # PRACH IQ samples bitwidth after compression.
  enable_ul_static_compr_hdr: false                               # Configures if the compression header is present for uplink User-Plane messages (false) or not present (true).
  enable_dl_static_compr_hdr: false                               # Configures if the compression header is present for downlink User-Plane messages (false) or not present (true).
  iq_scaling: 0.8                                                 # IQ samples scaling factor applied before compression, should be a positive value smaller than 10.
  cells:
    - network_interface: eth1                                 # Ethernet interface name used to communicate with the RU.
      ru_mac_addr: x:x:x:x                              # RU MAC address.
      du_mac_addr: x:x:x:x                              # DU MAC address.
      vlan_tag: 1                                                 # VLAN tag value.
      prach_port_id: [4, 5, 6, 7]                                 # PRACH eAxC port values.
      dl_port_id: [0, 1, 2, 3]                                             # Downlink eAxC port values.
      ul_port_id: [0, 1, 2, 3]                                    # Uplink eAxC port values.

cell_cfg:
  dl_arfcn: 630684                                                # ARFCN of the downlink carrier (center frequency).
  band: 78                                                        # The NR band.
  channel_bandwidth_MHz: 100                                       # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                                                  # Subcarrier spacing in kHz used for data.
  plmn: "00101"                                                   # PLMN broadcasted by the gNB.
  tac: 1                                                          # Tracking area code (needs to match the core configuration).
  pci: 11                                                          # Physical cell ID.
  nof_antennas_dl: 4                                              # Number of transmission antennas.
  nof_antennas_ul: 1                                              # Number of reception antennas.
[... 150 lines cut ...]
<!-- RRH_LO_FREQUENCY_KHZ: Tx and Rx PLL LO Frequency in kHz(internal or external LO) -->
RRH_LO_FREQUENCY_KHZ = 3460260
<!-- RRH_TX_ATTENUATION: Tx attenuation value with 1-digit fraction for each layer (>10dB. NOTE: The attenuation value must be larget than 10dB) -->
RRH_TX_ATTENUATION = 30.0, 30.0, 30.0, 30.0
<!-- RRH_RX_ATTENUATION: Rx attenuation value with 1-digit fraction for each layer (<30dB) -->
RRH_RX_ATTENUATION = 0.0, 0.0, 0.0, 0.0
<!-- RRH_BB_GENERAL_CTRL: General control words for Baseband      -->
<!-- Bit[0] of 1st word: Enable the filtering of MAC address      -->
<!-- Bit[1] of 1st word: Enable the UL slot tick packets per port -->
<!-- Others: Reserved                                             -->
RRH_BB_GENERAL_CTRL = 0x0, 0x0, 0x0, 0x0
<!-- RRH_RF_GENERAL_CTRL: General control words for RF -->
RRH_RF_GENERAL_CTRL = 0x3, 0x0, 0x0, 0x0
<!--                           -->
<!--            PTPV2 Related  -->
<!--                           -->
<!-- RRH_PTPV2_GRAND_MASTER_MODE: 0: Unicast, 1:Multicast -->
RRH_PTPV2_GRAND_MASTER_MODE = 0
<!-- RRH_PTPV2_OBSERVATION_TIME: 0: Stop xRAN immediately when OOL; >0: Stop xRAN after specific seconds when OOL -->
RRH_PTPV2_OBSERVATION_TIME = 0
<!-- RRH_PTPV2_GRAND_MASTER_IP: IP address of grand-master -->
RRH_PTPV2_GRAND_MASTER_IP = x.x.x.x
<!-- RRH_PTPV2_SUB_DOMAIN_NUM: The sub-domain number -->
RRH_PTPV2_SUB_DOMAIN_NUM = 44
```

xfoukas, 2023-10-22:
I managed to detect the cell with both UEs with the v3.1 RU firmware, by changing my above config file with the following:

```
iq_scaling: 1.6

dl_port_id: [0]

nof_antennas_dl: 1
```

Now the attachement is failing during the random access process with the following:

```
2023-10-22T15:04:36.950119 [UL-PHY0 ] [I] [  758.19] PRACH:  rssi=-24.0dB detected_preambles=[{idx=1 ta=0.91us power=-1.4dB snr=-1.4dB}] t=150.7us
2023-10-22T15:04:36.952525 [SCHED   ] [I] [  759.10] Slot decisions cell=0 t=12us (1 PDSCH, 0 PUSCHs): RAR: ra-rnti=0x10b rb=[0..3) tbs=9
2023-10-22T15:04:36.952576 [DL-PHY10] [I] [  759.10] PDCCH: rnti=0x010b ss_id=1 format=1_0 cce=0 al=4 t=34.8us
2023-10-22T15:04:36.952622 [DL-PHY10] [I] [  759.10] PDSCH: rnti=0x010b prb=[0, 3) symb=[2, 14) mod=QPSK rv=0 tbs=9 t=42.1us
2023-10-22T15:04:36.956021 [SCHED   ] [I] [  759.17] Slot decisions cell=0 t=2us (0 PDSCHs, 1 PUSCH): UL: ue=1024 rnti=0x4605 h_id=0 ss_id=1 rb=[3..6) rv=0 tbs=11 msg3_delay=
7
2023-10-22T15:04:36.959363 [UL-PHY1 ] [I] [  759.17] PUSCH: rnti=0x4605 h_id=0 prb=[3, 6) symb=[0, 14) mod=QPSK rv=0 tbs=11 crc=KO iter=6.0 sinr=-19.1dB t=201.2us uci_t=0.0us
 ret_t=0.0us
2023-10-22T15:04:36.959567 [DL-PHY4 ] [I] [   760.4] PDCCH: rnti=0x4605 ss_id=1 format=0_0 cce=0 al=4 t=32.2us
2023-10-22T15:04:36.961509 [SCHED   ] [I] [   760.8] Slot decisions cell=0 t=2us (0 PDSCHs, 1 PUSCH): UL: ue=1024 rnti=0x4605 h_id=0 ss_id=1 rb=[3..6) rv=0 tbs=11 k2=4
2023-10-22T15:04:36.964977 [UL-PHY1 ] [I] [   760.8] PUSCH: rnti=0x4605 h_id=0 prb=[3, 6) symb=[0, 14) mod=QPSK rv=0 tbs=11 crc=KO iter=6.0 sinr=-27.1dB t=342.0us uci_t=0.0us
 ret_t=0.0us
2023-10-22T15:04:36.965104 [DL-PHY15] [I] [  760.15] PDCCH: rnti=0x4605 ss_id=1 format=0_0 cce=0 al=4 t=66.8us
2023-10-22T15:04:36.967017 [SCHED   ] [I] [  760.19] Slot decisions cell=0 t=6us (0 PDSCHs, 1 PUSCH): UL: ue=1024 rnti=0x4605 h_id=0 ss_id=1 rb=[3..6) rv=0 tbs=11 k2=4
2023-10-22T15:04:36.970654 [UL-PHY1 ] [I] [  760.19] PUSCH: rnti=0x4605 h_id=0 prb=[3, 6) symb=[0, 14) mod=QPSK rv=0 tbs=11 crc=KO iter=6.0 sinr=-20.8dB t=386.7us uci_t=0.0us
 ret_t=0.0us
2023-10-22T15:04:36.972565 [DL-PHY10] [I] [  761.10] PDCCH: rnti=0x4605 ss_id=1 format=0_0 cce=0 al=4 t=35.5us
2023-10-22T15:04:36.976015 [SCHED   ] [I] [  761.17] Slot decisions cell=0 t=2us (0 PDSCHs, 1 PUSCH): UL: ue=1024 rnti=0x4605 h_id=0 ss_id=1 rb=[3..6) rv=0 tbs=11 k2=7
2023-10-22T15:04:36.979916 [UL-PHY1 ] [I] [  761.17] PUSCH: rnti=0x4605 h_id=0 prb=[3, 6) symb=[0, 14) mod=QPSK rv=0 tbs=11 crc=KO iter=6.0 sinr=-31.9dB t=184.3us uci_t=0.0us
 ret_t=0.0us
2023-10-22T15:04:36.980067 [DL-PHY5 ] [I] [   762.5] PDCCH: rnti=0x4605 ss_id=1 format=0_0 cce=0 al=4 t=36.1us
2023-10-22T15:04:36.982018 [SCHED   ] [I] [   762.9] Slot decisions cell=0 t=3us (0 PDSCHs, 1 PUSCH): UL: ue=1024 rnti=0x4605 h_id=0 ss_id=1 rb=[3..6) rv=0 tbs=11 k2=4
2023-10-22T15:04:36.985871 [UL-PHY1 ] [I] [   762.9] PUSCH: rnti=0x4605 h_id=0 prb=[3, 6) symb=[0, 14) mod=QPSK rv=0 tbs=11 crc=KO iter=6.0 sinr=-28.2dB t=213.3us uci_t=0.0us
 ret_t=0.0us
2023-10-22T15:04:36.986017 [SCHED   ] [I] [  762.17] UL HARQ rnti=0x4605 cell=0 h_id=0: Discarding HARQ with tbs=11. Cause: Maximum number of reTxs 4 exceeded
```

Is this still related to the fronthaul or could there be some issue with my config or real-time processing?

ismagom, 2023-10-22:
It seems there is no uplink. Can you try lowering the RX attenuation? 

Also with that iq_scale the tx_power will be higher and if you don't adjust the ssb block power, the UE might send too low power. Do you have a spectrum analyzer or some kind of equipment that can look at the signal quality?

xfoukas, 2023-10-22:
Thanks for the suggestions. I tried to reduce the iq_scale and the RX attenuation, but it didn't make any difference. 

A couple of questions:

1. What is the approximate expected throughput on the fronthaul UL with my current configuration? At the moment, I can see 14.5Mbps on the UL (and 1.13Gbps on the DL). Is this normal or could it be that not all the UL traffic is being sent from the RU?

2. I observed the following in my gnb log file:

```
2023-10-22T15:04:41.569506 [OFH     ] [I] Received packets: rx_total=7880 rx_early=0, rx_on_time=1610, rx_late=6270
2023-10-22T15:04:42.569507 [OFH     ] [I] Received packets: rx_total=4800 rx_early=0, rx_on_time=701, rx_late=4099
2023-10-22T15:04:43.569504 [OFH     ] [I] Received packets: rx_total=4800 rx_early=0, rx_on_time=1081, rx_late=3719
2023-10-22T15:04:44.569505 [OFH     ] [I] Received packets: rx_total=4800 rx_early=0, rx_on_time=1266, rx_late=3534
2023-10-22T15:04:45.569507 [OFH     ] [I] Received packets: rx_total=4800 rx_early=0, rx_on_time=538, rx_late=4262
2023-10-22T15:04:46.569505 [OFH     ] [I] Received packets: rx_total=4800 rx_early=0, rx_on_time=851, rx_late=3949
2023-10-22T15:04:47.569506 [OFH     ] [I] Received packets: rx_total=4800 rx_early=0, rx_on_time=506, rx_late=4294
2023-10-22T15:04:48.569506 [OFH     ] [I] Received packets: rx_total=4800 rx_early=0, rx_on_time=1300, rx_late=3500
2023-10-22T15:04:49.569506 [OFH     ] [I] Received packets: rx_total=4800 rx_early=0, rx_on_time=2082, rx_late=2718
2023-10-22T15:04:50.569505 [OFH     ] [I] Received packets: rx_total=4800 rx_early=0, rx_on_time=767, rx_late=4033
2023-10-22T15:04:51.569507 [OFH     ] [I] Received packets: rx_total=4800 rx_early=0, rx_on_time=961, rx_late=3839
```

Is this normal? Could there be some thread issue in processing the received packets? My server has 32 cores, so I don't understand why there are so many rx_late.

AlaiaL, 2023-10-23:
> Is this normal? Could there be some thread issue in processing the received packets? My server has 32 cores, so I don't understand why there are so many rx_late.

All the lates you are seeing are related to `ta4_max: 75 ` parameter in your configuration, that value seems to low, so try to increase it. As a reference, we are using here `ta4_max: 200` , but it may need some adjustment for your specific configuration.

ismagom, 2023-10-25:
Can you send gnb.log and console trace please?

ismagom, 2023-11-08:
Hi @xfoukas, any update on the original issue?

closed 2023-12-15 by ismagom