srsran/srsRAN_Project #271: ACC100 LDPC decoder

Shadowcaster77, 2023-10-24:
## Issue Description ##
LDPC accelerator through bbdev interface

## Setup Details ##
I attended the first srsRAN workshop this week, and during the meeting, I noticed during the presentation that in the Release 23.10, there is a feature that supports the Intel ACC100 FEC, but I was not able to find it in the codebase. It would really be helpful if you can point me to the correct location. Thanks for your time.

ofontbach, 2023-10-25:
Hi @Shadowcaster77,

first of all, thanks for your interest in the srsRAN project and for attending the workshop. We hope you enjoyed it and found it fruitful.

The ACC100 FEC support has not been yet released publicly but will be included in an upcoming release. Please, make sure to stay tuned for updates to the srsRAN project repository.

Best regards

ismagom, 2023-10-26:
Hey @Shadowcaster77 , thanks for coming to the workshop! Next time, please use Discussion section to ask questions like this. Use Issues only for bugs or problems with the code. 

thanks!

closed 2023-10-26 by ismagom