srsran/srsRAN_Project #280: du_example Segmentation fault error

FengShaocong, 2023-10-31:
## Issue Description ##
[Describe the issue in detail]
When I perform the build/apps/examples/du/du_example file, happen Segmentation fault error.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
srsRAN Project 23.5、B210

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]
```
Ubuntu:~/tools/srsRAN_Project/build/apps/examples/du$ sudo ./du_example -P b200_20MHz_n7
2023-10-31T05:44:23.912645 [DU_APP  ] [I] Loading profile: 'b200_20MHz_n7'
2023-10-31T05:44:23.912646 [DU_APP  ] [I] Starting du_example with DL_ARFCN=536020, UL_ARFCN=512020.0, DL center frequency 2680100000.0 Hz, UL center frequency 2560100000.0 Hz, tx_gain=60.0 dB, rx_gain=70.0 dB
2023-10-31T05:44:23.913092 [ALL     ] [I] Task worker "low_dl" started...
2023-10-31T05:44:23.913122 [ALL     ] [I] Task worker "low_ul" started...
2023-10-31T05:44:23.913765 [ALL     ] [I] Task worker "Ctrl-GNB" started...
2023-10-31T05:44:23.913800 [ALL     ] [I] Task worker "DU-CELL#0" started...
2023-10-31T05:44:23.914006 [ALL     ] [I] Task worker "UE#0" started...
2023-10-31T05:44:23.914060 [ALL     ] [I] Task worker "PHY_DL" started...
Available radio types: uhd and zmq.
2023-10-31T05:44:23.914109 [ALL     ] [I] Task worker "PHY_UL" started...
2023-10-31T05:44:23.914111 [ALL     ] [I] Task worker "low_prach" started...
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; DPDK_19.11; UHD_4.5.0.0-0-g471af98f
2023-10-31T05:44:23.914134 [ALL     ] [I] Task worker "radio_thread" started...
Making USRP object with args 'type=b200'
2023-10-31T05:44:25.389540 [RF      ] [D] Setting PPS source to 'internal' and clock source to 'internal'.
2023-10-31T05:44:25.392231 [RF      ] [D] Setting Tx Rate to 23.04 MHz.
2023-10-31T05:44:25.392301 [RF      ] [D] Setting Rx Rate to 23.04 MHz.
2023-10-31T05:44:25.392359 [RF      ] [D] Setting channel 0 Tx gain to 60.00 dB.
2023-10-31T05:44:25.399890 [RF      ] [D] Setting channel 0 Tx frequency to 2680.1 MHz.
2023-10-31T05:44:25.424170 [RF      ] [D] Setting channel 0 Rx gain to 70.00 dB.
2023-10-31T05:44:25.424213 [RF      ] [D] Setting channel 0 Rx frequency to 2560.1 MHz.
Segmentation fault
```

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

sauka, 2023-11-29:
Hi @FengShaocong ,

thanks for reporting the issue, it will be fixed in the coming release.

sauka, 2023-12-14:
Hi @FengShaocong ,

Could you try it again with the latest code? It should be fixed already.

closed 2023-12-15 by ismagom