srsran/srsRAN_Project #281: SR period of 16ms when using SCS 30 kHz

improperaffo, 2023-10-31:
## Issue Description ##
When starting the gNB with `sr_period_ms: 16` and `common_scs: 30` we get an assertion failed. In fact, looking at `srsran::srs_du::make_asn1_rrc_sr_resource()` in `lib/du_manager/converters/asn1_rrc_config_helpers.cpp:1081`, there is no option for 32 slots periodicity.

## Setup Details ##
```
cell_cfg:
  dl_arfcn: 632628
  band: 78
  channel_bandwidth_MHz: 20
  common_scs: 30
  plmn: "00101"
  tac: 1
  pci: 1
  tdd_ul_dl_cfg:
    dl_ul_tx_period: 4
    nof_dl_slots: 2
    nof_dl_symbols: 7
    nof_ul_slots: 1
    nof_ul_symbols: 6
  pucch:
    sr_period_ms: 16
```

## Expected Behavior ##
As the reference config states that SR period of 16 ms is supported, we were expecting it to work.

## Actual Behaviour ##
```
/opt/srsRAN_Project/lib/du_manager/converters/asn1_rrc_config_helpers.cpp:1148: asn1::rrc_nr::sched_request_res_cfg_s srsran::srs_du::make_asn1_rrc_sr_resource(const srsran::scheduling_request_resource_config&): 
Assertion failed - Invalid SR periodicity=32.
```

## Steps to reproduce the problem ##
Run the gNB with `sr_period_ms: 16` and `common_scs: 30`.

## Additional Information ##
In `include/srsran/asn1/rrc_nr/serving_cell.h:9008`,  `struct types_opts` also does not have `sl32`.

herlesupreeth, 2023-10-31:
Thanks for catching this (we will fix the validation in the next release). But in the spec its not possible to have value 32 slots for SR period (16 ms in SCS 30 Khz equates to 32 slots).

[image]

herlesupreeth, 2023-11-09:
The validation has been implemented in our internal code you should be able to see it in our next release

closed 2023-11-09 by herlesupreeth