srsran/srsRAN_Project #282: srsRAN Project install errors

hearmedrone, 2023-10-31:
## Issue Description ##
I have been trying to install srsRAN_project on a machine that has previously had it installed. I am installing on Arch Linux and have tried to install from source and using yay -Sy srsran-project-git and both appear to fail install at the same point

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
Linux archlinux 6.1.59-1-lts
USRP B210 with UHD installed

## Actual Behaviour ##
Building srsRAN_Project...
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAS_MAYBE_UNINITIALIZED
-- Performing Test HAS_MAYBE_UNINITIALIZED - Success
-- Performing Test HAVE_NON_VIRTUAL_DTOR
-- Performing Test HAVE_NON_VIRTUAL_DTOR - Success
-- Performing Test HAVE_SUGGEST_OVERRIDE
-- Performing Test HAVE_SUGGEST_OVERRIDE - Success
-- Performing Test HAVE_SHADOW
-- Performing Test HAVE_SHADOW - Success
-- Assertion level set to NORMAL
-- Found libdw: /usr/lib/libdw.so  
-- Found libbfd: /usr/lib/libbfd.so  
-- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR) 
-- Found Backward: /home/nuradio/.cache/yay/srsran-project-git/src/srsRAN_Project/cmake/modules  
-- Building with backward-cpp support
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- Checking for module 'mbedtls'
--   Package 'mbedtls', required by 'virtual:world', not found
-- MBEDTLS LIBRARIES: /lib/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /lib/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Found MbedTLS: /lib/libmbedcrypto.so  
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.10
-- FFTW3F LIBRARIES: /usr/lib/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: FFTW3F_STATIC_LIBRARY-NOTFOUND
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found FFTW3F: /usr/lib/libfftw3f.so  
-- Found GTest: /usr/lib/cmake/GTest/GTestConfig.cmake (found version "1.14.0")  
-- Found Threads: TRUE  
-- UHD LIBRARIES /lib/libuhd.so
-- UHD INCLUDE DIRS /usr/include
-- Found UHD: /lib/libuhd.so  
-- Checking for module 'yaml-cpp'
--   Found yaml-cpp, version 0.8.0
-- YAMLCPP LIBRARIES: /usr/lib/libyaml-cpp.so
-- YAMLCPP INCLUDE DIRS: /usr/include
-- Found YAMLCPP: /usr/lib/libyaml-cpp.so  
-- Performing Test HAVE_SSE
-- Performing Test HAVE_SSE - Success
-- SSE4.1 is enabled - target CPU must support it
-- Performing Test HAVE_AVX
-- Performing Test HAVE_AVX - Success
-- AVX is enabled - target CPU must support it
-- Performing Test HAVE_AVX2
-- Performing Test HAVE_AVX2 - Success
-- AVX2 is enabled - target CPU must support it
-- Performing Test HAVE_FMA
[... 209 lines cut ...]
                 from /home/nuradio/.cache/yay/srsran-project-git/src/srsRAN_Project/lib/scheduler/ue_scheduling/ue_configuration.cpp:23:
In member function ‘std::array<_Tp, _Nm>::value_type& std::array<_Tp, _Nm>::operator[](size_type) [with _Tp = long unsigned int; long unsigned int _Nm = 3]’,
    inlined from ‘constexpr srsran::bounded_bitset<N, LowestInfoBitIsMSB>& srsran::bounded_bitset<N, LowestInfoBitIsMSB>::operator=(const srsran::bounded_bitset<N, LowestInfoBitIsMSB>&) [with long unsigned int N = 135; bool LowestInfoBitIsMSB = false]’ at /home/nuradio/.cache/yay/srsran-project-git/src/srsRAN_Project/include/srsran/adt/bounded_bitset.h:346:15,
    inlined from ‘typename __gnu_cxx::__enable_if<(! std::__is_scalar<_Tp>::__value), void>::__type std::__fill_a1(_ForwardIterator, _ForwardIterator, const _Tp&) [with _ForwardIterator = srsran::bounded_bitset<135>*; _Tp = srsran::bounded_bitset<135>]’ at /usr/include/c++/13.2.1/bits/stl_algobase.h:919:11,
    inlined from ‘void std::__fill_a(_FIte, _FIte, const _Tp&) [with _FIte = srsran::bounded_bitset<135>*; _Tp = srsran::bounded_bitset<135>]’ at /usr/include/c++/13.2.1/bits/stl_algobase.h:977:21,
    inlined from ‘void std::fill(_ForwardIterator, _ForwardIterator, const _Tp&) [with _ForwardIterator = srsran::bounded_bitset<135>*; _Tp = srsran::bounded_bitset<135>]’ at /usr/include/c++/13.2.1/bits/stl_algobase.h:1007:20,
    inlined from ‘void apply_pdcch_candidate_monitoring_limits(frame_pdcch_candidate_list&, const srsran::bwp_info&)’ at /home/nuradio/.cache/yay/srsran-project-git/src/srsRAN_Project/lib/scheduler/ue_scheduling/ue_configuration.cpp:469:16,
    inlined from ‘void generate_crnti_monitored_pdcch_candidates(srsran::bwp_info&, srsran::rnti_t, srsran::pci_t)’ at /home/nuradio/.cache/yay/srsran-project-git/src/srsRAN_Project/lib/scheduler/ue_scheduling/ue_configuration.cpp:567:42:
/usr/include/c++/13.2.1/array:203:24: error: array subscript 3 is above array bounds of ‘std::__array_traits<long unsigned int, 3>::_Type’ {aka ‘long unsigned int [3]’} [-Werror=array-bounds=]
  203 |         return _M_elems[__n];
      |                ~~~~~~~~^
/usr/include/c++/13.2.1/array: In function ‘void generate_crnti_monitored_pdcch_candidates(srsran::bwp_info&, srsran::rnti_t, srsran::pci_t)’:
/usr/include/c++/13.2.1/array:109:55: note: while referencing ‘std::array<long unsigned int, 3>::_M_elems’
  109 |       typename __array_traits<_Tp, _Nm>::_Type        _M_elems;
      |                                                       ^~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [lib/scheduler/ue_scheduling/CMakeFiles/ue_sched.dir/build.make:90: lib/scheduler/ue_scheduling/CMakeFiles/ue_sched.dir/ue_configuration.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:9404: lib/scheduler/ue_scheduling/CMakeFiles/ue_sched.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
make: Leaving directory '/home/nuradio/.cache/yay/srsran-project-git/src/build'
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: srsran-project-git-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
srsran-project-git - exit status 4

robertfalkenberg, 2023-11-01:
Confirming this on Archlinux with GCC 13.2.1.
(This is probably a false alarm from very aggressive code optimization and analysis inside GCC.)

As a workaround, you can use the compiler `clang`:

```
# install clang if not already installed
yay -Sy clang

# set clang as default compiler in this terminal
export CC=/usr/bin/clang
export CXX=/usr/bin/clang++

# build srsRAN project, e.g.
yay -Sy srsran-project-git

```

closed 2023-11-01 by robertfalkenberg

robertfalkenberg, 2023-11-02:
I updated the AUR package today (disable WError), to allow build with GCC 13.2.1.