srsran/srsRAN_Project #285: Connection to NearRT-RIC failed

Mhdkb, 2023-11-01:
## Issue Description ##

When starting the connection to nearRT-RIC this crashes. 

## Setup Details ##
I am using srsRAN version 23.10 with flexRIC version e2ap-v2 and the patch specified in the documentation:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html

Both srsRAN and FlexRIC are on the same host. 

Log from the gNB:

--== srsRAN gNB (commit 374200dee) ==--

Connecting to AMF on 192.168.70.132:38412
Available radio types: uhd and zmq.
Connecting to NearRT-RIC on 127.0.0.1:36421
Cell pci=1, bw=10 MHz, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz

==== gNodeB started ===
Type <t> to view trace

## Actual Behaviour ##
The message error I get from RIC is the next:

Setting the config -c file to /usr/local/etc/flexric/flexric.conf
Setting path -p for the shared libraries to /usr/local/lib/flexric/
[NEAR-RIC]: nearRT-RIC IP Address = 127.0.0.1, PORT = 36421
[NEAR-RIC]: Initializing
[NEAR-RIC]: Loading SM ID = 147 with def = ORAN-E2SM-KPM
[NEAR-RIC]: Loading SM ID = 144 with def = PDCP_STATS_V0
[NEAR-RIC]: Loading SM ID = 148 with def = GTP_STATS_V0
[NEAR-RIC]: Loading SM ID = 142 with def = MAC_STATS_V0
[NEAR-RIC]: Loading SM ID = 145 with def = SLICE_STATS_V0
[NEAR-RIC]: Loading SM ID = 143 with def = RLC_STATS_V0
[NEAR-RIC]: Loading SM ID = 146 with def = TC_STATS_V0
[iApp]: Initializing ...
[iApp]: nearRT-RIC IP Address = 127.0.0.1, PORT = 36422
fd created with 6
Received message with id = 411, port = 50382
[E2AP] Received SETUP-REQUEST from PLMN   1. 1 Node ID 411 RAN type ngran_gNB
**nearRT-RIC: /home/flexric/src/ric/msg_handler_ric.c:88: generate_setup_response: Assertion `req->ran_func_item[i].def.len < 1023' failed.
Aborted (core dumped)**

yagoda, 2023-11-01:
Hi,
Check out discussion here: https://github.com/srsran/srsRAN_Project/discussions/270

For a the solution to this issue.

Regards,
Justin

Mhdkb, 2023-11-01:
Thank you very much!

closed 2023-11-14 by yagoda