srsran/srsRAN_Project #289: Assertion `has_value()' failed - Invalid tiny_optional<T> access

raul-parada, 2023-11-06:
## Issue Description ##
`Assertion `has_value()' failed - Invalid tiny_optional<T> access `when executing `sudo ./gnb -c gnb_zmq.yaml`

## Setup Details ##
Ubuntu 22.04.3 LTS
Docker Compose version v2.21.0
srsRAN_4G 23.04.1
srsRAN_Project 23.10.1

## Expected Behavior ##
Run gnb succesfully

## Actual Behaviour ##
Lower PHY in executor blocking mode.

--== srsRAN gNB (commit 374200dee) ==--

Connecting to AMF on 10.53.1.2:38412
Available radio types: zmq.
/home/rparada/srsRAN_Project/include/srsran/adt/tiny_optional.h:78: constexpr T& srsran::detail::base_tiny_optional<T, Args>::value() & [with T = std::unique_ptr<srsran::dft_processor>; T ...Args = {}]: 
Assertion `has_value()' failed - Invalid tiny_optional<T> access.
Aborted

## Steps to reproduce the problem ##
Followed this tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html
Installed everything from source except zmq
Run docker from srsRAN_Project  (log attached)
[log.txt](https://github.com/srsran/srsRAN_Project/files/13265511/log.txt)

 
## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

faluco, 2023-11-06:
Hello Raul,

Could you please recompile the gnb in Debug mode and share the backtrace printed after the assertion gets triggered?

To do so pass this `-DCMAKE_BUILD_TYPE=Debug` to cmake before rebuilding the gnb.

raul-parada, 2023-11-06:
```
sudo cmake ../ -DENABLE_EXPORT=ON -DENABLE_ZEROMQ=ON -DCMAKE_BUILD_TYPE=Debug

-- Assertion level set to PARANOID
-- Found libdw: /usr/lib/x86_64-linux-gnu/libdw.so  
-- Could NOT find libbfd (missing: LIBBFD_LIBRARY LIBBFD_INCLUDE_DIR) 
-- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR) 
-- Found Backward: /home/rparada/srsRAN_Project/cmake/modules  
-- Building with backward-cpp support
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.8
-- FFTW3F LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- UHD LIBRARIES UHD_LIBRARIES-NOTFOUND
-- UHD INCLUDE DIRS UHD_INCLUDE_DIRS-NOTFOUND
-- Could NOT find UHD (missing: UHD_LIBRARIES UHD_INCLUDE_DIRS) 
-- YAMLCPP LIBRARIES: /usr/lib/x86_64-linux-gnu/libyaml-cpp.so
-- YAMLCPP INCLUDE DIRS: /usr/include
-- FINDING ZEROMQ.
-- SSE4.1 is enabled - target CPU must support it
-- AVX is enabled - target CPU must support it
-- AVX2 is enabled - target CPU must support it
-- FMA is enabled - target CPU must support it
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- FFT_LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Building srsRAN version 23.10.1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rparada/srsRAN_Project/build
```

faluco, 2023-11-06:
Yes, build the gnb and run it, then share the backtrace after the assertion message is printed

raul-parada, 2023-11-06:
I got this now:

```
Lower PHY in executor blocking mode.

--== srsRAN gNB (commit 374200dee) ==--

Connecting to AMF on 10.53.1.2:38412
Failed to connect SCTP socket to 10.53.1.2:38412. error="Connection timed out" timeout=1396ms
srsGNB ERROR: Failed to create SCTP gateway.

Aborted (core dumped)
```

faluco, 2023-11-06:
This message is printed when the gnb cannot connect to the core network. Please make sure you are running a core network and that the correct IP addresses are set. You can check these settings in the following documentation link: https://docs.srsran.com/projects/project/en/latest/tutorials/source/amariUE/source/index.html?highlight=zmq#id4

raul-parada, 2023-11-06:
I have running 5GCore, however, it is not clear the IPs I should configure and where

raul-parada, 2023-11-06:
In the https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html says to use a given gnb.yaml file with a given configuration. Which IP rang should I use then? I would like to do the mentioned tutorial.

faluco, 2023-11-07:
As it seems you are using open5gs in a docker the ip addresses should be the ones in the gnb config file. Please check the open5gs logs for any errors when the gnb tries to connect and that those ip addresses are accessible. Also, check this troubleshooting section: https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html#core-network-not-running

ismagom, 2023-11-08:
Hi @raul-parada, do you have any update on the original issue?

mas-oliveira, 2023-12-18:
Hello,

I'm having problems with the same thing. 

marco@marco-virtual-machine:~/srsRAN_Project/build/apps/gnb$ sudo ./gnb -c gnb_zmq.yaml 

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in executor blocking mode.

--== srsRAN gNB (commit 55c984b55) ==--

Connecting to AMF on 127.0.0.5:38412
Available radio types: zmq.
/home/marco/srsRAN_Project/include/srsran/adt/tiny_optional.h:78: constexpr T& srsran::detail::base_tiny_optional<T, Args>::value() & [with T = std::unique_ptr<srsran::dft_processor>; T ...Args = {}]: 
Assertion `has_value()' failed - Invalid tiny_optional<T> access.
Aborted

I have tried to build and make the project again. I cannot see any error in the gnb.log file.