srsran/srsRAN_Project #302: Real-time failure in FAPI: Received late DL_TTI.request from slot 545.4

doudoulail, 2023-11-09:
## Issue Description ##
[Describe the issue in detail]
I tried to ofh link ran550, but there was such an error in the log. Besides, I also found that no IQ data was transmitted through packet capture
## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
amf:
  addr: xxxx                                                # The address or hostname of the AMF.
  bind_addr: xxxx                                       # A local IP that the gNB binds to for traffic from the AMF.

ru_ofh:
  gps_alpha: 0                          # Optional FLOAT (0). Sets the GPS alpha. Supported: [0 - 1.2288e+07].
  gps_beta: 0                           # Optional INT (0). Sets the GPS beta. Supported: [-32768 - +32767].
  ru_bandwidth_MHz: 100                                           # RU instantaneous bandwidth.
  t1a_max_cp_dl: 500                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 250                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 465                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 250                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 250                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 80                                                  # Minimum T1a on User-Plane in microseconds.
  ta4_max: 300                          # Optional UINT (300). Sets the Ta4 maximum value for User-Plane. Supported: [0 - 1960].
  ta4_min: 85                           # Optional UINT (85). Sets the Ta4 maximum value for User-Plane. Supported: [0 - 1960].
  is_prach_cp_enabled: true                                      # Configures if Control-Plane messages should be used to receive PRACH messages.
  is_dl_broadcast_enabled: false                                   # Set to true for a workaround over a firmware bug in the RAN550 when operating in SISO mode.
  ignore_ecpri_payload_size: true          # Optional BOOLEAN (0). Sets whether or not to ignore eCPRI payload size field value. Supported [0. 1].
  compr_method_ul: bfp                                            # Uplink compression method.  
  compr_bitwidth_ul: 9                                            # Uplink IQ samples bitwidth after compression.
  compr_method_dl: bfp                                            # Downlink compression method.
  compr_bitwidth_dl: 9                                            # Downlink IQ samples bitwidth after compression.
  compr_method_prach: bfp                                         # PRACH compression method.
  compr_bitwidth_prach: 16                                         # PRACH IQ samples bitwidth after compression.
  enable_ul_static_compr_hdr: false                               # Configures if the compression header is present for uplink User-Plane messages (false) or not present (true).
  enable_dl_static_compr_hdr: false                                # Configures if the compression header is present for downlink User-Plane messages (false) or not present (true).
  
  iq_scaling: 0.8                                                # IQ samples scaling factor applied before compression, should be a positive value smaller than 1.
  cells:
    - network_interface: ens3f0                                 # Ethernet interface name used to communicate with the RU.
      ru_mac_addr: xxxx                              # RU MAC address.
      du_mac_addr: xxxx                              # DU MAC address.
      vlan_tag: 3                                                 # VLAN tag value.
      prach_port_id: [0,1,2,3]                                          # PRACH eAxC port value.
      dl_port_id: [0,1,2,3]   #[0,1]                                           # Downlink eAxC port values.
      ul_port_id: [0,1,2,3]                                             # Uplink eAxC port values.

cell_cfg:
  dl_arfcn: 504990                                                # ARFCN of the downlink carrier (center frequency).
  band: 41                                                        # The NR band.
  channel_bandwidth_MHz: 20                                       # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                                                  # Subcarrier spacing in kHz used for data.
  plmn: "xxxx"                                                   # PLMN broadcasted by the gNB.
  tac: 1                                                          # Tracking area code (needs to match the core configuration).
  pci: 1                                                          # Physical cell ID.
  nof_antennas_dl: 4                                              # Number of transmission antennas.
  nof_antennas_ul: 1                                              # Number of reception antennas.    
  prach:
    prach_config_index: 159                                         # PRACH configuration index.
    prach_root_sequence_index: 1                                  # PRACH root sequence index.
    zero_correlation_zone: 0                                      # Zero correlation zone.
    prach_frequency_start: 12      # Offset in PRBs of lowest PRACH transmission occasion in frequency domain respective to PRB 0.
  tdd_ul_dl_cfg:
    nof_dl_slots: 7
    nof_dl_symbols: 6                         # Optional INT (0). Number of Downlink symbols at the beginning of the slot following full Downlink slots. Supported: [0-13]
    nof_ul_slots: 2
    nof_ul_symbols: 4                         # Optional INT (0). Number of Uplink symbols at the end of the slot preceding the first full Uplink slot. Supported: [0-13].
  pucch:
    #min_k1: 2
  pusch:
    #min_k2: 2

      #expert_cfg:
[... 37 lines cut ...]
2023-11-09T08:00:59.160809 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.160828 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.180151 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.180195 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.180413 [OFH     ] [W] Detected 2 lost messages
2023-11-09T08:00:59.180434 [OFH     ] [W] Detected 2 lost messages
2023-11-09T08:00:59.180503 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.180536 [OFH     ] [W] Detected 2 lost messages
2023-11-09T08:00:59.180738 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.180763 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.181093 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.181267 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.181284 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.181379 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.181398 [OFH     ] [W] Detected 2 lost messages
2023-11-09T08:00:59.181700 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.181823 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.181842 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.181961 [OFH     ] [W] Detected 1 lost messages
2023-11-09T08:00:59.181985 [OFH     ] [W] Detected 1 lost messages
## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

ismagom, 2023-11-09:
Hi @doudoulail , can you edit your comment and format it accordingly? 

```
use this for code or config files
```

It's very difficult to read otherwise. Thanks

doudoulail, 2023-11-09:
Okay, I'll upload it again

doudoulail, 2023-11-09:
> Hi @doudoulail , can you edit your comment and format it accordingly?
> 
> ```
> use this for code or config files
> ```
> 
> It's very difficult to read otherwise. Thanks

Hi @ismagom ,I have changed, thank you, the log I intercepted, because it is the same warning after

doudoulail, 2023-11-09:
> Hi @doudoulail , can you edit your comment and format it accordingly?
> 
> ```
> use this for code or config files
> ```
> 
> It's very difficult to read otherwise. Thanks

ismagom, 2023-11-09:
Can you also provide the following information? 

- CPU model
- RAN550 firmware version
- srsRAN commit hash
- Network setup, do you have a switch between the RAN550 and the gNB? 
- PTP setup. What devices are you using for PTP. Can you share PTP logs from the gNB server and RAN550? 

thanks

ismagom, 2023-11-09:
Also, I took the liberty to edit the subject, since this better describes the real issue

doudoulail, 2023-11-09:
Okay, thanks

> I used xeon Platinum 6336y@ 3.10GHz dual socket with 24 real cores / 48 logical cores each, 64G RAM, a 10G fiber link to the RU,Use linuxptp and a synce network card for synchronization on the machine where srsran is located.

> RAN550 firmware version: sw V0.8.1

> srsRAN commit hash：srsRAN Project 23.10.1 

> Not have a switch between the RAN550 and the gNB

> the logs
ptp4l[1221.966]: rms    6 max   14 freq -25784 +/-   9 delay   172 +/-   1
ptp4l[1223.091]: rms    5 max   10 freq -25778 +/-   8 delay   170 +/-   1
ptp4l[1224.216]: rms    6 max   11 freq -25781 +/-   9 delay   169 +/-   1
ptp4l[1225.341]: rms    5 max   10 freq -25783 +/-   8 delay   170 +/-   1

doudoulail, 2023-11-09:
I made sure the ptp was working and synchronized on the ran550 side

ismagom, 2023-11-09:
And in addition to the above comment: 

- Have you verified if you see any signal coming out the RU? 
- Have you looked at the C/U-plane counters at RU logs?

doudoulail, 2023-11-10:
> And in addition to the above comment:
> 
> * Have you verified if you see any signal coming out the RU?
> * Have you looked at the C/U-plane counters at RU logs?

> Yeah, I saw a U-plan emanating from the ru side and found a C/U-plane input from the oru counter

doudoulail, 2023-11-10:
I made some configuration changes, and no cpu frequency scaling is applied during the application run，But it still shows up as 0.
```
is_dl_broadcast_enabled: true 
 iq_scaling: 1.7 #0.27                                                # IQ samples scaling factor applied before compression, should be a positive value smaller than 1.
  cells:
    - network_interface: ens3f0                                 # Ethernet interface name used to communicate with the RU.
      ru_mac_addr: 00:11:22:33:44:55                              # RU MAC address.
      du_mac_addr: 50:7c:6f:53:ea:4c                              # DU MAC address.
      vlan_tag: 3                                                 # VLAN tag value.
      prach_port_id: [0]                                          # PRACH eAxC port value.
      dl_port_id: [0,1]   #[0,1]                                           # Downlink eAxC port values.
      ul_port_id: [0]                                             # Uplink eAxC port values.

cell_cfg:
  dl_arfcn: 504990                                                # ARFCN of the downlink carrier (center frequency).
  band: 41                                                        # The NR band.
  channel_bandwidth_MHz: 100                                       # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                                                  # Subcarrier spacing in kHz used for data.
  plmn: "xxxx"                                                   # PLMN broadcasted by the gNB.
  tac: 1                                                          # Tracking area code (needs to match the core configuration).
  pci: 1                                                          # Physical cell ID.
  nof_antennas_dl: 1                                              # Number of transmission antennas.
  nof_antennas_ul: 1                                              # Number of reception antennas.  
```
I found some other problems, and still no correlation IQ transfer
[gnb.log](https://github.com/srsran/srsRAN_Project/files/13314810/gnb.log)

doudoulail, 2023-11-11:
Hi @ismagom ,Thanks,I have now discovered a new problem and I am not sure if it is because of my problem,Whether I set whether to compress or not, type1 always shows no compression and the compression bit width is 0.
[image]
[gnb.log](https://github.com/srsran/srsRAN_Project/files/13324533/gnb.log)

```
amf:
  addr: xxxx                                               # The address or hostname of the AMF.
  bind_addr: xxxx                                            # A local IP that the gNB binds to for traffic from the AMF.

ru_ofh:
  ru_bandwidth_MHz: 100                                           # RU instantaneous bandwidth.
  t1a_max_cp_dl: 400                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 150                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 465                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 250                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 250                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 80                                                  # Minimum T1a on User-Plane in microseconds.
  is_prach_cp_enabled: true                                      # Configures if Control-Plane messages should be used to receive PRACH messages.
  is_dl_broadcast_enabled: true 
  compr_method_ul: bfp                                            # Uplink compression method.
  compr_bitwidth_ul: 9                                            # Uplink IQ samples bitwidth after compression.
  compr_method_dl: bfp                                            # Downlink compression method.
  compr_bitwidth_dl: 9                                            # Downlink IQ samples bitwidth after compression.
  compr_method_prach: bfp                                         # PRACH compression method.
  compr_bitwidth_prach: 9                                         # PRACH IQ samples bitwidth after compression.
  enable_ul_static_compr_hdr: false                                # Configures if the compression header is present for uplink User-Plane messages (false) or not present (true).
  enable_dl_static_compr_hdr: false                                # Configures if the compression header is present for downlink User-Plane messages (false) or not present (true).
  iq_scaling: 1.5                                                # IQ samples scaling factor applied before compression, should be a positive value smaller than 10.
  cells:
    - network_interface: ens3f0                                 # Ethernet interface name used to communicate with the RU.
      ru_mac_addr: 00:11:22:33:44:55                              # RU MAC address.
      du_mac_addr: 50:7c:6f:53:ea:4c                              # DU MAC address.
      vlan_tag: 3                                                 # VLAN tag value.
      prach_port_id: [4]                                          # PRACH eAxC port value.
      dl_port_id: [0]                                          # Downlink eAxC port values.
      ul_port_id: [0]                                          # Uplink eAxC port values.

cell_cfg:
  dl_arfcn: 504990                                                # ARFCN of the downlink carrier (center frequency).
  band: 41                                                        # The NR band.
  channel_bandwidth_MHz: 100                                      # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                                                  # Subcarrier spacing in kHz used for data.
  plmn: "xxxx"                                                   # PLMN broadcasted by the gNB.
  tac: 1                                                          # Tracking area code (needs to match the core configuration).
  pci: 1                                                          # Physical cell ID.
  nof_antennas_dl: 1                                              # Downlink number of antennas.
  nof_antennas_ul: 1                                              # Uplink number of anntennas.
  prach:
    prach_config_index: 159                                         # PRACH configuration index.
    prach_root_sequence_index: 1                                  # PRACH root sequence index.
    zero_correlation_zone: 0                                      # Zero correlation zone.
    prach_frequency_start: 0                                      # Offset in PRBs of lowest PRACH transmission occasion in frequency domain respective to PRB 0.

log:
  filename: /home/sun/gnb.log                                          # Path of the log file.
  all_level: info                                              # Logging level applied to all layers.

pcap:
  mac_enable: false                                               # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap                                 # Path where the MAC PCAP is stored.
  ngap_enable: false                                              # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap                               # Path where the NGAP PCAP is stored.
```

doudoulail, 2023-11-11:
and type 3 is displayed normally

doudoulail, 2023-11-11:
I was using the latest srsran code, I did a print trace of the type1 module code and found that there was indeed a related problem, I did not continue to trace down, but changed the value at the end of the assignment

doudoulail, 2023-11-14:
> Hi @doudoulail ,
> 
> I see above you have a dual-socket CPU. It might be worth trying my suggestion [in this comment](https://github.com/srsran/srsRAN_Project/discussions/245#discussioncomment-7526944).
> 
> Let us know how it works.

Thank you. It's worked out one way or another

faluco, 2023-11-14:
Hi @doudoulail

Regarding the compression header stating that data is uncompressed when configured to be BFP9, this is expected. Taken from the OFH spec: point 7.5.2.10
```
This means the udCompHdr in a C-Plane message only has meaning for UL
U-Plane data (dataDirection = 0b) and should be set to 0x00 when dataDirection = 1b (DL data), and the receiving O-
RU shall ignore the received udCompHdr value.
```
This is why this field is all set to 0s. The user plane packet should contain the expected compression fields. This part of the spec is quite tricky and introduces some confussion. Hope this helps.

doudoulail, 2023-11-14:
Hi @faluco Thank you very much for your answer. I did have some misunderstandings about this part before.

closed 2023-11-14 by doudoulail