srsran/srsRAN_Project #305: MAC pcap file shows that the radio type is FDD in the uplink when using n41

kyrlon, 2023-11-11:
## Issue Description ##
While monitoring the power headroom report, I noticed that in the uplink of my MAC layer pcap that it reports that the radio type is FDD when I am using n41. Is this a mistake? Attached is the pcap where it reports the radio type as FDD in the uplink, but TDD in the downlink.
[gnb_mac.txt](https://github.com/srsran/srsRAN_Project/files/13324103/gnb_mac.txt)

## Expected Behavior ##
Radio type to show TDD

## Actual Behaviour ##
[image]

kyrlon, 2023-11-11:
I found here in this [file](https://github.com/srsran/srsRAN_Project/blob/374200deefd8e1b96fab7328525fd593a808a641/lib/mac/mac_ul/pdu_rx_handler.cpp#L384) that it is hard coded to FDD. Would this be the cause?

closed 2023-11-14 by kyrlon

andrepuschmann, 2023-11-18:
It's hard-coded because the duplex type doesn't reach the PCAP writer to set the actual value. It's a bit confusing I agree but I don't think it's causing any problem.