srsran/srsRAN_Project #306: Assertion `req->ran_func_item[i].def.len < 1023' failed.

raul-parada, 2023-11-13:
I'm following this tutorial https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html

I follow this steps:
1) Execute in a terminal `docker-compose up --build 5gc` from srsRAN_Project/docker ([log_open5gs.txt](https://github.com/srsran/srsRAN_Project/files/13333573/log_open5gs.txt))
2) Execute in another terminal `./flexric/build/examples/ric/nearRT-RIC` (
[log_ric.txt](https://github.com/srsran/srsRAN_Project/files/13333578/log_ric.txt)
)
3) Execute in another terminal `sudo ./gnb -c gnb_zmq.yaml `on  /srsRAN_Project/build/apps/gnb (
[log_gnb.txt](https://github.com/srsran/srsRAN_Project/files/13333581/log_gnb.txt)
)

When executing this third step, the terminal from step 2 receives this error:
```
[E2AP] Received SETUP-REQUEST from PLMN   1. 1 Node ID 411 RAN type ngran_gNB
nearRT-RIC: /home/rparada/flexric/src/ric/msg_handler_ric.c:88: generate_setup_response: Assertion `req->ran_func_item[i].def.len < 1023' failed.
Aborted (core dumped)
```
I've executed the step 2 and there is not error, however, when I run one UE  `sudo ip netns` add ue1 and `sudo ./srsue ue_zmq.conf `on srsRAN_4G/build/srsue/src (
[log_ue.txt](https://github.com/srsran/srsRAN_Project/files/13333606/log_ue.txt)
)

The UE stays on Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=511

yagoda, 2023-11-13:
Hi,
Check out the discussion here: #270 

For the solution to this issue.

Regards,
Justin

raul-parada, 2023-11-13:
I've modified msg_handler_ric.c l88 as `assert(req->ran_func_item[i].def.len < 2048 );`

And now I get: 

```
Received message with id = 411, port = 56531 
[E2AP] Received SETUP-REQUEST from PLMN   1. 1 Node ID 411 RAN type ngran_gNB
[NEAR-RIC]: Accepting RAN function ID 147 with def = `0ORAN-E2SM-KPM 
[NEAR-RIC]: Accepting interfaceType 0
*** stack smashing detected ***: terminated
Aborted (core dumped)
```

raul-parada, 2023-11-14:
Any solution?

yagoda, 2023-11-14:
This is detailed in the discussion as mentioned earlier #270, if you have increased the size in the assert and not modified the size of the dependent array you could see a stack smashing although I can't say for sure.

raul-parada, 2023-11-14:
I've modified the specific line mentioned in #270. What else should I modify? Why the same configuration worked 4 weeks ago and it is failing now? Please, give me more specific directions to fix the issue beside the mentioned #270. Thanks

yagoda, 2023-11-14:
Please see my above comment, the array above the assert, the assert protects the size of the memcpy below, if you increase the size of the assert, you must increase the size of the array involved in the copy. Again please note that this is not our software so we are not in a position to provide advice on its modification that we can stand over, this is just a guess as to what could be your issue.

closed 2023-11-14 by yagoda