srsran/srsRAN_Project #307: No radio devices available using BladeRF

ayamoheddine, 2023-11-13:
I'm trying to carry on the work done [https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html](url) using two machines: PC1 for srsRAN Project with gnb and open5gs CN and bladeRF and PC2 for srsRAN-4G for simulated UE. 

I'm getting this error: No radio devices available. 
srsGNB ERROR: unable to create radio session.
Aborted (core dumped).

Any idea if bladeRF is supported with srsRAN_Project as it was supported with srsRAN_4G? 

Any help is highly appreciated.

closed 2023-11-13 by brendan-mcauliffe

ayamoheddine, 2023-11-14:
Is there any possibility of extending the SDR options to support bladeRF as in srsRAN_4G?

brendan-mcauliffe, 2023-11-14:
A concrete roadmap for adding new drivers hasn't been confirmed yet. One option would be to try using the [SoapyUHD](https://github.com/pothosware/SoapyUHD/wiki) driver, which allows you to use a SoapySDR device with UHD. You would also need to use [SoapyBladeRF](https://github.com/pothosware/SoapyBladeRF/wiki) to enable you to use SoapySDR with the BladeRF. We have not tested this ourselves, so would not be able to provide much more guidance than this.