srsran/srsRAN_Project #309: srsUE attach fails through srsGNB to Open5GS (ZMQ)

hs3387, 2023-11-14:
## Issue Description ##
I wasn't sure whether to post this in the 4g git or the 5g git, but since it's a 5g core, gnb, and nr band use in srsUE, I'm posting here.

My UE attachment keeps getting rejected. It receives the release command then sits there until I exit the program
[image]

I don't see anything happening in the gnb console during this, it's just blank.
[image]

I see this series of packets flowing between gnb and the core
[image]

UE Sim config 
[image]

Core UE config
[image]

## Setup Details ##
srsUE and GNB running on ubuntu 20.04 LTS - connected through ZeroMQ
Open5GS running on another Ubuntu 20.04 (was able to attach a UE with a GNB running a USRP B210)

## Expected Behavior ##
UE should attach and authenticate successfully

## Actual Behaviour ##
UE/Core rejects authentication
Core logs and pcap don't specify reason for rejection

herlesupreeth, 2023-11-14:
Please post a pcap of NGAP interface and all the relevant configuration files.

just by looking at screenshot of pcap I see Authentication Reject so that indicates there is some issue in the UE SIM configuration (either at 5GC or Ki or OPc is wrong)

hs3387, 2023-11-20:
[gnbx.yaml.txt](https://github.com/srsran/srsRAN_Project/files/13406092/gnbx.yaml.txt)
[uezmq2.conf.txt](https://github.com/srsran/srsRAN_Project/files/13406093/uezmq2.conf.txt)
[tail amf.log.txt](https://github.com/srsran/srsRAN_Project/files/13406094/tail.amf.log.txt)
[pcaps.zip](https://github.com/srsran/srsRAN_Project/files/13406084/pcaps.zip)

This is the only info i could gather from pcap but no idea what to make of it

[image]

hs3387, 2023-11-20:
[amf.yaml.txt](https://github.com/srsran/srsRAN_Project/files/13406947/amf.yaml.txt)

herlesupreeth, 2023-11-20:
[image]

As mentioned earlier Authentication Reject indicates that there is some issue in the UE SIM configuration (either at 5GC or Ki or OPc is wrong)

I would suggest changing the following parameters in UE configuration file and give it a try (to eliminate the possibility of wrongly generated OPc value from Ki and OP)

```
opc  = 8E27B6AF0E692E750F32667A3B14605D
k    = 8baf473f2f8fd09487cccbd7097c6862
```

hs3387, 2023-11-27:
I changed this, still getting auth reject
[image]

hs3387, 2023-11-27:
Given that I can't find a specific reason for the auth reject (and logically the reason being in the sim config)

Could you walk me through setting up a new subscriber on Open5gs and creating a UE conf for srsue

hs3387, 2023-11-28:
@herlesupreeth 
It looks like srsue is only taking the first 5 digits of my IMSI as the PLMN
It needs to take 6?
How do I set this?

herlesupreeth, 2023-11-29:
Please post the pcap to know better where the issue is. I would suggest changing MNC from 010 to say 100 i.e. non-leading zero number.

pgawlowicz, 2024-01-23:
@hs3387 Any update on this issue?

hs3387, 2024-01-25:
Nope, I have tried different official guides and some random thoughts too.
Nothing seems to work.

Regardless of whyther Im using a physical sim (pcsc) or typing the usim
info in the ue config, I keep getting MAC failures. Even tried a sim that
works on a mobile phone with a physical gnb with the same core.

pgawlowicz, 2024-01-25:
Do you run the dockerized Open5gs provided in srsRAN_Project repo?

Azukaa, 2024-01-29:
I am facing exactly the same issue when I run an installed open5gs on my desktop. But when I run the dockerized version from srsran, it works. ANY tip to solve this?

azzukky, 2024-01-29:
I am facing exactly the same issue when I run an installed open5gs on my desktop. But when I run the dockerized version from srsran, it works. ANY tip to solve this?

pgawlowicz, 2024-01-29:
Note that the dockerized version is shipped with configs for each Open5GC service and the UE database is properly populated. You should be able to achieve the same effect with your own installation.

closed 2024-02-29 by pgawlowicz