srsran/srsRAN_Project #313: UE does not finish connection - remains Random Access Transmission

raul-parada, 2023-11-14:
I'm following this tutorial https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html

I follow this steps:

    1-Execute in a terminal `docker-compose up --build 5gc `from srsRAN_Project/docker
    2-Execute in another terminal .`/flexric/build/examples/ric/nearRT-RIC `    
    3-Execute in another terminal `sudo ./gnb -c gnb_zmq.yaml` on /srsRAN_Project/build/apps/gnb 
    4-Execute UE `sudo ip netns add ue1` and `sudo ./srsue ue_zmq.conf `on srsRAN_4G/build/srsue/src 

However, the UE stays on `Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=511` without succesfully complete the procedure (
[log_ue.txt](https://github.com/srsran/srsRAN_Project/files/13349563/log_ue.txt)
) 
All three first steps are successfully running.

pgawlowicz, 2023-11-14:
Hi,
please check if your gnb config has the following lines:

```
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE
```

raul-parada, 2023-11-14:
I don't have any gnb config file in srsRAN_4G/srsgnb as gnb config I'm using the yaml file (srsRAN_Project) provided by the tutorial named gnb_zmq.yaml. Should I include that line in this file?

pgawlowicz, 2023-11-14:
Could you provide logs from gnb?

raul-parada, 2023-11-14:
[log_gnb.txt](https://github.com/srsran/srsRAN_Project/files/13350042/log_gnb.txt)

pgawlowicz, 2023-11-14:
this is cmd output, logs are at /tmp/gnb.log

pgawlowicz, 2023-11-14:
Please check the gnb config, probably you are missing `prach_config_index: 1` see here: https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html#gnb

raul-parada, 2023-11-14:
Solved!

closed 2023-11-14 by raul-parada