srsran/srsRAN_Project #315: Issue Connecting UE to gNB on srsRAN Project

oliverxsch, 2023-11-15:
## Issue Description ##
UE can't attach.

## Setup Details ##
UHD 4.6.0
Ubuntu 22.04.1 LTS
i5 8500
First PC: 8GB Ram
Second PC: 16GB Ram

## Expected Behavior ##
UE doesn't attach

## Actual Behaviour ##
See attachment.

## Steps to reproduce the problem ##
Hello srsRAN community,

I'm encountering difficulties connecting the UE to the gNB in my setup. I am utilizing two PCs, both connected to a USRP X310 via 1GB Ethernet.

On the first PC, I am configuring the UE using the first USRP, with UHD and srsRAN(4G) installed, as srsRAN(5G) does not include a UE. I've attached the configuration file and console output from the attempted UE startup.

The second PC, connected to the second USRP, functions as the gNB. It has UHD, srsRAN5G, and Open5G installed. I've attached the configuration file and console output obtained when starting the gNB.

I followed the instructions provided at https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html.

Despite searching on Github for a solution, I haven't been able to resolve the issue. If you require any additional files, please let me know. I'm still a beginner in this field, so any guidance would be greatly appreciated.

## Additional Information ##
[gnb ue files.zip](https://github.com/srsran/srsRAN_Project/files/13366029/gnb.ue.files.zip)

brendan-mcauliffe, 2023-11-17:
Hi @oliverxsch, can you increase the gain values for both the UE and gNB? They are possibly too low. You should also run the performance script found [here](https://github.com/srsran/srsRAN_Project/tree/main/scripts) on both the UE and gNB machines. 

As a side note, when reporting issues it is always best to include the logs for both the gNB and UE in info mode, and the PCAPs. 

Try again with the increased gain values, and if it doesnt work then attach the logs and PCAPs.

oliverxsch, 2023-11-17:
> Hi @oliverxsch, can you increase the gain values for both the UE and gNB? They are possibly too low. You should also run the performance script found [...]

Hi, i would like to try that but because i tried resolving the issue for my self, the gNB now will not even start anymore. 
It always says: srsGNB Error: Failed to create SCTP gateway.
Do you have a solution?
I have the open5gs files and the gnb files in the appendix.
[files.zip](https://github.com/srsran/srsRAN_Project/files/13392030/files.zip)

brendan-mcauliffe, 2023-11-17:
Something is likely running in the background and using the port. Make sure that you have killed all previous instances of the gNB and restart the open5GS instance you are running.

oliverxsch, 2023-11-20:
Thanks for your help, i can try your tips on tuesday.

oliverxsch, 2023-11-21:
> Hi @oliverxsch, can you increase the gain values for both the UE and gNB? They are possibly too low. You should also run the performance script found [...]

> Something is likely running in the background and using the port. Make sure that you have killed all previous instances of the gNB and restart the open5GS instance you are running.

Thanks for your tips, and Open5Gs seams to be working now. But the UE still can't attach with different db-values.
I will send all logs and confs you might need.

[ue_gnb_5g_logs-and-conf.zip](https://github.com/srsran/srsRAN_Project/files/13427715/ue_gnb_5g_logs-and-conf.zip)

brendan-mcauliffe, 2023-11-22:
It looks like you have an issue with the MTU size on the gNB machine. Take a look at [this section](https://files.ettus.com/manual/page_usrp_x3x0_config.html#:~:text=core.wmem_max%3D33554432-,Configuring%20the%20MTU,-In%20order%20to) of the X310 user manual. Don't forget to run [this script](https://github.com/srsran/srsRAN_Project/tree/main/scripts) on both the UE and gNB machine and try again

oliverxsch, 2023-11-23:
> It looks like you have an issue with the MTU size on the gNB machine. Take a look at [this [...]

I tried that now, but it didnt work either.

brendan-mcauliffe, 2023-11-23:
Can you attach updated logs and pcaps?

oliverxsch, 2023-11-23:
Sure.
[23.11.23.zip](https://github.com/srsran/srsRAN_Project/files/13451339/23.11.23.zip)

oliverxsch, 2023-11-23:
I now tried running both from the same machine, same issue.

oliverxsch, 2023-11-24:
[srsranlogconf.zip](https://github.com/srsran/srsRAN_Project/files/13459357/srsranlogconf.zip)

> Your logs clearly show that the UE received the RRC connection setup and transmitted the acknowledgment in the PUCCH. You can see this here:
> 
> ```
> 2023-11-23T14:47:50.448779 [PHY1-NR] [I] [ 5485] PDCCH: cc=0, c-rnti=0x4601 dci=1_0 ss=common0 L=2 cce=0 f_alloc=0x1df t_alloc=0x0 vrb_to_prb_map=0 mcs=4 ndi=1 rv=2 harq_id=0 dai=0 pucch_tpc=1 pucch_res=3 harq_feedback=3 
> 2023-11-23T14:47:50.449080 [PHY1-NR] [I] [ 5485] PDSCH: cc=0 pid=0 c-rnti=0x4601 prb=(1,40) symb=(2,13) CW0: mod=QPSK tbs=325 R=0.303 rv=2 CRC=OK iter=4.0 evm=0.59 t_us=248 epre=+25.8 snr=-0.7 cfo=-8.3 delay=-0.0  ack_tti_tx=5489
>     0000: 3e 1f 84 b9 2d 0e 26 40 01 38 20 40 04 09 9a e0
>     0010: 05 80 08 8b d7 63 80 83 0f 00 03 e0 10 23 41 e0
> 2023-11-23T14:47:50.449083 [MAC-NR ] [I] [ 5485] DL 0:  reTX  tbs=325, rv=2, ack=OK, ndi=1
> 2023-11-23T14:47:50.449127 [PHY1-NR] [I] [ 5485] PUCCH: cc=0, f=1 prb=3:1 symb=0:14 cs=0 occ=0 rnti=0x4601 ack=1 , tti_tx=5489
> ```
> 
> Firstly, the gNB detected the MSG3 with very poor SNR (ideally this would be around 20 dB or more). After this, it does not detect the PUCCH transmission from the UE.
> 
> ```
> 2023-11-23T14:47:50.462574 [UL-PHY1 ] [I] [   546.6] PUSCH: rnti=0x4601 h_id=0 prb=[3, 6) symb=[0, 14) mod=QPSK rv=0 tbs=11 crc=OK iter=2.0 sinr=-2.5dB t=191.3us uci_t=0.0us ret_t=0.0us
> ...
> 2023-11-23T14:47:50.485450 [UL-PHY1 ] [I] [   548.9] PUCCH: rnti=0x4601 format=1 prb1=3 prb2=48 symb=[0, 14) cs=0 occ=0 ack=2 t=36.5us
> ...
> 2023-11-23T14:47:50.482519 [UL-PHY1 ] [I] [   548.6] PUCCH: rnti=0x4601 format=2 prb=[50, 51) prb2=na symb=[10, 12) csi1=2222 t=44.8us
> ```
> 
> Can you run the tests again, this time running the gNB PHY logs in debug? This will help us to know if the PUCCH is not detected by the gNB due to poor RF channel conditions or something else. If you don't know how to do this, check the documentation.

Okay, i hope this can help.

oliverxsch, 2023-11-24:
Ok, now this is weird. Sometimes im actually able to connect the UE to the gNB (with no change in the configuration). But shortly afterwards the connection is released. 
I will now list my setup because i believe its now hard to keep up with it with all the changes i have made.

First PC: 
Specs: i5 8500, 16GB Ram
OS: Ubuntu 22.04.03 LTS
UHD 3.15 LTS
Newest srsRAN Project and newest Open5Gs.
Connected via a 10G-Link to a USRP X310

Second PC:
Specs: i5 8500, 8GB Ram
OS: Ubuntu 22.04.03 LTS
UHD 4.1.0.5-3
Newest srsRAN-4G Project.
Connected via a 10G-Link to another USRP X310

Both USRPs are connected to an octoclock for their sync and clock.

Both USRPs are connected via cables not antennas with 6db attenuators.

Configuration and Logs (for the case when the UE actually connects, but releases the connection after a few seconds.)
Logs for PHY Level gnb are in debug, everything else is in info.
[logs and conf ping no answer.zip](https://github.com/srsran/srsRAN_Project/files/13459944/logs.and.conf.ping.no.answer.zip)

brendan-mcauliffe, 2023-11-28:
@oliverxsch your gain values are very low. The gNB reports an RLF (Radio Link Failure) just before it initiates the release. This points to a problem with your channel conditions. Tweak these values and see if you can improve stability. You should change one  at a time, i.e. increase the UE values, then gNB. Repeat until you get a stable connection. 

For reference this is the line in the logs: 
```
2023-11-24T13:59:01.708379 [DU-MNG  ] [I] ue=0: RLF detected. Timer of 31000 msec to release UE started...
```

oliverxsch, 2023-11-28:
> @oliverxsch your gain values are very low. The gNB reports an RLF (Radio Link Failure) just before it initiates the release. This points to a problem with your channel conditions. Tweak these values [...]

I played with the values a bit and i found out that for 
UE: tx 15 rx 10
GNB: tx 10 rx 10 it was stable (i could ping the UE from the gNB) but as soon as i pinged the gNB from the UE the connection crashed.
As a result i tried changing these values. But without success. Now these exact gain values lead to nothing. The UE can't connect anymore, just like before. 
The bahavior seems to be very random to me..
In the UE log i can now see that it is out of sync. But both USRPs are still connected to an external octoclock...
[logs28.11 no connection.zip](https://github.com/srsran/srsRAN_Project/files/13488909/logs28.11.no.connection.zip)

oliverxsch, 2023-11-28:
After shutting down the USRPs, waiting 3min and turning them back on. I was able to connect the UE to the GNB again, but the connection crashed again after a ping.

oliverxsch, 2023-11-28:
Now it works again, with no changes towards the gain values, but after a ping the connection crashed again. The behavior seems to be very random and not always reproducable.

oliverxsch, 2023-11-28:
I will now close the issue and open a new one, because the issue isn't that it does not connect but that it does not stay connected.

oliverxsch, 2023-11-28:
https://github.com/srsran/srsRAN_Project/issues/343
Here's the new issue. I would be very grateful for help.

closed 2023-11-28 by oliverxsch