srsran/srsRAN_Project #317: Ping from Core to UE doesn't work

raul-parada, 2023-11-16:
I'm following this tutorial https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html

I follow this steps:

    1-Execute in a terminal `docker-compose up --build 5gc `from srsRAN_Project/docker
    2-Execute in another terminal .`/flexric/build/examples/ric/nearRT-RIC `    
    3-Execute in another terminal `sudo ./gnb -c gnb_zmq.yaml` on /srsRAN_Project/build/apps/gnb 
    4-Execute UE `sudo ip netns add ue1` and `sudo ./srsue ue_zmq.conf `on srsRAN_4G/build/srsue/src 

I can ping from the UE by `sudo ip netns exec ue1 ping -i 0.1 10.45.1.1`

However, I cannot ping from core to UE using the following command:

`ping -i 0.1 10.45.1.2`

The route is established correctly: ```
sudo ip ro add 10.45.0.0/16 via 10.53.1.2
RTNETLINK answers: File exists
```
I do have access to internet ping directly. 

I add the tmp log of the UE
[ue_log_20231116.txt](https://github.com/srsran/srsRAN_Project/files/13378282/ue_log_20231116.txt)

From gnb part there are no traces:

```
==== gNodeB started ===
Type <t> to view trace
```

pgawlowicz, 2023-11-16:
You need to also add a default route for the UE:

sudo ip netns exec ue1 ip route add default via 10.45.1.1 dev tun_srsue

raul-parada, 2023-11-16:
I forgot to mention I've done the troubleshooting section, however, still the same issue

pgawlowicz, 2023-11-16:
Did UE connect and got IP?

raul-parada, 2023-11-16:
This output from open5gs: 

```
open5gs_5gc  | 11/16 11:15:28.263: [sbi] INFO: [4b4e72e0-8471-41ee-afa7-6dcb918cdb30] (NF-discover) NF Profile updated (../lib/sbi/nnrf-handler.c:763)
open5gs_5gc  | 11/16 11:15:28.264: [smf] INFO: [Added] Number of SMF-UEs is now 1 (../src/smf/context.c:1012)
open5gs_5gc  | 11/16 11:15:28.264: [smf] INFO: [Added] Number of SMF-Sessions is now 1 (../src/smf/context.c:3108)
open5gs_5gc  | 11/16 11:15:28.272: [smf] INFO: UE SUPI[imsi-001010123456780] DNN[srsapn] IPv4[10.45.1.2] IPv6[] (../src/smf/npcf-handler.c:528)
open5gs_5gc  | 11/16 11:15:28.273: [upf] INFO: [Added] Number of UPF-Sessions is now 1 (../src/upf/context.c:194)
open5gs_5gc  | 11/16 11:15:28.273: [gtp] INFO: gtp_connect() [127.0.0.4]:2152 (../lib/gtp/path.c:60)
open5gs_5gc  | 11/16 11:15:28.273: [upf] INFO: UE F-SEID[UP:0x1 CP:0x1] APN[srsapn] PDN-Type[1] IPv4[10.45.1.2] IPv6[] (../src/upf/context.c:467)
open5gs_5gc  | 11/16 11:15:28.273: [upf] INFO: UE F-SEID[UP:0x1 CP:0x1] APN[srsapn] PDN-Type[1] IPv4[10.45.1.2] IPv6[] (../src/upf/context.c:467)
open5gs_5gc  | 11/16 11:15:28.273: [gtp] INFO: gtp_connect() [10.53.1.2]:2152 (../lib/gtp/path.c:60)
open5gs_5gc  | 11/16 11:15:28.343: [gmm] INFO: [imsi-001010123456780] No GUTI allocated (../src/amf/gmm-sm.c:1242)
open5gs_5gc  | 11/16 11:15:28.480: [gtp] INFO: gtp_connect() [10.53.1.1]:2152 (../lib/gtp/path.c:60)
open5gs_5gc  | 11/16 12:15:28.264: [sbi] INFO: [4b4e72e0-8471-41ee-afa7-6dcb918cdb30] NF expired (../lib/sbi/nf-sm.c:318)
```

pgawlowicz, 2023-11-17:
Any progress with this issue?

pgawlowicz, 2023-11-17:
Could you print routing table from ue1?
```
sudo ip netns exec ue1 route -n
```

and also host:
```
route -n
```
Could you try to ping from open 5gs docker?
```
docker exec -it open5gs_5gc bash 
```
then inside docker:
```
ping 8.8.8.8
```

pgawlowicz, 2023-11-17:
try:
```
echo 1 | sudo tee /proc/sys/net/ipv4/ip_forward 1>/dev/null
sudo iptables -t nat -A POSTROUTING -o <IFNAME> -j MASQUERADE
```

raul-parada, 2023-11-17:
> try:
> 
> ```
> sudo sysctl -w net.ipv4.ip_forward=1
> sudo iptables -t nat -A POSTROUTING -o <IFNAME> -j MASQUERADE
> ```

Doesn't work

```
parada@rparada-NUC7i7BNH:~$ sudo sysctl -w net.ipv4.ip_forward=1
[sudo] password for rparada: 
net.ipv4.ip_forward = 1
rparada@rparada-NUC7i7BNH:~$ sudo iptables -t nat -A POSTROUTING -o wlp58s0 -j MASQUERADE
rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 ping -i 1 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
10 packets transmitted, 0 received, 100% packet loss, time 9213ms
```

pgawlowicz, 2023-11-17:
On host machine, the route to `10.45.0.0` goes over a different interface than the route to `10.53.1.0`, but is should be the same as the command is:
```
sudo ip ro add 10.45.0.0/16 via 10.53.1.2
```

could you try to remove it first and add again with interface:

```
sudo ip ro del 10.45.0.0/16 via 10.53.1.2
sudo ip ro add 10.45.0.0/16 via 10.53.1.2 dev br-dfaa76d5109f
```

and print the host routing again.
```
route -n
```

raul-parada, 2023-11-17:
> On host machine, the route to `10.45.0.0` goes over a different interface than the route to `10.53.1.0`, but is should be the same as the command is:
>
> ```
> sudo ip ro add 10.45.0.0/16 via 10.53.1.2 [...]

```
rparada@rparada-NUC7i7BNH:~$ sudo ip ro add 10.45.0.0/16 via 10.53.1.2
RTNETLINK answers: File exists
rparada@rparada-NUC7i7BNH:~$ sudo ip ro del 10.45.0.0/16 via 10.53.1.2
sudo ip ro add 10.45.0.0/16 via 10.53.1.2 dev br-dfaa76d5109f
RTNETLINK answers: No such process
RTNETLINK answers: File exists
rparada@rparada-NUC7i7BNH:~$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.3.10.1       0.0.0.0         UG    600    0        0 wlp58s0
10.3.10.0       0.0.0.0         255.255.255.0   U     600    0        0 wlp58s0
10.45.0.0       0.0.0.0         255.255.0.0     U     0      0        0 ogstun
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-dfaa76d5109f
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 wlp58s0
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
```

pgawlowicz, 2023-11-17:
chould you run:
```
sudo iptables -t nat -S
```

raul-parada, 2023-11-17:
> chould you run:
> 
> ```
> sudo iptables -t nat -S
> ```

```
rparada@rparada-NUC7i7BNH:~$ sudo iptables -t nat -S
-P PREROUTING ACCEPT
-P INPUT ACCEPT
-P OUTPUT ACCEPT
-P POSTROUTING ACCEPT
-N DOCKER
-A PREROUTING -m addrtype --dst-type LOCAL -j DOCKER
-A OUTPUT ! -d 127.0.0.0/8 -m addrtype --dst-type LOCAL -j DOCKER
-A POSTROUTING -s 172.17.0.0/16 ! -o docker0 -j MASQUERADE
-A POSTROUTING -s 10.53.1.0/24 ! -o br-dfaa76d5109f -j MASQUERADE
-A POSTROUTING -s 10.53.1.2/32 -d 10.53.1.2/32 -p tcp -m tcp --dport 3000 -j MASQUERADE
-A POSTROUTING -s 10.45.0.0/16 ! -o wlp58s0 -j MASQUERADE
-A POSTROUTING -o wlp58s0 -j MASQUERADE
-A DOCKER -i docker0 -j RETURN
-A DOCKER -i br-dfaa76d5109f -j RETURN
-A DOCKER ! -i br-dfaa76d5109f -p tcp -m tcp --dport 3000 -j DNAT --to-destination 10.53.1.2:3000
```

pgawlowicz, 2023-11-17:
could you add the following and check ping:

```
sudo iptables -t nat -A POSTROUTING -s 10.45.0.0/16 ! -o ogstun -j MASQUERADE
```

raul-parada, 2023-11-17:
> `sudo iptables -t nat -A POSTROUTING -s 10.45.0.0/16 ! -o ogstun -j MASQUERADE`

```
rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 ping -i 0.1 10.45.1.1
PING 10.45.1.1 (10.45.1.1) 56(84) bytes of data.
64 bytes from 10.45.1.1: icmp_seq=1 ttl=64 time=83.9 ms
64 bytes from 10.45.1.1: icmp_seq=2 ttl=64 time=70.9 ms
64 bytes from 10.45.1.1: icmp_seq=3 ttl=64 time=42.4 ms
64 bytes from 10.45.1.1: icmp_seq=4 ttl=64 time=70.0 ms
64 bytes from 10.45.1.1: icmp_seq=5 ttl=64 time=54.4 ms
64 bytes from 10.45.1.1: icmp_seq=6 ttl=64 time=93.3 ms
64 bytes from 10.45.1.1: icmp_seq=7 ttl=64 time=64.1 ms
64 bytes from 10.45.1.1: icmp_seq=8 ttl=64 time=102 ms
64 bytes from 10.45.1.1: icmp_seq=9 ttl=64 time=70.1 ms
64 bytes from 10.45.1.1: icmp_seq=10 ttl=64 time=91.8 ms
64 bytes from 10.45.1.1: icmp_seq=11 ttl=64 time=76.8 ms
64 bytes from 10.45.1.1: icmp_seq=12 ttl=64 time=48.1 ms
^C
--- 10.45.1.1 ping statistics ---
12 packets transmitted, 12 received, 0% packet loss, time 1105ms
rtt min/avg/max/mdev = 42.406/72.313/102.012/17.651 ms
rparada@rparada-NUC7i7BNH:~$ ping -i 0.1 10.45.1.2
PING 10.45.1.2 (10.45.1.2) 56(84) bytes of data.
^C
--- 10.45.1.2 ping statistics ---
56 packets transmitted, 0 received, 100% packet loss, time 5772ms

rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 ping -i 1 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
4 packets transmitted, 0 received, 100% packet loss, time 3060ms
```

pgawlowicz, 2023-11-17:
great! thanks, we will update the tutorial.

raul-parada, 2023-11-17:
Still doesn't work. Should I wait for the updated tutorial? thanks

pgawlowicz, 2023-11-17:
ah true, I noticed only the first ping and assumed it was correct.

did you try to setup everything again after the host PC reboot?

raul-parada, 2023-11-17:
No, I'll  reboot and run everything again

raul-parada, 2023-11-17:
Still doesn't work.

```
rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 ping -i 0.1 10.45.1.1
[sudo] password for rparada: 
PING 10.45.1.1 (10.45.1.1) 56(84) bytes of data.
64 bytes from 10.45.1.1: icmp_seq=1 ttl=64 time=107 ms
64 bytes from 10.45.1.1: icmp_seq=2 ttl=64 time=87.4 ms
64 bytes from 10.45.1.1: icmp_seq=3 ttl=64 time=59.0 ms
64 bytes from 10.45.1.1: icmp_seq=4 ttl=64 time=99.7 ms
64 bytes from 10.45.1.1: icmp_seq=5 ttl=64 time=70.1 ms
64 bytes from 10.45.1.1: icmp_seq=6 ttl=64 time=91.5 ms
64 bytes from 10.45.1.1: icmp_seq=7 ttl=64 time=76.9 ms
64 bytes from 10.45.1.1: icmp_seq=8 ttl=64 time=68.5 ms
64 bytes from 10.45.1.1: icmp_seq=9 ttl=64 time=40.8 ms
64 bytes from 10.45.1.1: icmp_seq=10 ttl=64 time=76.5 ms
^C
--- 10.45.1.1 ping statistics ---
11 packets transmitted, 10 received, 9.09091% packet loss, time 1003ms
rtt min/avg/max/mdev = 40.755/77.771/107.288/18.711 ms, pipe 2
rparada@rparada-NUC7i7BNH:~$ ping -i 0.1 10.45.1.2
PING 10.45.1.2 (10.45.1.2) 56(84) bytes of data.
^C
--- 10.45.1.2 ping statistics ---
24 packets transmitted, 0 received, 100% packet loss, time 2392ms

rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 ping -i 1 8.8.8.8
ping: connect: Network is unreachable
rparada@rparada-NUC7i7BNH:~$ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=114 time=15.5 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=114 time=21.8 ms
^C
--- 8.8.8.8 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 15.495/18.651/21.807/3.156 ms
rparada@rparada-NUC7i7BNH:~$ docker exec -it open5gs_5gc bash 
root@b9887e011fa5:/open5gs# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=113 time=15.6 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=113 time=15.5 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=113 time=19.3 ms
^C
--- 8.8.8.8 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2001ms
rtt min/avg/max/mdev = 15.527/16.828/19.329/1.768 ms
```

pgawlowicz, 2023-11-17:
do you have a firewall running? please check and disable (just for testing) if active.

```
sudo ufw status
Status: active
sudo ufw disable
```

raul-parada, 2023-11-17:
Status: inactive

pgawlowicz, 2023-11-17:
do not execute:
```
sudo iptables -t nat -A POSTROUTING -s 10.45.0.0/16 ! -o wlp58s0 -j MASQUERADE
```
but this one:
```
sudo iptables -t nat -A POSTROUTING -o wlp58s0 -j MASQUERADE
```

if you executed the first command then remove the corresponding iptables rule with:
```
sudo iptables -t nat -D POSTROUTING -s 10.45.0.0/16 ! -o wlp58s0 -j MASQUERADE
```

and print the iptables:
```
sudo iptables -t nat -S
```

raul-parada, 2023-11-17:
If I try to delete the first command as you mentioned I got this:

rparada@rparada-NUC7i7BNH:~$ sudo iptables -t nat -D POSTROUTING -s 10.45.0.0/16 ! -o wlp58s0 -j MASQUERADE
iptables: Bad rule (does a matching rule exist in that chain?).

Does it mean it did work?

pgawlowicz, 2023-11-17:
seems it was not there. please print iptables

raul-parada, 2023-11-17:
> seems it was not there. please print iptables

```
rparada@rparada-NUC7i7BNH:~$ sudo iptables -t nat -A POSTROUTING -o wlp58s0 -j MASQUERADE
rparada@rparada-NUC7i7BNH:~$ sudo iptables -t nat -S
-P PREROUTING ACCEPT
-P INPUT ACCEPT
-P OUTPUT ACCEPT
-P POSTROUTING ACCEPT
-N DOCKER
-A PREROUTING -m addrtype --dst-type LOCAL -j DOCKER
-A OUTPUT ! -d 127.0.0.0/8 -m addrtype --dst-type LOCAL -j DOCKER
-A POSTROUTING -s 172.17.0.0/16 ! -o docker0 -j MASQUERADE
-A POSTROUTING -s 10.53.1.0/24 ! -o br-dfaa76d5109f -j MASQUERADE
-A POSTROUTING -s 10.53.1.2/32 -d 10.53.1.2/32 -p tcp -m tcp --dport 3000 -j MASQUERADE
-A POSTROUTING -o wlp58s0 -j MASQUERADE
-A DOCKER -i docker0 -j RETURN
-A DOCKER -i br-dfaa76d5109f -j RETURN
-A DOCKER ! -i br-dfaa76d5109f -p tcp -m tcp --dport 3000 -j DNAT --to-destination 10.53.1.2:3000

```

raul-parada, 2023-11-17:
> and routing table on the host pc and ue1

could you provide me with the specific commands?

pgawlowicz, 2023-11-17:
see above

raul-parada, 2023-11-17:
> see above

```
rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
10.45.1.0       0.0.0.0         255.255.255.0   U     0      0        0 tun_srsue
rparada@rparada-NUC7i7BNH:~$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.3.10.1       0.0.0.0         UG    600    0        0 wlp58s0
10.3.10.0       0.0.0.0         255.255.255.0   U     600    0        0 wlp58s0
10.45.0.0       0.0.0.0         255.255.0.0     U     0      0        0 ogstun
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-dfaa76d5109f
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 wlp58s0
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0

```

pgawlowicz, 2023-11-17:
the ue1 does not have the default route, please addd

sudo ip netns exec ue1 ip route add default via 10.45.1.1 dev tun_srsue

raul-parada, 2023-11-17:
> the ue1 does not have the default route, please add
> 
> sudo ip netns exec ue1 ip route add default via 10.45.1.1 dev tun_srsue
> 
> and print routing table again.

IP tables:

```
rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 ip route add default via 10.45.1.1 dev tun_srsue
rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.45.1.1       0.0.0.0         UG    0      0        0 tun_srsue
10.45.1.0       0.0.0.0         255.255.255.0   U     0      0        0 tun_srsue
rparada@rparada-NUC7i7BNH:~$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.3.10.1       0.0.0.0         UG    600    0        0 wlp58s0
10.3.10.0       0.0.0.0         255.255.255.0   U     600    0        0 wlp58s0
10.45.0.0       0.0.0.0         255.255.0.0     U     0      0        0 ogstun
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-dfaa76d5109f
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 wlp58s0
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
rparada@rparada-NUC7i7BNH:~$ sudo iptables -t nat -S
-P PREROUTING ACCEPT
-P INPUT ACCEPT
-P OUTPUT ACCEPT
-P POSTROUTING ACCEPT
-N DOCKER
-A PREROUTING -m addrtype --dst-type LOCAL -j DOCKER
-A OUTPUT ! -d 127.0.0.0/8 -m addrtype --dst-type LOCAL -j DOCKER
-A POSTROUTING -s 172.17.0.0/16 ! -o docker0 -j MASQUERADE
-A POSTROUTING -s 10.53.1.0/24 ! -o br-dfaa76d5109f -j MASQUERADE
-A POSTROUTING -s 10.53.1.2/32 -d 10.53.1.2/32 -p tcp -m tcp --dport 3000 -j MASQUERADE
-A POSTROUTING -o wlp58s0 -j MASQUERADE
-A DOCKER -i docker0 -j RETURN
-A DOCKER -i br-dfaa76d5109f -j RETURN
-A DOCKER ! -i br-dfaa76d5109f -p tcp -m tcp --dport 3000 -j DNAT --to-destination 10.53.1.2:3000
```

PINGs:

```
rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 ping -i 0.1 10.45.1.1
[sudo] password for rparada: 
PING 10.45.1.1 (10.45.1.1) 56(84) bytes of data.
64 bytes from 10.45.1.1: icmp_seq=1 ttl=64 time=71.7 ms
64 bytes from 10.45.1.1: icmp_seq=2 ttl=64 time=40.4 ms
64 bytes from 10.45.1.1: icmp_seq=3 ttl=64 time=66.0 ms
64 bytes from 10.45.1.1: icmp_seq=4 ttl=64 time=44.1 ms
64 bytes from 10.45.1.1: icmp_seq=5 ttl=64 time=88.1 ms
64 bytes from 10.45.1.1: icmp_seq=6 ttl=64 time=58.2 ms
64 bytes from 10.45.1.1: icmp_seq=7 ttl=64 time=86.3 ms
64 bytes from 10.45.1.1: icmp_seq=8 ttl=64 time=69.3 ms
64 bytes from 10.45.1.1: icmp_seq=9 ttl=64 time=95.3 ms
64 bytes from 10.45.1.1: icmp_seq=10 ttl=64 time=74.7 ms
64 bytes from 10.45.1.1: icmp_seq=11 ttl=64 time=50.3 ms
64 bytes from 10.45.1.1: icmp_seq=12 ttl=64 time=75.8 ms
^C
--- 10.45.1.1 ping statistics ---
12 packets transmitted, 12 received, 0% packet loss, time 1104ms
rtt min/avg/max/mdev = 40.411/68.358/95.326/16.718 ms
rparada@rparada-NUC7i7BNH:~$ ping -i 0.1 10.45.1.2
PING 10.45.1.2 (10.45.1.2) 56(84) bytes of data.
^C
--- 10.45.1.2 ping statistics ---
16 packets transmitted, 0 received, 100% packet loss, time 1571ms

rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 ping -i 1 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
2 packets transmitted, 0 received, 100% packet loss, time 1012ms
root@b9887e011fa5:/open5gs# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=113 time=16.6 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=113 time=15.3 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=113 time=16.5 ms
^C
--- 8.8.8.8 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2002ms
rtt min/avg/max/mdev = 15.317/16.113/16.562/0.564 ms
```

pgawlowicz, 2023-11-17:
do you have Open5GS running on the host machine that was installed manually (i.e., not running with our docker)?

I do not like those two entries in the host routing table:
```
10.45.0.0       0.0.0.0         255.255.0.0     U     0      0        0 ogstun
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-dfaa76d5109f
```
they should use the same interface.

could you run:
```
ifconfig ogstun
ifconfig br-dfaa76d5109f
```
note that you might have to install some package.

raul-parada, 2023-11-17:
Yes, I'm using your docker to run Open5GS, I do also have it installed manually (Open5GS v2.6.6). Here the output of the ifconfig:

```
rparada@rparada-NUC7i7BNH:~$ ifconfig ogstun
ogstun: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1400
        inet 10.45.0.1  netmask 255.255.0.0  destination 10.45.0.1
        inet6 2001:db8:cafe::1  prefixlen 48  scopeid 0x0<global>
        inet6 fe80::8ca5:9b99:c035:edcb  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 50  bytes 3920 (3.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

rparada@rparada-NUC7i7BNH:~$ ifconfig br-dfaa76d5109f
br-dfaa76d5109f: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.53.1.1  netmask 255.255.255.0  broadcast 10.53.1.255
        inet6 fe80::42:9dff:fe18:90d6  prefixlen 64  scopeid 0x20<link>
        ether 02:42:9d:18:90:d6  txqueuelen 0  (Ethernet)
        RX packets 620  bytes 54264 (54.2 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 661  bytes 68487 (68.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

pgawlowicz, 2023-11-17:
could you stop the Open5GS that was installed manually:
```
sudo systemctl stop open5gs-mmed
sudo systemctl stop open5gs-sgwcd
sudo systemctl stop open5gs-smfd
sudo systemctl stop open5gs-amfd
sudo systemctl stop open5gs-sgwud
sudo systemctl stop open5gs-upfd
sudo systemctl stop open5gs-hssd
sudo systemctl stop open5gs-pcrfd
sudo systemctl stop open5gs-nrfd
sudo systemctl stop open5gs-scpd
sudo systemctl stop open5gs-ausfd
sudo systemctl stop open5gs-udmd
sudo systemctl stop open5gs-pcfd
sudo systemctl stop open5gs-nssfd
sudo systemctl stop open5gs-bsfd
sudo systemctl stop open5gs-udrd
sudo systemctl stop open5gs-webui
```
turn off the ogstun interface:
```
sudo ifconfig ogstun 0.0.0.0 down
```

print host routing table:
```
route -n
```

if an entry for `10.45.0.0` is not present then add:
```
sudo ip ro add 10.45.0.0/16 via 10.53.1.2
```

print host routing table:
```
route -n
```

try ping from ue1

raul-parada, 2023-11-17:
Now, it works. 

```
rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 ping -i 0.1 10.45.1.1
PING 10.45.1.1 (10.45.1.1) 56(84) bytes of data.
64 bytes from 10.45.1.1: icmp_seq=1 ttl=64 time=78.2 ms
64 bytes from 10.45.1.1: icmp_seq=2 ttl=64 time=48.1 ms
64 bytes from 10.45.1.1: icmp_seq=3 ttl=64 time=77.9 ms
64 bytes from 10.45.1.1: icmp_seq=4 ttl=64 time=66.9 ms
64 bytes from 10.45.1.1: icmp_seq=5 ttl=64 time=102 ms
64 bytes from 10.45.1.1: icmp_seq=6 ttl=64 time=90.4 ms
64 bytes from 10.45.1.1: icmp_seq=7 ttl=64 time=67.2 ms
64 bytes from 10.45.1.1: icmp_seq=8 ttl=64 time=45.5 ms
64 bytes from 10.45.1.1: icmp_seq=9 ttl=64 time=91.5 ms
64 bytes from 10.45.1.1: icmp_seq=10 ttl=64 time=65.8 ms
64 bytes from 10.45.1.1: icmp_seq=11 ttl=64 time=106 ms
64 bytes from 10.45.1.1: icmp_seq=12 ttl=64 time=73.3 ms
64 bytes from 10.45.1.1: icmp_seq=13 ttl=64 time=43.3 ms
^C
--- 10.45.1.1 ping statistics ---
14 packets transmitted, 13 received, 7.14286% packet loss, time 1309ms
rtt min/avg/max/mdev = 43.296/73.557/106.177/19.656 ms, pipe 2
rparada@rparada-NUC7i7BNH:~$ ping -i 0.1 10.45.1.2
PING 10.45.1.2 (10.45.1.2) 56(84) bytes of data.
64 bytes from 10.45.1.2: icmp_seq=1 ttl=63 time=69.6 ms
64 bytes from 10.45.1.2: icmp_seq=2 ttl=63 time=105 ms
64 bytes from 10.45.1.2: icmp_seq=3 ttl=63 time=74.1 ms
64 bytes from 10.45.1.2: icmp_seq=4 ttl=63 time=114 ms
64 bytes from 10.45.1.2: icmp_seq=5 ttl=63 time=45.9 ms
64 bytes from 10.45.1.2: icmp_seq=6 ttl=63 time=121 ms
64 bytes from 10.45.1.2: icmp_seq=7 ttl=63 time=45.2 ms
64 bytes from 10.45.1.2: icmp_seq=8 ttl=63 time=124 ms
64 bytes from 10.45.1.2: icmp_seq=9 ttl=63 time=52.9 ms
64 bytes from 10.45.1.2: icmp_seq=10 ttl=63 time=71.8 ms
^C
--- 10.45.1.2 ping statistics ---
11 packets transmitted, 10 received, 9.09091% packet loss, time 1018ms
rtt min/avg/max/mdev = 45.216/82.372/124.073/29.452 ms, pipe 2
rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 ping -i 1 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=112 time=108 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=112 time=115 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=112 time=131 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=112 time=120 ms
^C
--- 8.8.8.8 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3003ms
rtt min/avg/max/mdev = 107.986/118.448/130.620/8.192 ms
```
Should I remove the installed Open5GS?

pgawlowicz, 2023-11-17:
It is not needed, so you can remove it. 
The problem was routing was sending packets to the Open5GS that was manually installed.

closed 2023-11-17 by pgawlowicz

chelimi, 2024-05-15:
Good evening sir,
I'm getting error while running the command
waxwing@waxwing:~$ sudo ip netns exec ue1 ip route add default via 10.45.0.1 dev tun_srsue
Error: Nexthop has invalid gateway.

waxwing@waxwing:~$ sudo srsenb
[sudo] password for waxwing: 
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /root/.config/srsran/enb.conf
Reading configuration file /root/.config/srsran/enb.conf...
Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
Couldn't open rr.conf, trying /root/.config/srsran/rr.conf
Couldn't open rb.conf, trying /root/.config/srsran/rb.conf

Built in Release mode using commit ce8a3cae1 on branch HEAD.

Opening 1 channels in RF device=zmq with args=fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=23.04e6
Supported RF device list: UHD zmq file
CHx base_srate=23.04e6
CHx id=enb
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2000
CH0 fail_on_disconnect=true
NG connection successful

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Setting manual TX/RX offset to 100 samples
Setting frequency: DL=1842.5 Mhz, DL_SSB=1842.05 Mhz (SSB-ARFCN=368410), UL=1747.5 MHz for cc_idx=0 nof_prb=52

● open5gs-amfd.service - Open5GS AMF Daemon
     Loaded: loaded (/lib/systemd/system/open5gs-amfd.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2024-05-09 11:32:06 IST; 6 days ago
   Main PID: 3239 (open5gs-amfd)
      Tasks: 2 (limit: 18951)
     Memory: 113.7M
        CPU: 11.923s
     CGroup: /system.slice/open5gs-amfd.service
             └─3239 /usr/bin/open5gs-amfd -c /etc/open5gs/amf.yaml

May 15 16:48:18 waxwing open5gs-amfd[3239]: 05/15 16:48:18.246: [amf] INFO: gNB-N2 accepted[192.168.134.187]:38394 in ng-path module (../src/amf/ngap-sctp.c:113)
May 15 16:48:18 waxwing open5gs-amfd[3239]: 05/15 16:48:18.247: [amf] INFO: gNB-N2 accepted[192.168.134.187] in master_sm module (../src/amf/amf-sm.c:741)
May 15 16:48:18 waxwing open5gs-amfd[3239]: 05/15 16:48:18.253: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1231)
May 15 16:48:18 waxwing open5gs-amfd[3239]: 05/15 16:48:18.253: [amf] INFO: gNB-N2[192.168.134.187] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:780)
May 15 16:48:55 waxwing open5gs-amfd[3239]: 05/15 16:48:55.555: [amf] INFO: gNB-N2[192.168.134.187] connection refused!!! (../src/amf/amf-sm.c:793)
May 15 16:48:55 waxwing open5gs-amfd[3239]: 05/15 16:48:55.555: [amf] INFO: [Removed] Number of gNBs is now 0 (../src/amf/context.c:1258)
May 15 16:54:09 waxwing open5gs-amfd[3239]: 05/15 16:54:09.670: [amf] INFO: gNB-N2 accepted[192.168.134.187]:59629 in ng-path module (../src/amf/ngap-sctp.c:113)
May 15 16:54:09 waxwing open5gs-amfd[3239]: 05/15 16:54:09.670: [amf] INFO: gNB-N2 accepted[192.168.134.187] in master_sm module (../src/amf/amf-sm.c:741)
May 15 16:54:09 waxwing open5gs-amfd[3239]: 05/15 16:54:09.677: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1231)
May 15 16:54:09 waxwing open5gs-amfd[3239]: 05/15 16:54:09.677: [amf] INFO: gNB-N2[192.168.134.187] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:780)

waxwing@waxwing:~$ sudo srsue
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Couldn't open , trying /root/.config/srsran/ue.conf
Reading configuration file /root/.config/srsran/ue.conf...

Built in Release mode using commit ce8a3cae1 on branch HEAD.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6
Supported RF device list: UHD zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Waiting PHY to initialize ... done!
Attaching UE...

waxwing@waxwing:~$ sudo ip ro add 10.45.0.0/16 via 10.53.1.2
RTNETLINK answers: File exists
waxwing@waxwing:~$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.134.1   0.0.0.0         UG    100    0        0 enp2s0
10.45.0.0       0.0.0.0         255.255.0.0     U     0      0        0 ogstun
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-c38cb30b6938
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 enp2s0
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-4460feeebb98
192.168.134.0   0.0.0.0         255.255.254.0   U     100    0        0 enp2s0
waxwing@waxwing:~$ sudo ip netns exec ue1 ip route add default via 10.45.0.1 dev tun_srsue
Error: Nexthop has invalid gateway.

please help me to solve the error.

pgawlowicz, 2024-05-15:
please follow this tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup

chelimi, 2024-05-16:
waxwing@waxwing:~/srsran/srsRAN_Project/build/configs$ sudo gnb -c gnb_zmq.yaml 
[sudo] password for waxwing: 

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in executor blocking mode.

--== srsRAN gNB (commit 0b2702cca) ==--

Connecting to AMF on 10.53.1.2:38412
Failed to connect SCTP socket to 10.53.1.2:38412. error="Connection timed out" timeout=1385ms
srsGNB ERROR: Failed to create SCTP gateway.

sir we getting same error after following that document also

pgawlowicz, 2024-05-16:
did you start the dockerized open5gs on the same host pc as gnb?

chelimi, 2024-05-16:
waxwing@waxwing:~/srsran/srsRAN_Project/docker$ sudo docker compose up --build 5gc
[sudo] password for waxwing: 
[+] Building 4.4s (16/16) FINISHED                               docker:default
 => [5gc internal] load build definition from Dockerfile                   0.6s
 => => transferring dockerfile: 3.11kB                                     0.1s
 => [5gc internal] load .dockerignore                                      0.8s
 => => transferring context: 2B                                            0.0s
 => [5gc internal] load metadata for docker.io/library/ubuntu:22.04        2.2s
 => [5gc base 1/8] FROM docker.io/library/ubuntu:22.04@sha256:a6d2b38300c  0.0s
 => [5gc internal] load build context                                      0.2s
 => => transferring context: 142B                                          0.0s
 => CACHED [5gc base 2/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 3/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 4/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 5/8] RUN echo v2.6.1 > ./open5gsversion               0.0s
 => CACHED [5gc base 6/8] RUN git clone --depth 1 --branch $(cat ./open5g  0.0s
 => CACHED [5gc base 7/8] RUN curl -fsSL https://deb.nodesource.com/setup  0.0s
 => CACHED [5gc base 8/8] RUN python3 -m pip install pymongo click pyrout  0.0s
 => CACHED [5gc open5gs 1/3] WORKDIR /open5gs                              0.0s
 => CACHED [5gc open5gs 2/3] COPY open5gs-5gc.yml open5gs-5gc.yml.in       0.0s
 => CACHED [5gc open5gs 3/3] COPY open5gs_entrypoint.sh add_users.py setu  0.0s
 => [5gc] exporting to image                                               0.1s
 => => exporting layers                                                    0.0s
 => => writing image sha256:812e7c17097e3b1db38fac053b4be0c2e4fc8f82d1495  0.1s
 => => naming to docker.io/library/docker-5gc                              0.1s
[+] Running 1/0
 ✔ Container open5gs_5gc  Created                                          0.0s 
Attaching to open5gs_5gc
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | {"t":{"$date":"2024-05-16T11:04:42.225Z"},"s":"I",  "c":"CONTROL",  "id":20697,   "ctx":"-","msg":"Renamed existing log file","attr":{"oldLogPath":"/tmp/mongodb.log","newLogPath":"/tmp/mongodb.log.2024-05-16T11-04-41"}}
open5gs_5gc  | 
open5gs_5gc  | > open5gs@2.6.1 dev
open5gs_5gc  | > node server/index.js
open5gs_5gc  | 
open5gs_5gc  | 
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | Connection to 127.0.0.1 27017 port [tcp/*] succeeded!
open5gs_5gc  | SUBSCRIBER_DB=001010123456780,00112233445566778899aabbccddeeff,opc,63bfa50ee6523365ff14c1f45f88737d,8000,9,10.45.1.2
[... 120 lines cut ...]
open5gs_5gc  | 05/16 11:05:11.139: [gtp] INFO: gtp_server() [127.0.0.4]:2152 (../lib/gtp/path.c:30)
open5gs_5gc  | 05/16 11:05:11.139: [pfcp] INFO: pfcp_server() [127.0.0.4]:8805 (../lib/pfcp/path.c:30)
open5gs_5gc  | 05/16 11:05:11.139: [pfcp] INFO: ogs_pfcp_connect() [127.0.0.7]:8805 (../lib/pfcp/path.c:61)
open5gs_5gc  | 05/16 11:05:11.139: [upf] INFO: PFCP associated [127.0.0.4]:8805 (../src/upf/pfcp-sm.c:173)
open5gs_5gc  | 05/16 11:05:11.168: [sbi] INFO: NF Service [nsmf-pdusession] (../lib/sbi/context.c:1436)
open5gs_5gc  | 05/16 11:05:11.168: [sbi] INFO: nghttp2_server() [http://127.0.0.4]:7777 (../lib/sbi/nghttp2-server.c:395)
open5gs_5gc  | 05/16 11:05:11.168: [app] INFO: SMF initialize...done (../src/smf/app.c:31)
[... cut ...]
open5gs_5gc  | 05/16 11:05:13.220: [upf] WARNING: PFCP[RSP] has already been associated [127.0.0.4]:8805 (../src/upf/pfcp-sm.c:213)
open5gs_5gc  | > Using external babel configuration
open5gs_5gc  | > Location: "/open5gs/webui/.babelrc"
open5gs_5gc  |  DONE  Compiled successfully in 2013ms11:05:21 AM
open5gs_5gc  | 
open5gs_5gc  | 
open5gs_5gc  | Mongoose: accounts.count({}, {})
open5gs_5gc  | > Ready on http://560ba178d2c9:3000
open5gs_5gc  | Mongoose: subscribers.createIndex({ imsi: 1 }, { unique: true, background: true })
open5gs_5gc  | Mongoose: accounts.createIndex({ username: 1 }, { unique: true, background: true })

chelimi, 2024-05-16:
i should do that proces again sirr

chelimi, 2024-05-16:
waxwing@waxwing:~/srsran/srsRAN_Project/docker$ sudo docker compose up --build 5gc
[+] Building 3.5s (16/16) FINISHED                               docker:default
 => [5gc internal] load build definition from Dockerfile                   0.2s
 => => transferring dockerfile: 3.11kB                                     0.0s
 => [5gc internal] load .dockerignore                                      0.4s
 => => transferring context: 2B                                            0.0s
 => [5gc internal] load metadata for docker.io/library/ubuntu:22.04        2.0s
 => [5gc base 1/8] FROM docker.io/library/ubuntu:22.04@sha256:a6d2b38300c  0.0s
 => [5gc internal] load build context                                      0.2s
 => => transferring context: 142B                                          0.0s
 => CACHED [5gc base 2/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 3/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 4/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 5/8] RUN echo v2.6.1 > ./open5gsversion               0.0s
 => CACHED [5gc base 6/8] RUN git clone --depth 1 --branch $(cat ./open5g  0.0s
 => CACHED [5gc base 7/8] RUN curl -fsSL https://deb.nodesource.com/setup  0.0s
 => CACHED [5gc base 8/8] RUN python3 -m pip install pymongo click pyrout  0.0s
 => CACHED [5gc open5gs 1/3] WORKDIR /open5gs                              0.0s
 => CACHED [5gc open5gs 2/3] COPY open5gs-5gc.yml open5gs-5gc.yml.in       0.0s
 => CACHED [5gc open5gs 3/3] COPY open5gs_entrypoint.sh add_users.py setu  0.0s
 => [5gc] exporting to image                                               0.1s
 => => exporting layers                                                    0.0s
 => => writing image sha256:812e7c17097e3b1db38fac053b4be0c2e4fc8f82d1495  0.0s
 => => naming to docker.io/library/docker-5gc                              0.0s
[+] Running 1/0
 ✔ Container open5gs_5gc  Created                                          0.0s 
Attaching to open5gs_5gc
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | {"t":{"$date":"2024-05-16T11:17:30.250Z"},"s":"I",  "c":"CONTROL",  "id":20697,   "ctx":"-","msg":"Renamed existing log file","attr":{"oldLogPath":"/tmp/mongodb.log","newLogPath":"/tmp/mongodb.log.2024-05-16T11-17-30"}}
open5gs_5gc  | 
open5gs_5gc  | > open5gs@2.6.1 dev
open5gs_5gc  | > node server/index.js
open5gs_5gc  | 
open5gs_5gc  | 
open5gs_5gc  | > Using external babel configuration
open5gs_5gc  | > Location: "/open5gs/webui/.babelrc"
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  |  DONE  Compiled successfully in 1017ms11:17:32 AM
open5gs_5gc  | 
open5gs_5gc  | 
open5gs_5gc  | Connection to 127.0.0.1 27017 port [tcp/*] succeeded!
open5gs_5gc  | Mongoose: accounts.count({}, {})
open5gs_5gc  | > Ready on http://560ba178d2c9:3000
open5gs_5gc  | Mongoose: subscribers.createIndex({ imsi: 1 }, { unique: true, background: true })
open5gs_5gc  | Mongoose: accounts.createIndex({ username: 1 }, { unique: true, background: true })
open5gs_5gc  | SUBSCRIBER_DB=001010123456780,00112233445566778899aabbccddeeff,opc,63bfa50ee6523365ff14c1f45f88737d,8000,9,10.45.1.2
open5gs_5gc  | Reading subscriber data from cmd.
open5gs_5gc  | UE (IMSI=001010123456780) already exists, updating it.
open5gs_5gc  | Attempting to update IMSI 001010123456780
open5gs_5gc  | UpdateResult({'n': 1, 'nModified': 0, 'ok': 1.0, 'updatedExisting': True}, acknowledged=True)
open5gs_5gc  | True
open5gs_5gc  | Open5GS daemon v2.6.1
open5gs_5gc  | 
open5gs_5gc  | 05/16 11:17:34.473: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:126)
open5gs_5gc  | 
open5gs_5gc  | Open5GS daemon v2.6.1
open5gs_5gc  | 
open5gs_5gc  | 05/16 11:17:34.480: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:126)
open5gs_5gc  | 
open5gs_5gc  | 05/16 11:17:34.483: [sbi] INFO: nghttp2_server() [http://127.0.0.10]:7777 (../lib/sbi/nghttp2-server.c:395)
open5gs_5gc  | 05/16 11:17:34.483: [app] INFO: NRF initialize...done (../src/nrf/app.c:31)
open5gs_5gc  | Open5GS daemon v2.6.1
open5gs_5gc  | 
open5gs_5gc  | 05/16 11:17:34.531: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:126)
open5gs_5gc  | 
open5gs_5gc  | 05/16 11:17:34.533: [sbi] INFO: nghttp2_server() [http://127.0.1.10]:7777 (../lib/sbi/nghttp2-server.c:395)
open5gs_5gc  | 05/16 11:17:34.534: [app] INFO: SCP initialize...done (../src/scp/app.c:31)
open5gs_5gc  | 05/16 11:17:34.534: [nrf] INFO: [e4f804ae-1375-41ef-9fc9-959ea0eefcd4] NF registered [Heartbeat:10s] (../src/nrf/nf-sm.c:190)
[... 113 lines cut ...]
open5gs_5gc  | 05/16 11:17:34.936: [app] INFO: BSF initialize...done (../src/bsf/app.c:31)
open5gs_5gc  | 05/16 11:17:34.937: [nrf] INFO: [e5357866-1375-41ef-bdbf-532c3bb75b9b] NF registered [Heartbeat:10s] (../src/nrf/nf-sm.c:190)
open5gs_5gc  | 05/16 11:17:34.937: [sbi] INFO: [e5357866-1375-41ef-bdbf-532c3bb75b9b] (NRF-notify) NF registered (../lib/sbi/nnrf-handler.c:632)
open5gs_5gc  | 05/16 11:17:34.937: [sbi] INFO: [e5357866-1375-41ef-bdbf-532c3bb75b9b] (NRF-notify) NF Profile updated (../lib/sbi/nnrf-handler.c:642)
open5gs_5gc  | 05/16 11:17:34.938: [sbi] INFO: [e5357866-1375-41ef-bdbf-532c3bb75b9b] (NRF-notify) NF registered (../lib/sbi/nnrf-handler.c:632)
[... cut ...]
open5gs_5gc  | 05/16 11:17:34.995: [sbi] INFO: [e53e11d8-1375-41ef-ba2f-67e63b8bc9be] (NRF-notify) NF Profile updated (../lib/sbi/nnrf-handler.c:642)
open5gs_5gc  | 05/16 11:17:34.995: [sbi] INFO: [e53e11d8-1375-41ef-ba2f-67e63b8bc9be] NF registered [Heartbeat:10s] (../lib/sbi/nf-sm.c:214)
open5gs_5gc  | 05/16 11:17:37.095: [smf] WARNING: PFCP[REQ] has already been associated [127.0.0.7]:8805 (../src/smf/pfcp-sm.c:217)
open5gs_5gc  | 05/16 11:17:37.095: [upf] WARNING: PFCP[RSP] has already been associated [127.0.0.4]:8805 (../src/upf/pfcp-sm.c:213)

chelimi, 2024-05-16:
how to build that again

pgawlowicz, 2024-05-16:
remove the container and the image and run the same command again

chelimi, 2024-05-16:
sorry for inconvenience sir.. please let me know the what container should i remove i didn't get you sir.. please me know the what i should i change

pgawlowicz, 2024-05-16:
```
# list docker containers:
docker container list -a
# remove the open5gs container using its ID
docker container rm CONTAINER_ID

# same with image
docker image list
# remove the open5gs image using its ID
docker image rm IMAGE_ID
```

chelimi, 2024-05-16:
waxwing@waxwing:~/srsran/srsRAN_Project/docker$ sudo docker container list -a
CONTAINER ID   IMAGE        COMMAND                  CREATED       STATUS                       PORTS                                       NAMES
48819c4ef3f3   srsran/gnb   "gnb -c /gnb_config.…"   5 hours ago   Created                                                                  srsran_gnb
560ba178d2c9   docker-5gc   "./open5gs_entrypoin…"   6 hours ago   Exited (255) 8 minutes ago   0.0.0.0:3000->3000/tcp, :::3000->3000/tcp   open5gs_5gc
waxwing@waxwing:~/srsran/srsRAN_Project/docker$ sudo docker container rm CONTAINER ID
Error response from daemon: No such container: CONTAINER
Error response from daemon: No such container: ID
waxwing@waxwing:~/srsran/srsRAN_Project/docker$ sudo docker container rm CONTAINER_ID
Error response from daemon: No such container: CONTAINER_ID
WE  are getting this error sir

pgawlowicz, 2024-05-16:
sudo docker container rm 48819c4ef3f3

chelimi, 2024-05-16:
done sir.. i removed that

chelimi, 2024-05-16:
done sir.. i removed that.....should i run again this command  sir 
sudo docker compose up --build 5gc

chelimi, 2024-05-16:
waxwing@waxwing:~/srsran/srsRAN_Project/docker$ sudo docker compose up --build 5gc
[+] Building 2.4s (16/16) FINISHED                                                                                                                                                           docker:default
 => [5gc internal] load build definition from Dockerfile                                                                                                                                               0.3s
 => => transferring dockerfile: 3.11kB                                                                                                                                                                 0.0s
 => [5gc internal] load .dockerignore                                                                                                                                                                  0.2s
 => => transferring context: 2B                                                                                                                                                                        0.0s
 => [5gc internal] load metadata for docker.io/library/ubuntu:22.04                                                                                                                                    1.0s
 => [5gc internal] load build context                                                                                                                                                                  0.2s
 => => transferring context: 142B                                                                                                                                                                      0.0s
 => [5gc base 1/8] FROM docker.io/library/ubuntu:22.04@sha256:a6d2b38300ce017add71440577d5b0a90460d0e57fd7aec21dd0d1b0761bbfb2                                                                         0.0s
 => CACHED [5gc base 2/8] RUN DEBIAN_FRONTEND=noninteractive apt-get update     && apt install -y software-properties-common     && rm -rf /var/lib/apt/lists/*                                        0.0s
 => CACHED [5gc base 3/8] RUN DEBIAN_FRONTEND=noninteractive apt-get update     && apt-get install -y     python3-pip     python3-setuptools     python3-wheel     ninja-build     build-essential     0.0s
 => CACHED [5gc base 4/8] RUN DEBIAN_FRONTEND=noninteractive apt-get update && apt-get install -y --no-install-recommends wget gnupg     && wget -qO - https://www.mongodb.org/static/pgp/server-6.0.  0.0s
 => CACHED [5gc base 5/8] RUN echo v2.6.1 > ./open5gsversion                                                                                                                                           0.0s
 => CACHED [5gc base 6/8] RUN git clone --depth 1 --branch $(cat ./open5gsversion) https://github.com/open5gs/open5gs open5gs    && cd open5gs     && meson build --prefix=`pwd`/install     && ninja  0.0s
 => CACHED [5gc base 7/8] RUN curl -fsSL https://deb.nodesource.com/setup_20.x | bash -     && apt-get install -y nodejs     && cd open5gs/webui     && npm ci --no-optional                           0.0s
 => CACHED [5gc base 8/8] RUN python3 -m pip install pymongo click pyroute2 ipaddress python-iptables                                                                                                  0.0s
 => CACHED [5gc open5gs 1/3] WORKDIR /open5gs                                                                                                                                                          0.0s
 => CACHED [5gc open5gs 2/3] COPY open5gs-5gc.yml open5gs-5gc.yml.in                                                                                                                                   0.0s
 => CACHED [5gc open5gs 3/3] COPY open5gs_entrypoint.sh add_users.py setup_tun.py subscriber_db.cs[v] ./                                                                                               0.0s
 => [5gc] exporting to image                                                                                                                                                                           0.1s
 => => exporting layers                                                                                                                                                                                0.0s
 => => writing image sha256:812e7c17097e3b1db38fac053b4be0c2e4fc8f82d1495e9483fb36a1ff3c158c                                                                                                           0.0s
 => => naming to docker.io/library/docker-5gc                                                                                                                                                          0.0s
[+] Running 1/0
 ✔ Container open5gs_5gc  Created                                                                                                                                                                      0.0s 
Attaching to open5gs_5gc
open5gs_5gc  | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc  | waiting for mongodb
open5gs_5gc  | {"t":{"$date":"2024-05-16T12:35:54.277Z"},"s":"I",  "c":"CONTROL",  "id":20697,   "ctx":"-","msg":"Renamed existing log file","attr":{"oldLogPath":"/tmp/mongodb.log","newLogPath":"/tmp/mongodb.log.2024-05-16T12-35-54"}}
open5gs_5gc  | 
open5gs_5gc  | > open5gs@2.6.1 dev
open5gs_5gc  | > node server/index.js
open5gs_5gc  | 
open5gs_5gc  | 
[... cut ...]
open5gs_5gc  | 05/16 12:35:59.013: [sbi] INFO: [d90e1a9c-1380-41ef-a91f-d3af74137705] (NRF-notify) NF Profile updated (../lib/sbi/nnrf-handler.c:642)
open5gs_5gc  | 05/16 12:36:01.109: [smf] WARNING: PFCP[REQ] has already been associated [127.0.0.7]:8805 (../src/smf/pfcp-sm.c:217)
open5gs_5gc  | 05/16 12:36:01.109: [upf] WARNING: PFCP[RSP] has already been associated [127.0.0.4]:8805 (../src/upf/pfcp-sm.c:213)

we got the connection reufused is successed but aslo we are getting same error in gnb

pgawlowicz, 2024-05-16:
could you share your gnb config?

chelimi, 2024-05-16:
# This configuration file example shows how to configure the srsRAN Project gNB to allow srsUE to connect to it. 
# This specific example uses ZMQ in place of a USRP for the RF-frontend, and creates an FDD cell with 10 MHz bandwidth. 
# To run the srsRAN Project gNB with this config, use the following command: 
#   sudo ./gnb -c gnb_zmq.yaml

amf:
  addr: 127.0.0.5        # The address or hostname of the AMF.
  bind_addr: 127.0.0.5         # A local IP that the gNB binds to for traffic from the AMF.

ru_sdr:
  device_driver: zmq           # The RF driver name.
  device_args: tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=23.04e6            # Optionally pass arguments to the selected RF driver.
  srate: 23.04                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 20         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "00101"                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 12            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.

pgawlowicz, 2024-05-16:
Please follow the shared tutorial.
You have to install zmq and compile the project with ZMQ support.

chelimi, 2024-05-20:
Attaching to open5gs_5gc
Error response from daemon: driver failed programming external connectivity on endpoint open5gs_5gc (0e7b784417e89be92767f5694d59ffa3a38bb97403b5d6b41fb571833ba1424c): Error starting userland proxy: listen tcp4 0.0.0.0:9999: bind: address already in use

chelimi, 2024-05-20:
waxwing@waxwing:~/srsran/srsRAN_Project/docker$ sudo docker-compose up --build 5gc
[+] Building 2.5s (16/16) FINISHED                               docker:default
 => [5gc internal] load build definition from Dockerfile                   0.2s
 => => transferring dockerfile: 3.11kB                                     0.0s
 => [5gc internal] load .dockerignore                                      0.3s
 => => transferring context: 2B                                            0.0s
 => [5gc internal] load metadata for docker.io/library/ubuntu:22.04        1.0s
 => [5gc base 1/8] FROM docker.io/library/ubuntu:22.04@sha256:a6d2b38300c  0.0s
 => [5gc internal] load build context                                      0.2s
 => => transferring context: 142B                                          0.0s
 => CACHED [5gc base 2/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 3/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 4/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 5/8] RUN echo v2.7.0 > ./open5gsversion               0.0s
 => CACHED [5gc base 6/8] RUN git clone --depth 1 --branch $(cat ./open5g  0.0s
 => CACHED [5gc base 7/8] RUN curl -fsSL https://deb.nodesource.com/setup  0.0s
 => CACHED [5gc base 8/8] RUN python3 -m pip install pymongo click pyrout  0.0s
 => CACHED [5gc open5gs 1/3] WORKDIR /open5gs                              0.0s
 => CACHED [5gc open5gs 2/3] COPY open5gs-5gc.yml open5gs-5gc.yml.in       0.0s
 => CACHED [5gc open5gs 3/3] COPY open5gs_entrypoint.sh add_users.py setu  0.0s
 => [5gc] exporting to image                                               0.1s
 => => exporting layers                                                    0.0s
 => => writing image sha256:08070cfd35e4496cf9aacafe920e3eac994d45af1a598  0.0s
 => => naming to docker.io/library/docker-5gc                              0.0s
Attaching to open5gs_5gc
Error response from daemon: driver failed programming external connectivity on endpoint open5gs_5gc (0e7b784417e89be92767f5694d59ffa3a38bb97403b5d6b41fb571833ba1424c): Error starting userland proxy: listen tcp4 0.0.0.0:9999: bind: address already in use