srsran/srsRAN_Project #327: Cannot connect more than one UE to gNB (ZMQ)

raul-parada, 2023-11-22:
## Issue Description ##
I cannot connect more than one UE to gNB (ZMQ)

## Setup Details ##
I'm following this tutorial https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html

I follow this steps:

  ```
  1-Execute in a terminal `docker-compose up --build 5gc `from srsRAN_Project/docker
    2-Execute in another terminal .`/flexric/build/examples/ric/nearRT-RIC `    
    3-Execute in another terminal `sudo ./gnb -c gnb_zmq.yaml` on /srsRAN_Project/build/apps/gnb 
    4-Execute UE `sudo ip netns add ue1` and `sudo ./srsue ue_zmq.conf `on srsRAN_4G/build/srsue/src 
    5-Add addicional arguments to gnb_zmq.yaml device_args: tx_port=tcp://127.0.0.1:2002,rx_port=tcp://127.0.0.1:2003,base_srate=11.52e6
    6-Create UE.conf for UE2 (attached). I've modified zmq ports, logs names and uSIM IDs
    7- Execute UE `sudo ip netns add ue2` and `sudo ./srsue ue2_zmq.conf `on srsRAN_4G/build/srsue/src
```

## Expected Behavior ##
Successful PDU session establishment for UE2 (together with UE1)

## Actual Behaviour ##
r```
parada@rparada-NUC7i7BNH:~/srsRAN_4G/build/srsue/src$ sudo ./srsue ue2_zmq.conf
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue2_zmq.conf...

Built in Release mode using commit fa56836b1 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2003,rx_port=tcp://127.0.0.1:2002,base_srate=11.52e6
Supported RF device list: zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://127.0.0.1:2002
CH0 tx_port=tcp://127.0.0.1:2003
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
```

Is it possible to connect multiple UEs? How gNB (5GCore) provide an IP to UE2 and more UEs?
[u2conf.txt](https://github.com/srsran/srsRAN_Project/files/13435643/u2conf.txt)

pgawlowicz, 2023-11-22:
Hi, 
Currently, it is not supported to connect multiple UEs directly to the gnb. You need to add a "channel" in-between. 
See those two discussions: 
- https://github.com/srsran/srsRAN_Project/discussions/131
- https://github.com/srsran/srsRAN_4G/discussions/1203#discussioncomment-6465385

pgawlowicz, 2023-11-23:
This is a third-party tutorial, that I pointed to only as a reference explaining the approach in general.

The basic idea is as follows:
1. Currently, srsGNB and srsUE use ZMQ in point-to-point mode, so it is not possible to connect multiple UEs to the same GNB.
2. We need a splitter in DL and an aggregator in UL, that can be implemented in GnuRadio:
[image]

As you can see in the above figure, there is a ZMQ REQ source that is connected to GNB. It sends a copy of samples to 3 ZMQ REP Sinks that are then connected to 3 UEs.

Similar in UL, but here we need to add signals from all 3 UEs and then send the aggregated signal to GNB.

raul-parada, 2023-11-23:
I've already done that. Here is my comment to the forwarded issue:

_Hi. Did someone make it work? I have built the above flowchart on GNU Radio. I have some doubts: 1) Do we need to create the ue.conf from all UEs? I have created 3 additional UEs with the following arguments: device_args = tx_port=tcp://127.0.0.1:2X00,rx_port=tcp://127.0.0.1:2X01,base_srate=11.52e6 is it correct? If run the ue.conf then the address is already used and by running the flowchart I receive that the bind address is already used. If I execute the processes vice versa, I receive the following error when executing the ue.conf ```
Error: binding transmitter socket (tcp://127.0.0.1:2001): Address already in use
[zmq] Error: opening transmitter_
_Error initializing radio._

So, I did understand why GNU Radio is used, however, I don't know how to merge both deployments. Thanks.

pgawlowicz, 2023-11-23:
Could you share a screenshot of your gnuradio flowgraph and GNB as well as 2 UEs configs?

raul-parada, 2023-11-23:
The UE are located within srsRAN4G while gNB in srsRAN_Project.
[image]
[ue_conf.txt](https://github.com/srsran/srsRAN_Project/files/13451131/ue_conf.txt)
[ue2_conf.txt](https://github.com/srsran/srsRAN_Project/files/13451132/ue2_conf.txt)
[gnb_conf.txt](https://github.com/srsran/srsRAN_Project/files/13451140/gnb_conf.txt)

pgawlowicz, 2023-11-23:
The current config is:

GNB has 2 lines with `device_args`, but should be one:
```
  device_args: tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=11.52e6 #
  device_args: tx_port=tcp://127.0.0.1:2002,rx_port=tcp://127.0.0.1:2003,base_srate=11.52e6 #Optionally pass arguments to the selected RF driver.
```
UE:
```
device_args = tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=11.52e6
```
UE2:
```
device_args = tx_port=tcp://127.0.0.1:2100,rx_port=tcp://127.0.0.1:2101,base_srate=11.52e6
```

So gnb uses `rx_port=tcp://127.0.0.1:2001` and UE uses `tx_port=tcp://127.0.0.1:2001` that is why you get the info that the address is in use already.

You need the following changes:
- Sample rate in gnuradio from 23.04 M to 11.52M. 
- The gnuradio flowgraph is for 3 UEs, but you only have 2, so  you need to remove one ZMQ REP Source (with port=2301) and one ZMQ REP Sink (with port 2300). 

Then you need those changes in configs:
GNB:
```
  device_args: tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=11.52e6 
```
UE:
```
device_args = tx_port=tcp://127.0.0.1:2101,rx_port=tcp://127.0.0.1:2100,base_srate=11.52e6
```
UE2:
```
device_args = tx_port=tcp://127.0.0.1:2201,rx_port=tcp://127.0.0.1:2200,base_srate=11.52e6
```

raul-parada, 2023-11-23:
I've just uploaded 2 out of the three configured. 
I've tried to run everything following these steps:
1-Execute in a terminal sudo docker-compose up --build 5gc from srsRAN_Project/docker 
2-Execute in another terminal ./flexric/build/examples/ric/nearRT-RIC 
3-Execute in another terminal sudo ./gnb -c gnb_zmq.yaml on srsRAN_Project/build/apps/gnb 
4-Execute in another terminal sudo ip netns add ue1, sudo ip netns add ue2 and sudo ip netns add ue3 on srsRAN_4G/build/srsue/src
5- When Executing sudo ./srsue ue_zmq.conf on srsRAN_4G/build/srsue/src the output remains as:

```
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue_zmq.conf...

Built in Release mode using commit fa56836b1 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2101,rx_port=tcp://127.0.0.1:2100,base_srate=11.52e6
Supported RF device list: zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://127.0.0.1:2100
CH0 tx_port=tcp://127.0.0.1:2101
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...

```  
And no IP is assigned. Then, I run the GNU Radio flowchart without error. However, the above output remains the same. Obviously even when executing the following command to check the traffic `sudo ip netns exec ue1 ping -i 0.1 10.45.1.1` the network is unreachable. I'm doing the right procedure?

pgawlowicz, 2023-11-23:
I think you need to start all three UEs, otherwise the gnuradio flowchart will be waiting for samples. 
You can execute `nload` command to see the traffic on the loopback interface, if samples are sent it should be a quite high load.

raul-parada, 2023-11-23:
It works executing all three UE.conf files before starting GNU Radio. However, I'm wondering how can I send data from UE to the Internet through the 5GCore when no IP address has been assigned to the UE, generally, 10.45.1.2.

pgawlowicz, 2023-11-23:
Could you execute `nload` command and see whether there is traffic on the loopback (lo) interface? If the signal samples are passed between gnb and UEs, the traffic volume should be quite high. (you can check it, running the working setup with 1 gnb and 1 ue connected directly without the gnuradio).
I suspect that the signal samples transferred that is why UEs do not get any signal samples, and hence cannot get IP address.

raul-parada, 2023-11-23:
[image]
yes, there is traffic. So, there is not option right now to get IP? Otherwise, with this setup can re-direct specific data from another source?

pgawlowicz, 2023-11-23:
could you share your configs and gnuradio graph? i will replicate it locally.

raul-parada, 2023-11-23:
[UE.zip](https://github.com/srsran/srsRAN_Project/files/13452125/UE.zip)
[gnb_zmq.zip](https://github.com/srsran/srsRAN_Project/files/13452130/gnb_zmq.zip)
[MultiUE5G.zip](https://github.com/srsran/srsRAN_Project/files/13452137/MultiUE5G.zip)
Let me know if you require more files. I use the srsRAN_Project/docker as 5GCore

raul-parada, 2023-11-28:
Did you have the chance to replicate this environment?

pgawlowicz, 2023-11-28:
Yes, it seems to be a bigger issue with srsUE. We are working on the fix. I will keep you informed.

Zplusless, 2023-12-07:
Below is my tests and logs. Two UEs, both "RRC connected", but only ue2 got IP with poor throughput.

https://github.com/srsran/srsRAN_Project/discussions/40#discussioncomment-7759997

pgawlowicz, 2024-01-15:
@raul-parada Yes, we do have a patch for the srsUE. Just need more testing. It will be soon released.

raul-parada, 2024-01-23:
Great! please, @pgawlowicz, let me know when the new version is released

pgawlowicz, 2024-01-31:
@raul-parada Please check the new section of our gnb+srsUE tutorial that describes how to emulate multiple-UEs:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#multi-ue-emulation

Please remember to clone the newest version of the srsRAN-4G repository.

raul-parada, 2024-02-01:
Hi, I've followed the tutorial (and installed the new srs4G repository). However, only one out of the three UEs receive an IP

pgawlowicz, 2024-02-01:
Note that, due to the way the Open5GC docker container is built,  you have to name the [subscriber_db.csv](https://docs.srsran.com/projects/project/en/latest/_downloads/5d59e27a3d39c6f9b72df6ca2b01d0e2/subscriber_db.csv) as `subscriber_db.csv` and put it into `srsRAN_Project/docker/open5gs/`

Moreover, you have to set exactly this env variable `SUBSCRIBER_DB="subscriber_db.csv"` in your `srsRAN_Project/docker/open5gs/open5gs.env`

raul-parada, 2024-02-01:
[image]
Please, re-open the issue since it hasn't been solved.

pgawlowicz, 2024-02-01:
hmm, ok, could you provide the screenshot of the Open5GC webgui?

raul-parada, 2024-02-01:
[image]

pgawlowicz, 2024-02-01:
while running the Open5gc docker container, could you execute: 
```
docker exec -it open5gs_5gc /bin/bash
```

and then inside the docker 
```
echo $SUBSCRIBER_DB
cat subscriber_db.csv
```

raul-parada, 2024-02-01:
[log5gc.txt](https://github.com/srsran/srsRAN_Project/files/14128654/log5gc.txt)

pgawlowicz, 2024-02-01:
hmm you have
```
root@dfefceb4a4f3:/open5gs# echo $SUBSCRIBER_DB
001010123456780,00112233445566778899aabbccddeeff,opc,63bfa50ee6523365ff14c1f45f88737d,8000,9,10.45.1.2
```
and it should be set to `SUBSCRIBER_DB="subscriber_db.csv"`.
Not sure, why the value is not propagated correctly. 
Anyway, inside the open5gs docker container, you can add the subscribers manually:
```
root@dfefceb4a4f3:/open5gs# python3 ./add_users.py --subscriber_data=./subscriber_db.csv
```

raul-parada, 2024-02-02:
It works. Trying to transmit data via iperf3 I got this error (server is listening):

```
rparada@rparada-NUC7i7BNH:~$ sudo ip netns exec ue1 iperf3 -c 10.45.1.1 -b 10M -i 1 -t 60
iperf3: error - unable to send control message: Bad file descriptor
```

pgawlowicz, 2024-02-02:
Thanks for pointing this out. Indeed, the iperf3 client from UE will not reach the server running on the host machine with the command from tutorial.

I will update the tutorial. Meanwhile, you can try the following solutions:
**Solution 1.** Start iperf3 server inside open5gs docker, i.e.:
```
docker exec -it open5gs_5gc iperf3 -s -i 1
```
Then use the command from the tutorial:
```
sudo ip netns exec ue1 iperf3 -c 10.45.1.1 -u -b 10M -i 1 -t 60
```

**Solution 2.** Start iperf3 server on the host machine (one that is running the open5gs container):
```
iperf3 -s -i 1
```
then start iperf3 client with:
```
sudo ip netns exec ue1 iperf3 -c 10.53.1.1 -u -b 10M -i 1 -t 60
```
**Note:** all routes have to be configured as presented in the tutorial.

**Solution 3.** Use iperf client that in UDP mode does not require iperf server to be connected. 
```
sudo ip netns exec ue1 iperf -c 10.45.1.1 -u -b 10M -i 1 -t 60
```
**Note:** this way you can inject UL traffic, but you will not get a proper data rate values from iperf. You can check the data rate inside srsUE console by typing `t`.
To inject DL traffic:
```
docker exec -it open5gs_5gc iperf -c 10.45.1.2 -u -b 50M -i 1 -t 60
```

pgawlowicz, 2024-02-02:
@raul-parada does it work?

raul-parada, 2024-02-02:
Solution 2 works

closed 2024-02-02 by pgawlowicz

chelimi, 2024-05-31:
Trying to transmit data via iperf3 I got this error (server is listening):
waxwing@waxwing:~$ sudo ip netns exec ue1 iperf3 -c 10.53.1.1 -u -b 10M -i 1 -t 60
[sudo] password for waxwing: 
iperf3: error - unable to send control message: Bad file descriptor

waxwing@waxwing:~$ sudo ip netns exec ue1 iperf3 -c 10.45.1.1 -u -b 10M -i 1 -t 60
iperf3: error - unable to send control message: Bad file descriptor