srsran/srsRAN_Project #332: srsran7.2 Message acceptance problem

doudoulail, 2023-11-24:
I am debugging srsran and Foxconn rru encountered such a problem, receiving messages too late or too early, what is the reason?
```
2023-11-23T03:19:44.371826 [OFH     ] [I] Received packets: rx_total=57203 rx_early=0, rx_on_time=0, rx_late=57203
2023-11-23T03:19:45.371824 [OFH     ] [I] Received packets: rx_total=57194 rx_early=0, rx_on_time=0, rx_late=57194
2023-11-23T03:19:46.371825 [OFH     ] [I] Received packets: rx_total=57199 rx_early=0, rx_on_time=0, rx_late=57199
2023-11-23T03:19:47.371822 [OFH     ] [I] Received packets: rx_total=57200 rx_early=0, rx_on_time=0, rx_late=57200
2023-11-23T03:19:48.371824 [OFH     ] [I] Received packets: rx_total=57194 rx_early=0, rx_on_time=0, rx_late=57194
2023-11-23T03:19:49.371825 [OFH     ] [I] Received packets: rx_total=57200 rx_early=0, rx_on_time=0, rx_late=57200
2023-11-23T03:19:50.371825 [OFH     ] [I] Received packets: rx_total=57208 rx_early=0, rx_on_time=0, rx_late=57208
2023-11-23T03:19:51.371825 [OFH     ] [I] Received packets: rx_total=57191 rx_early=0, rx_on_time=0, rx_late=57191
2023-11-23T03:19:52.371823 [OFH     ] [I] Received packets: rx_total=57201 rx_early=0, rx_on_time=0, rx_late=57201
2023-11-23T03:19:53.371822 [OFH     ] [I] Received packets: rx_total=57204 rx_early=0, rx_on_time=0, rx_late=57204
2023-11-23T03:19:54.371825 [OFH     ] [I] Received packets: rx_total=57199 rx_early=0, rx_on_time=0, rx_late=57199
```
[gnb.log](https://github.com/srsran/srsRAN_Project/files/13454385/gnb.log)
Configuration file of srsran
[test.txt](https://github.com/srsran/srsRAN_Project/files/13454415/test.txt)

sauka, 2023-11-24:
Hi @doudoulail ,

The lates in the log mean that the reception window is too short, you need to increase the `ta4_max` value in the gNB config. It may need some adjustment depending on your setup and system configuration. If you see "earlies", then the `ta4_min` needs to be decreased, for example you can put it to 0.

doudoulail, 2023-11-25:
Hi @sauka I've tried setting it to 0, but it's still there

ismagom, 2023-11-25:
It could also be that sync is not correct. Do you see lates on the RU side?

doudoulail, 2023-11-25:
Yeah, I saw the sync on the RU side
```
ptp4l[1048.287]: master offset          4 s2 freq     -22 path delay       364
ptp4l[1049.287]: master offset          0 s2 freq     -25 path delay       364
ptp4l[1050.287]: master offset        -10 s2 freq     -35 path delay       365
ptp4l[1051.287]: master offset          4 s2 freq     -24 path delay       365
ptp4l[1052.287]: master offset          4 s2 freq     -23 path delay       365
ptp4l[1053.287]: master offset          7 s2 freq     -19 path delay       366
ptp4l[1054.287]: master offset         -1 s2 freq     -25 path delay       366
ptp4l[1055.288]: master offset         -7 s2 freq     -31 path delay       366
ptp4l[1056.287]: master offset         -4 s2 freq     -30 path delay       366
ptp4l[1057.287]: master offset          3 s2 freq     -24 path delay       366
ptp4l[1058.287]: master offset          3 s2 freq     -23 path delay       368
ptp4l[1059.288]: master offset          0 s2 freq     -26 path delay       368
ptp4l[1060.288]: master offset         -3 s2 freq     -28 path delay       368
ptp4l[1061.288]: master offset         -4 s2 freq     -30 path delay       368
ptp4l[1062.288]: master offset          1 s2 freq     -27 path delay       366
ptp4l[1063.288]: master offset          1 s2 freq     -26 path delay       366
ptp4l[1064.288]: master offset         -3 s2 freq     -30 path delay       366
ptp4l[1065.288]: master offset         -8 s2 freq     -36 path delay       366
ptp4l[1066.288]: master offset          5 s2 freq     -25 path delay       366
ptp4l[1067.289]: master offset         -1 s2 freq     -30 path delay       366
ptp4l[1068.289]: master offset         -9 s2 freq     -38 path delay       366
ptp4l[1069.289]: master offset         -1 s2 freq     -33 path delay       366
ptp4l[1070.289]: master offset         -9 s2 freq     -41 path delay       366
ptp4l[1071.289]: master offset         -6 s2 freq     -41 path delay       365
ptp4l[1072.289]: master offset          3 s2 freq     -34 path delay       364
ptp4l[1073.289]: master offset         -2 s2 freq     -38 path delay       364
ptp4l[1074.289]: master offset          8 s2 freq     -28 path delay       364
ptp4l[1075.290]: master offset          0 s2 freq     -34 path delay       364
ptp4l[1076.290]: master offset          7 s2 freq     -27 path delay       364
ptp4l[1077.290]: master offset          8 s2 freq     -24 path delay       364
ptp4l[1078.290]: master offset          9 s2 freq     -20 path delay       364
ptp4l[1079.290]: master offset          9 s2 freq     -18 path delay       364
ptp4l[1080.290]: master offset          4 s2 freq     -20 path delay       364
ptp4l[1081.290]: master offset          7 s2 freq     -16 path delay       364
ptp4l[1082.291]: master offset         -2 s2 freq     -23 path delay       364
ptp4l[1083.291]: master offset          1 s2 freq     -20 path delay       364
ptp4l[1084.291]: master offset          0 s2 freq     -21 path delay       366
ptp4l[1085.291]: master offset          0 s2 freq     -21 path delay       366
ptp4l[1086.291]: master offset         -2 s2 freq     -23 path delay       366
ptp4l[1087.291]: master offset          5 s2 freq     -17 path delay       366
ptp4l[1088.291]: master offset          4 s2 freq     -16 path delay       366
ptp4l[1089.292]: master offset          6 s2 freq     -13 path delay       368
ptp4l[1090.291]: master offset         -2 s2 freq     -19 path delay       368
```
I looked up the ptp situation in sync,I guess there's some issues on the ru side

closed 2023-12-05 by doudoulail