srsran/srsRAN_Project #334: Split 7.2 OFH - Benetel RAN650 - high MCS issue

mateuszmichalskiacc, 2023-11-24:
## Issue Description ##
This bug report is outcome of this discussion: https://github.com/srsran/srsRAN_Project/discussions/325
The MCS>24 can not be achieved for DL max tput test.
Smaller PDSCH grants (less PRBs) are fine with the MCS 27 and MCS 28.
When full BW (100MHz) is used, the DL BLER increases and LA decreases the MCS.

## Setup Details ##
srsRAN version: release_23_10_1
core: open5gs
RU: Benetel RAN650 v0.8.1- 100MHz 2x2 
Amarisoft UE in Amarisoft e simbox

RF cables used instead of OTA

## Expected Behavior ##
Highest MCS for qam256 achievable in perfect RF cabled setup

## Actual Behaviour ##
MCS 27-28 used with successful for small DL packets (small number of PRB), 
MCS 26 used when PDSCH allocation limited to 130-180 PRB
[image]

MCS 24 used when full BW used (100MHz)
[image]

## Steps to reproduce the problem ##
start iperf3 DL UDP full tput test

## Additional Information ##
The same Amarisoft UE simbox can achieve DL MCS 27 with full DL tput test with different DU vendor and also Benetel RAN650

logs:  https://drive.google.com/drive/folders/1XfTFcnyU3LLxnpJXmvtV6XHb5DIyn7VC?usp=sharing

ismagom, 2023-11-24:
Thanks for reporting.

Have you tried increasing BLER target to 10%?

With the other DU, what BLER and throughput do you get when doing MCS 27 and same BW?

We don't have a RAN650 yet so we can't test ourselves, you need to find the optimal iq_scale value.

ismagom, 2023-11-24:
One more question: what compression mode you use with the other DU vendor?

ismagom, 2023-11-24:
After looking at the config again, we think the problem could be DC offset, can you try adding `dc_offset: center` in the `pdsch` section?

mateuszmichalskiacc, 2023-11-30:
Hi @ismagom 
1. static compression
2. We've moved to 4x2 mode on RU, and max achieved MCS is even lower (19-22) than for 2x2. 
The same MCS for UE which is 4x4 DL and 2x2 DL, when RU is 4x2 mode.
3. dc_offset doesn't seem to help, I see no difference. 
4. With real UE (but still not OTA) in 2x2 RU mode, we can achieve a little bit higher MCS (25-26) , but for 4x4 only 22.
5. I've tried already various of values for iq_scale, RU output power and UE rx_gain, no significant difference was noted.

We set up second setup with different Bentel RAN650 unit and results are similar.
Actually we can't reach peak tput on any of setups with any of UEs.

closed 2024-04-11 by pgawlowicz