srsran/srsRAN_Project #335: Couldn't parse amplitude control

AzeezEbrahim, 2023-11-26:
## Issue Description ##
So, i am training to maximize the transmit power level, so i found paramets in the GNB config file.
So when i try to parse the amplitude control, such as power_ceiling_dBFs, and tx_gain_backoff.
i am facing an error `INI was not able to parse ru_sdr.cells.++`

idk, how else to increase the power level, and idk if i am doing it right or wrong. is there another place to increase the transmit power? a

## Setup Details ##
I am using b210 usrp and open5gs core network. 

## Expected Behavior ##
it should parse it, unless if it is not implemented 

## Actual Behaviour ##
[image]

## Steps to reproduce the problem ##
this is my GNB yaml file.

```# This configuration file example shows how to configure the srsRAN Project gNB to connect to a COTS UE. As with the 
# associated tutorial, this config has been tested with a OnePlus 8T. A B210 USRP is used as the RF-frontend.   
# This config creates a TDD cell with 20 MHz bandwidth in band 78. 
# To run the srsRAN Project gNB with this config, use the following command: 
#   sudo ./gnb -c gnb_b210_20MHz_oneplus_8t.yaml

amf:
  addr: 127.0.1.100                                             # The address or hostname of the AMF.
  bind_addr: 127.0.0.1                                          # A local IP that the gNB binds to for traffic from the AMF.

ru_sdr:
  device_driver: uhd                                            # The RF driver name.
  device_args: type=b200,num_recv_frames=64,num_send_frames=64  # Optionally pass arguments to the selected RF driver.
  clock: internal                                               # Specify the clock source used by the RF.
  sync:                                                         # Specify the sync source used by the RF.
  srate: 30.72                                                  # RF sample rate might need to be adjusted according to selected bandwidth.
  # srate: 61.44
  otw_format: sc12
  tx_gain: 89.75                                                   # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 70                                                   # Receive gain of the RF might need to adjusted to the given situation.
  
  cells:                                # Optional TEXT. Sets the hardware specific cell configuration on a per cell basis. 
    amplitude control: 
      tx_gain_backoff: 12.0                     # Optional FLOAT (12.0). Sets baseband gain back-off in dB. This accounts for the signal PAPR and is applied regardless of clipping settings. Format: positive float.
      enable_clipping: false                    # Optional BOOL (false). Sets clipping of the baseband samples on or off. If enabled, samples that exceed the power ceiling are clipped.
      power_ceiling_dBFs: -0.1                  # Optional FLOAT (-0.1). Sets the power ceiling in dB, relative to the full scale amplitude of the radio. Format: negative float or 0.

cell_cfg:
  dl_arfcn: 627340                                              # ARFCN of the downlink carrier (center frequency).
  band: 78                                                      # The NR band.
  channel_bandwidth_MHz: 20                                     # Bandwith in MHz. Number of PRBs will be automatically derived.
  # channel_bandwidth_MHz: 50
  common_scs: 30                                                # Subcarrier spacing in kHz used for data.
  plmn: "99970"                                                 # PLMN broadcasted by the gNB.
  tac: 7                                                        # Tracking area code (needs to match the core configuration).
  pci: 1                                                        # Physical cell ID.

  pcg_p_nr_fr1: 23                      # Optional INT (10). Sets the maximum total TX power to be used by the UE in this NR cell group across in FR1. Supported: [-30 - +23].
  nof_antennas_dl: 2  # MIMO: 2 - SISO: 1
  nof_antennas_ul: 2  # MIMO: 2 - SISO: 1
  ssb:
    ssb_period: 20
    ssb_block_power_dbm: 50                # Optional INT (-16). Sets the SS PBCH block power in dBm. Supported: [-60 - +50].
  pdsch:
    mcs_table: qam256
  ul_common: 
    p_max: 23                                  # Optional TEXT. Sets maximum transmit power allowed in this serving cell. Supported: [-30 - +23].

# All gNB layers and components can be configur
![image](https://github.com/srsran/srsRAN_Project/assets/87777192/14b38d77-98cd-4ec5-9c31-9a7ddc372c5d)
ed independently to output at various levels of detail. Logs can be configured to the following levels (from lowest to highest levels of detail):
# none, error, warning, info, debug
 #filename:                             # Optional TEXT (/tmp/gnb.log). File path for logs
log: 
  all_level: debug                    # Optional TEXT (warning). Sets a common log level across PHY, MAC, RLC, PDCP, RRC, SDAP, NGAP and GTPU layers

pcap:
  mac_enable: true                                             # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap                               # Path where the MAC PCAP is stored.
  ngap_enable: false                                            # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap                             # Path where the NGAP PCAP is stored.
  
#   e2ap_enable: true                 # Set to true to enable E2AP PCAPs.
#   e2ap_filename: /tmp/gnb_e2ap.pcap # Path where the E2AP PCAP is stored.

# e2:
#   enable_du_e2: true                # Enable DU E2 agent (one for each DU instance)
#   e2sm_kpm_enabled: true            # Enable KPM service module```

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

AlaiaL, 2023-12-11:
Hello AzeezEbrahim,

I don't think your config is correct to configure these parameters. Can you try to replace your `ru_sdr` section with this one? 

```
ru_sdr:
  device_driver: uhd                                            # The RF driver name.
  device_args: type=b200,num_recv_frames=64,num_send_frames=64  # Optionally pass arguments to the selected RF driver.
  clock: internal                                               # Specify the clock source used by the RF.
  sync:                                                         # Specify the sync source used by the RF.
  srate: 30.72                                                  # RF sample rate might need to be adjusted according to selected bandwidth.
  # srate: 61.44
  otw_format: sc12
  tx_gain: 89.75                                                   # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 70                                                   # Receive gain of the RF might need to adjusted to the given situation.
  cells:                                # Optional TEXT. Sets the hardware specific cell configuration on a per cell basis.
    - amplitude_control:
        tx_gain_backoff: 12.0                     # Optional FLOAT (12.0). Sets baseband gain back-off in dB. This accounts for the signal PAPR and is applied regardless of clipping settings. Format: positive float.
        enable_clipping: false                    # Optional BOOL (false). Sets clipping of the baseband samples on or off. If enabled, samples that exceed the power ceiling are clipped.
        ceiling: -0.1                  # Optional FLOAT (-0.1). Sets the power ceiling in dB, relative to the full scale amplitude of the radio. Format: negative float or 0.

```

AzeezEbrahim, 2023-12-13:
ok it worked, thanks @AlaiaL .

i took it from docs, so i think it needs some updates
[https://docs.srsran.com/projects/project/en/latest/user_manuals/source/config_ref.html](url)

closed 2023-12-13 by AzeezEbrahim

KianooshSepas, 2024-10-07:
Hi all,

I've followed the [srsRAN installation guide](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/installation.html) and successfully set up Open5GS as the core network. However, when I try to run the gNB using the following command:

bash

./gnb -c ./gnb_zmq.yaml

I encounter the following error:

sql

--== srsRAN gNB (commit ) ==--

INI was not able to parse amf.++
Run with --help for more information.

Could someone help me resolve this issue? It seems to be related to the amf configuration in the gnb_zmq.yaml file, but I haven't been able to fix it.
and this I just download the files from here https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup

Thanks in advance!

AlaiaL, 2024-10-08:
Hi [KianooshSepas](https://github.com/KianooshSepas),

Can you share your config and the commit hash of the gNB? 
Also in the `configs` folder of the project, can you check if the file `amf.yml` is present (https://github.com/srsran/srsRAN_Project/blob/main/configs/amf.yml)?

Recently we did some changes in the YAML schema regarding the amf section.

Thanks

EspiMCF, 2024-10-08:
Hello @AlaiaL ,

Same problem as @KianooshSepas, this is my amf section (also attached the whole config file): 
[gnb-2x2-50-fdd.txt](https://github.com/user-attachments/files/17293883/gnb-2x2-50-fdd.txt)

```
amf:
  addr: 10.10.30.90
  bind_addr: 10.10.30.91
  n2_bind_addr: 10.10.30.91             # Optional TEXT. Sets local IP address to bind for N2 interface. Format: IPV4 or IPV6>
  n2_bind_interface: eno8303np0.306     # Optional TEXT. Sets the network device to bind to for N2 interface.
  n3_bind_addr: 10.10.30.101            # Optional TEXT. Sets local IP address to bind for N3 interface. Format: IPV4 or IPV6>
  n3_bind_interface: eno8303np0.307     # Optional TEXT. Sets the network device to bind to for N3 interface.
  n3_ext_addr: 10.10.30.100             # Optional TEXT. Sets the external IP address that is advertised to receive GTP-U pac>
```

This is the output:
```
$ ./gnb --help -c gnb-2x2-50-fdd.yaml

--== srsRAN gNB (commit e73b46182) ==--

INI was not able to parse amf.++
Run with --help for more information.
```
And I had also a previous issue using vlan_tag parameter for the ru_ofh section, that's why I needed to comment that line.

AlaiaL, 2024-10-08:
Hi [EspiMCF](https://github.com/EspiMCF),

`vlan_tag` was updated to allow to configure the 2 planes of OFH, `vlan_tag_cp` (for Control-Plane) and `vlan_tag_up` (for User-Plane). 

Regarding the amf section, that part of the configuration was recently changed, there are some examples in the configs folders, but it looks something like this:

```
cu_cp:
  amf:
    addr: 127.0.1.100
    port: 38412
    bind_addr: 127.0.0.1
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1
```

The n2 related parameters are in the `cu_cp --> amf:` section, while the n3 parameters are in `upf` section. 

Printing some help of the gNB may help you to update your configs (current main branch of the repository): 
```
aleal@Wazyx~/Repositories/Public/srsRAN_Project/build(main)$ ./apps/gnb/gnb cu_cp amf --help

--== srsRAN gNB (commit e73b46182) ==--

AMF configuration
Usage: ./apps/gnb/gnb cu_cp amf [OPTIONS]

Options:
  -h,--help                   Print this help message and exit
  --no_core BOOLEAN [false]   Allow CU-CP to run without a core
  --addr TEXT                 AMF IP address
  --port UINT:INT in [20000 - 40000] [38412] 
                              AMF port
  --bind_addr TEXT:IPV4       Local IP address to bind for N2 interface
  --bind_interface TEXT [auto] 
                              Network device to bind for N2 interface
  --sctp_rto_initial INT      SCTP initial RTO value
  --sctp_rto_min INT          SCTP RTO min
  --sctp_rto_max INT          SCTP RTO max
  --sctp_init_max_attempts INT
                              SCTP init max attempts
  --sctp_max_init_timeo INT   SCTP max init timeout 
  --sctp_nodelay BOOLEAN      Send SCTP messages as soon as possible without any Nagle-like algorithm
  --supported_tracking_areas TEXT ...
                              Sets the list of tracking areas supported by this AMF

aleal@Wazyx~/Repositories/Public/srsRAN_Project/build(main)$  ./apps/gnb/gnb upf --help

--== srsRAN gNB (commit e73b46182) ==--

UPF parameters
Usage: ./apps/gnb/gnb upf [OPTIONS]

Options:
  -h,--help                   Print this help message and exit
  --bind_addr TEXT:(IPV4) OR ({auto})
                              Local IP address to bind for N3 interface
  --bind_interface TEXT [auto] 
                              Network device to bind for N3 interface
  --ext_addr TEXT:(IPV4) OR ({auto})
                              External IP address that is advertised to receive GTP-U packets from UPF via N3 interface
  --udp_max_rx_msgs INT       Maximum amount of messages RX in a single syscall
  --pool_threshold FLOAT:FLOAT in [0 - 1]
                              Pool accupancy threshold after which packets are dropped
  --no_core BOOLEAN           Allow gNB to run without a core

```

Regards