srsran/srsRAN_Project #338: RRC Release

AEncinas19, 2023-11-27:
Hello,
I have followed the issue #254 where it indicates that
              > The issue has been fixed in https://github.com/srsran/srsRAN_Project/tree/test . Thanks @hajoha for confirming.

[logsandconfs.zip](https://github.com/srsran/srsRAN_Project/files/13450623/logsandconfs.zip)
How is the srsue configuration file? I'm using the new test commit and gnb configuration allocated in configs/gnb_rf_b210_fdd_srsUE.yml and my srsUE after PDU Session Establishment receives an RRC Release due to LCID 1 doesn't exist. 

The .zip file contains:
- gNB configuration
- srsue configuration
- gNB logs
- srsue logs

Thanks in advance.

_Originally posted by @AEncinas19 in https://github.com/srsran/srsRAN_Project/issues/254#issuecomment-1824470016_

pgawlowicz, 2023-11-27:
Hi, seems that the UL SNR is quite low (i.e, around 0-1dB), could you try to increase the tx_gain in the UE config?

AEncinas19, 2023-11-27:
Thank you for the support. I've put 80 in the tx_gain (the maximum value accepted), I reach SNRs around 10-20 dBs, but it still does not work. However, the ue.log has changed. I attatch them bellow.
[ue.log](https://github.com/srsran/srsRAN_Project/files/13474069/ue.log)

It seems that the NR Connection Reconfiguration fails, which implies that the gnb sends a UeContestReleaseRequest and removes the UE. The srsUE supports QoS with traffic filters? I attatch below the Open5gs QoS configuration.
[image]

pgawlowicz, 2023-11-27:
Your config seems to be running correctly over ZMQ RF channel. Are you sure you use the newest releases of srsRAN and srsRAN_4g?

AEncinas19, 2023-11-27:
I'm not ussing ZMQ, I'm using an over-the-air setup. For gNB I'm using the commit 0fd7c93 from the test branch and for srsUE I'm using commit eea87b1 from the master branch.

pgawlowicz, 2023-11-27:
Yes, but I have tested your configs with zmq radio just to confirm it is not a config-related problem. Seems to be OK, also with the versions you use. 
Do you have an option to use an external clock for both gnb and srsUE?

pgawlowicz, 2023-11-27:
Did you change the Open5GS configs?
Could you try to run Open5GS using a docker container that we provide? You can find the tutorial here:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html#open5gs-core

AEncinas19, 2023-11-28:
Well, my Open5GS implementation is slightly different due to be compliant with the 3GPP Architecture and each 5G element is deployed in a separated container. However, there is connection between gnB and AMF and between gnB and UPF and if I don't create PCC rules in the Open5GS webui all works perfectly. Anyway, tomorrow I will also try the Docker Container that you provide.

Thank you for the support.

AEncinas19, 2023-11-30:
Well, it works all fine until I define a QoS filter in Open5GS. If I don't define any uplink or downlink filter in the PCC rule, all works fine.

herlesupreeth, 2023-12-05:
> I think it is not a problem in qos.yaml from gNB because I also try it with a COTS UE and all works fine without that error.

Then it could be that srsUE does not support this. If you could attach the F1AP and MAC pcap then we can know where the issue is.

AEncinas19, 2023-12-05:
Yes. it seems that srsUE does not support QoS.

closed 2023-12-05 by AEncinas19

BENIEJAISON, 2025-02-17:
What must be done in order to disable PCC rules in open5gs so that RRC release error can be avoided in srsue