srsran/srsRAN_Project #343: UE <-> gNB connection crashes after ping

oliverxsch, 2023-11-28:
## Issue Description ##
The connection between the UE and gNB crashes after an while.

## Setup Details ##
I already opened another issue https://github.com/srsran/srsRAN_Project/issues/315
But I closed it because most of the time im now having a different problem. 
Im using an over the air setup.

First PC:
Specs: i9 13900, 64GB Ram
OS: Ubuntu 22.04.03 LTS
UHD 3.15 LTS
Newest srsRAN Project and newest Open5Gs.
Connected via a 10G-Link to a USRP X310
"gNB"

Second PC:
Specs: i9 13900, 64GB Ram
OS: Ubuntu 22.04.03 LTS
UHD 3.15 LTS
Newest srsRAN-4G Project.
Connected via a 10G-Link to another USRP X310
"UE"

Both USRPs are connected to an CDA 2990 G for their sync and clock. I have the same problem with and without an additional active GPS Antenna.

Both USRPs are connected via antennas.
I also tried cables with 6db attenuators (with different gain values). Same issue.

Sometimes the connection is very long intact but as soon as i ping the UE from the gNB or the other way around the connection crashes. 
Sometimes im able to ping but after about 20-40 seconds the connection also crashes.
Sometimes the connection does not even build up (see https://github.com/srsran/srsRAN_Project/issues/315)
I tried different gain-values on both sides, different cables, computers, USRPs etc.
Sometimes i can see: [PHY-SA ] [E] [ 5030] SYNC: detected out-of-sync... skipping slot ... in the UE log
Sometimes i can see:[DU-MNG  ] [I] ue=0: RLF detected. Timer of 31000 msec to release UE started... in the gNB log.

I have no idea, what i am supposed to try now, also which error i am getting or weither or not the UE connects at all seems to be random. Because if i don't change any configuration at all im still getting different errors from time to time.

Configuration and Logs (for the case when the UE actually connects, but releases the connection after a ping)
[UE_and_GNB_Logs_and_Confs.zip](https://github.com/srsran/srsRAN_Project/files/13491205/UE_and_GNB_Logs_and_Confs.zip)

Thanks in advance for your help.

oliverxsch, 2023-11-29:
Just as a side note, the configuration has to be correct, because with 2 B200 following this tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html its working.

pgawlowicz, 2023-11-29:
Could you try with the following parameters in the `ru_sdr` section:
```
  device_driver: uhd
  device_args: type=x300,addr=192.168.43.2,master_clock_rate=184.32e6,send_frame_size=8000,recv_frame_size=8000   
  clock: external
  sync: external                                                 
  srate: 15.36
```

Then for UE please change/add:
```
...
srate = 15.36e6
...
device_args = type=x300,addr=x.x.x.x,master_clock_rate=184.32e6,send_frame_size=8000,recv_frame_size=8000,clock=external,sync=external
...
[expert]
lte_sample_rates = true
```

Please run this script on both PCs: [srsran_performance](https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance)

oliverxsch, 2023-11-30:
Thank you so much, finally it works. Maybe add this tip to the tutorial on the website?

> Could you try with the following parameters in the `ru_sdr` section:
>
> ```
>   device_driver: uhd
>   device_args: [...]

Rinelli96, 2024-01-24:
Hi @pgawlowicz, we have exactly the same problem. Do you have any tips for us?

Our configuration consist of two URSP B210 connected via cables without attenuators. Their are synchonized with the same 10MHz external clock. The UE and GNB software run on Ubuntu 22.04.3 LTS.
The configuration files of GNB and UE respectively are attached 
[Config Files.zip](https://github.com/srsran/srsRAN_Project/files/14036466/Config.Files.zip)

pgawlowicz, 2024-01-24:
@Rinelli96 Does your configs work over zmq? Please check to be sure there are no non-RF related issues.

Then, could you check setting `time_adv_nsamples=600` in the srsUE config when running with USRPs?

Finally, I would recommend adding some attenuators (10-30dB) to the setup and tune TX/RX gains for both gnb and ue.

Rinelli96, 2024-01-24:
Thanks for your reply @pgawlowicz, 
1. we are checking the connection through ZMQ using the repo config files with the modification suggested in the doumentation guide. The GNB seems to work correctly, while the UE, when started, give us the following error:

Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_blade.so libsrsran_rf_zmq.so
Inactive RF plugins:
Reading configuration file /home/rinelli/Documents/Config_files/5G_Config/ue_zmq.conf...

Built in Release mode using commit eea87b1d8 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://[127.0.0.1:2001](http://127.0.0.1:2001/),rx_port=tcp://[127.0.0.1:2000](http://127.0.0.1:2000/),base_srate=11.52e6
Supported RF device list: UHD bladeRF zmq file
CHx base_srate=11.52e6
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://[127.0.0.1:2000](http://127.0.0.1:2000/)
CH0 tx_port=tcp://[127.0.0.1:2001](http://127.0.0.1:2001/)
Error: couldn't update sample rate. 23.04 is not divisible by 11.52
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)

2. Concerning the time_adv_nsamples, I think it is in some sense related to the delay on the communication channel. Can you provide me other informations about it (e.g. How it is related to the delay? Why it is set to 300 in the default configuration?, ...) ?.
3. What do you mean for TX/RX gain tuning? The problem can be caused by our higher gains?  

4. Just another question, Why you suggest to Oliver to set the "lte_sample_rates = true" in the UE config file and the "base_srate = 15.36" in both UE and GNB. We suppose that the srate issue have been fixed with the latest release ( base_srate = 23.04) do you confirm it ? 
Thank you for the support!!

Rinelli96, 2024-01-24:
1. I have already set the correct base_srate in the configurations: 
[Config.Files.zip](https://github.com/srsran/srsRAN_Project/files/14038524/Config.Files.zip).

2. Okay, thanks. I will try it!

3. So, do I need to check the SNR in the log file? Then, do I have to modify the TX/RX Gain for both GNB and UE to maximize the SNR. Is that correct?

4. OK

Thank you for your patience and support!

Rinelli96, 2024-01-24:
1. Note that the GNB Configuration file is a .yaml file format, while UE conf is a .conf file. I downloaded them directly from the [Tutorial](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup).

pgawlowicz, 2024-01-24:
i know, but you use `:` in srsue conf.

Rinelli96, 2024-01-24:
Yes! I included the wrong line in the zip folder. However, thank you for the support!! 
Regarding the Over-the-Air Setup[](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#over-the-air-setup), I have fixed the issue, and fine-tuning the parameters has solved the problem of connection instability. I aim to continue parameter tuning to maximize the pusch of the GNB and the snr in the UE, making the connection as reliable as possible.

pgawlowicz, 2024-01-24:
@Rinelli96 could you share the successful configs here?

Rinelli96, 2024-01-29:
Hi @pgawlowicz, here the configuration files working for us.
[config.zip](https://github.com/srsran/srsRAN_Project/files/14081761/config.zip)

pgawlowicz, 2024-01-29:
@Rinelli96 Great, thank you for your feedback!
I see the configuration is the *standard* one, except the tx/rx gains tuning for both gnb/srsUE.

Can you confirm that you get a stable connection and the srsUE, you can run ping/iperf and the srsUE keeps the connection (i.e., does not send a PRACH after a few seconds?)

Could you also print the console output on both gnb and srsUE (typing `t` after they start) and check the SNR levels for `pdsch` and `pusch` ?

pgawlowicz, 2025-01-20:
Could you try with a smaller BW? i am afraid that 1G link is too slow for the sampling rate of 30.72M

mohit4444, 2025-01-20:
@pgawlowicz the cell selection fails if i use 11.52 on both the UE and the gNB

[config1.zip](https://github.com/user-attachments/files/18478772/config1.zip)

mohit4444, 2025-01-20:
I removed it and i'm still getting the cell selection error

[conf.zip](https://github.com/user-attachments/files/18478967/conf.zip)

pgawlowicz, 2025-01-20:
seems the UE does not see the signal. Do you run over the air or over the cable? could you try to tune the rx/tx gains?

mohit4444, 2025-01-21:
I'm running it over the air. I was able to get the UE to select the cell with 11.52M but the 'PBCH-MIB: CRC failed' error still persists.

[confs.zip](https://github.com/user-attachments/files/18488760/confs.zip)

pgawlowicz, 2025-01-22:
please use `srate = 11.52e6` in srsUE config.

mohit4444, 2025-01-23:
Hi @pgawlowicz ,
Thank you! The UE has moved a little further in the process

[image]

The UE is not getting an ip address. I've tried changing the open 5gs env file to match the k and opc from the ue config file as well but the result is the same.

[config.zip](https://github.com/user-attachments/files/18518779/config.zip)

mohit4444, 2025-01-23:
Hi @pgawlowicz ,
Thank you! UE has completed the RRC connection but it's still not receiving an IP address.

[image]

[image]

[config.zip](https://github.com/user-attachments/files/18519422/config.zip)

pgawlowicz, 2025-01-24:
could you try to apply the patch from this comment to srsUE, recompile and try again? 
https://github.com/srsran/srsRAN_Project/issues/826#issuecomment-2370588502

mohit4444, 2025-01-26:
Hi @pgawlowicz ,
Thank you very much! I applied the patch, and the UE successfully connected and obtained an IP address.
However, the UE now disconnects on its own after around 10 seconds. Is there a way i can get it to stay connected?

[conf.zip](https://github.com/user-attachments/files/18550164/conf.zip)

Dipnarayan, 2025-08-01:
Hello, I am using srsRAN zmq where gnb is connected to AMF on  127.0.0.5:38412 
I am running open5gs, gnb and ue in same wsl2

**But the srsue is stuck with "Attaching UE"**
What can be the issue?

gnb.yml 

`
# cu_cp:
#   amf:
#     addr: 127.0.0.5       # FIX: Point to the AMF on the WSL IP.
#     bind_addr: 127.0.0.1   # FIX: Bind to the WSL IP.
#   inactivity_timer: 7200

# cu_up:
#   ngu:
#     socket:
#       - bind_addr: 127.0.0.1 # FIX: Bind the N3 interface to the WSL IP.

# ru_sdr:
#   device_driver: zmq
#   device_args: tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=11.52e6
#   srate: 11.52
#   tx_gain: 75
#   rx_gain: 75

# cell_cfg:
#   dl_arfcn: 368500
#   band: 3
#   channel_bandwidth_MHz: 10
#   common_scs: 15
#   plmn: "99970"
#   tac: 1
#   # ... rest of cell_cfg is the same ...
#   # pdcch:
#   #   common: {ss0_index: 0, coreset0_index: 12}
#   #   dedicated: {ss2_type: common, dci_format_0_1_and_1_1: false}
  
#   pdcch:
#     common:
#       ss0_index: 0                  # Set search space zero index to match srsUE capabilities
#       coreset0_index: 6             # Set search CORESET Zero index to match srsUE capabilities
#     dedicated:
#       ss2_type: common              # Search Space type, has to be set to common
#       dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
#   prach:
#     prach_config_index: 1
#   pdsch:
#     mcs_table: qam64
#   pusch:
#     mcs_table: qam64

# log:
#   filename: /tmp/gnb.log            # Path of the log file.
#   all_level: debug                   # Logging level applied to all layers.
#   hex_max_size: 0

# pcap:
#   mac_enable: true                 # Set to true to enable MAC-layer PCAPs.
#   mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
#   ngap_enable: true                # Set to true to enable NGAP PCAPs.
#   ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.

cu_cp:
  amf:
    addr: 127.0.0.5                 # The address or hostname of the AMF.
    port: 38412
    bind_addr: 127.0.0.2            # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas:
      - tac: 1
        plmn_list:
          - plmn: "99970"
            tai_slice_support_list:
              - sst: 1
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].

ru_sdr:
  device_driver: zmq                # The RF driver name.
  device_args: tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  srate: 23.04                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 20         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "99970"                     # PLMN broadcasted by the gNB.
  tac: 1                            # Tracking area code (needs to match the core configuration).
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 12            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE
  pdsch:
    mcs_table: qam64                # Sets PDSCH MCS to 64 QAM
  pusch:
    mcs_table: qam64                # Sets PUSCH MCS to 64 QAM

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
`

and ue_amq1.conf

`[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 23.04e6
nof_antennas = 1

device_name = zmq
device_args = tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6

[rat.eutra]
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 3
nof_carriers = 1
max_nof_prb = 106
nof_prb = 106

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
imsi = 999700000000001
k = 0C0A34601D4F07677303652C0462535B
opc = E8ED289DEBA952E4283B54E88E6183CA
imei = 358737060011223

[rrc]
release = 15
ue_category = 4

[nas]
apn = internet
apn_protocol = ipv4

[gw]
netns = ue1
ip_devname = tun_srsue
ip_netmask = 255.255.255.0

[gui]
enable = false
`