srsran/srsRAN_Project #357: Docker Compose Build and Pull issue

golisrikanth1989, 2023-12-06:
Hi all,

I am trying to setup the 5G SA end-end setup using dockers and successfully build Open5GS docker and srsRAN docker using respective Dockerfiles and also 5G SA network is also up and running. Trying to build and check the network metrics using the  metric-server, grafana and influxdb. But while building or running the docker compose -f docker-compose.yml up, I am getting below issue 
_[+] Running 1/1
 ! metrics-server Warning     3.1s 
[+] Building 0.0s (2/2) FINISHED     docker:default
 => [metrics-server internal] load .dockerignore    0.0s
 => => transferring context: 2B                              0.0s
 => [metrics-server internal] load build definition from Dockerfile     0.0s
 => => transferring dockerfile: 1.37kB         0.0s
failed to solve: no build stage in current context_

Setup Details: Setup is running on 12th Gen Intel(R) Core(TM) i7-12700 with 20 cores with ubunut 22.04.
followed these links
https://github.com/srsran/srsRAN_Project/tree/main/docker
https://github.com/srsran/srsRAN_Project/blob/main/docker/metrics_server/README.md 

Can anyone please provide any suggestions or documentation for the same?

Thanks
Srikanth

FredNys, 2023-12-12:
I experienced the same.

asaezper, 2023-12-12:
Hi!

If you take a look at the metrics_server/Dockerfile https://github.com/srsran/srsRAN_Project/blob/main/docker/metrics_server/Dockerfile you can see that the FROM line is missing.

It will be fixed in next push to main. Also metric reporting by the gnb is still in progress, which is required to use the metrics-server. I'll update this issue when both things are ready to use!

Thanks for reporting it!

golisrikanth1989, 2023-12-13:
Hi Alfredo,

Thanks for your reply! Besides Docker and Grafana, are there any other tools I should learn to fully understand this? Please recommend resources, as I'm very much interested in learning and contributing to automating and managing networks.

Best Regards,
Srikanth

closed 2023-12-15 by ismagom