srsran/srsRAN_Project #372: DPDK with OFH 100Mhz 4x2 Crash

IbrahimBabiker87, 2023-12-15:
## Issue Description ##
Srsran crash with DPDK OFH

## Setup Details ##
OFH with Benetel RAN650 CAT A

```
ru_ofh:
  ru_bandwidth_MHz: 100                                           # RU instantaneous bandwidth.
  t1a_max_cp_dl: 500                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 250                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 465                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 250                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 250                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 80
  ta4_max: 200                                                    # Maximum Ta4 on User-Plane in microseconds.
  ta4_min: 0                                                      # Minimum Ta4 on User-Plane in microseconds.
  is_prach_cp_enabled: false                                      # Configures if Control-Plane messages should be used to receive PRACH messages.
  compr_method_ul: bfp                                            # Uplink compression method.
  compr_bitwidth_ul: 9                                            # Uplink IQ samples bitwidth after compression.
  compr_method_dl: bfp                                            # Downlink compression method.
  compr_bitwidth_dl: 9                                            # Downlink IQ samples bitwidth after compression.
  compr_method_prach: bfp                                         # PRACH compression method.
  compr_bitwidth_prach: 9                                        # PRACH IQ samples bitwidth after compression.
  enable_ul_static_compr_hdr: true                                # Configures if the compression header is present for uplink User-Plane messages (false) or not present (true).
  enable_dl_static_compr_hdr: true                                # Configures if the compression header is present for downlink User-Plane messages (false) or not present (true).
  iq_scaling: 4                                             # IQ samples scaling factor applied before compression, should be a positive value smaller than 10.  
  cells:                                # Optional TEXT. Sets the hardware specific cell configuration on a per cell basis.
  - network_interface: enp1s0f0             # Optional TEXT (enp1s0f0). Sets the ethernet network interface name for the RU. Format: a string, e.g. [interface_name].
    ru_mac_addr: 70:b3:d5:e1:5d:3d       # Optional TEXT (70:b3:d5:e1:5b:06). Sets the RU MAC address. Format: a string, e.g. [AA:BB:CC:DD:11:22:33].
    du_mac_addr: 7c:c2:55:79:85:e8       # Optional TEXT (00:11:22:33:00:77). Sets the DU MAC address. Format: a string, e.g. [AA:BB:CC:DD:11:22:33].
    vlan_tag: 5                             # Optional UINT (1). Sets the V-LAN tag control information field. Supported: [1 - 4094].
    prach_port_id: [4,5]                   # Optional UINT (4). Sets the RU PRACH eAxC port ID. Supported: [0 - 65535].
    dl_port_id: [0,1,2,3]                   # Optional UINT (0, 1). Sets the RU downlink eAxC port ID. Format: vector containing all DL eXaC ports, e.g. [0, ...\ , N].
    ul_port_id: [0,1]                      # Optional UINT (0). Sets the RU uplink eAxC port ID. Supported: [0 - 65535].

hal:
  eal_args: -l 0-3 -n 4 --log-level lib.eal:debug                             # Optional TEXT. EAL configuration parameters used to initialize DPDK. 
```

## Expected Behavior ##
to work fine

## Actual Behaviour ##
```
/home/accelleran/srsRAN_Project/include/srsran/adt/bounded_bitset.h:1156: void srsran::bounded_bitset<N, LowestInfoBitIsMSB>::assert_within_bounds_(size_t, bool) const [with long unsigned int N = 1024; bool LowestInfoBitIsMSB = false; size_t = long unsigned int]: 
Assertion `pos < size() or (not strict and pos == size())' failed - ERROR: index='20' is out-of-bounds for bitset of size='1'.
Aborted

```
```
accelleran@acc-bm-du1:~/srsRAN_Project/scripts$ dpdk-devbind.py -s

Network devices using DPDK-compatible driver
============================================
0000:01:00.0 'Ethernet Controller X710 for 10GbE SFP+ 1572' drv=vfio-pci unused=i40e

Network devices using kernel driver
===================================
0000:01:00.1 'Ethernet Controller X710 for 10GbE SFP+ 1572' if=enp1s0f1 drv=i40e unused=vfio-pci 
0000:01:00.2 'Ethernet Controller X710 for 10GbE SFP+ 1572' if=enp1s0f2 drv=i40e unused=vfio-pci 
0000:01:00.3 'Ethernet Controller X710 for 10GbE SFP+ 1572' if=enp1s0f3 drv=i40e unused=vfio-pci *Active*
0000:07:00.0 'I210 Gigabit Network Connection 1533' if=eno1 drv=igb unused=vfio-pci 
0000:08:00.0 'I210 Gigabit Network Connection 1533' if=eno2 drv=igb unused=vfio-pci 

No 'Baseband' devices detected
==============================

Crypto devices using kernel driver
==================================
0000:0f:00.2 'VanGogh PSP/CCP 1649' drv=ccp unused=vfio-pci 

No 'DMA' devices detected
=========================

No 'Eventdev' devices detected
==============================

No 'Mempool' devices detected
=============================

No 'Compress' devices detected
==============================

No 'Misc (rawdev)' devices detected
===================================

No 'Regex' devices detected
===========================

No 'ML' devices detected
========================
```

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

sauka, 2024-01-08:
Hi @IbrahimBabiker87 

Thanks for reporting this. 
Are you using any affinity settings in `expert_execution` config? The problem seems to be related to your `eal_args` parameters limiting number of CPUs visible to the gNB to 1.

Could you compile the code with debug symbols and run it with gdb to provide more details?
To do so, pass `-DCMAKE_BUILD_TYPE=RelWithDebInfo` to the cmake command and recompile the gnb.
Then run it from gdb as follows:

`sudo gdb --args ./path/to/gnb -c /path/to/config.yml`
type `r`
when it stops, type `bt` and share the stack trace here. 
Thanks.

IbrahimBabiker87, 2024-01-08:
Hi @Sauka,

This is already solved by adding below in hal config:
hal:
  eal_args: "--lcores (0-1)@(0-31)"

closed 2024-01-08 by IbrahimBabiker87