srsran/srsRAN_Project #374: open5gs_5gc: Connection refused

WirelessRe, 2023-12-17:
## Issue Description ##
[[I am following the instruction for srsRAN gNB with srsUE here (https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html), on Ubuntu 22.04.3 LTSinstalled on a virtual box. After I run (sudo docker-compose up --build 5gc) from the folder (./srsRAN_Project/docker), I get the following:

## Actual Behaviour ##
[VirtualBox:~/srsRAN_Project_v2/docker$ sudo docker-compose up --build 5gc
WARNING: Some services (gnb) use the 'configs' key, which will be ignored. Compose does not support 'configs' configuration - use `docker stack deploy` to deploy to a swarm.
Building 5gc
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon   29.7kB
Step 1/20 : ARG OS_VERSION=22.04
Step 2/20 : FROM ubuntu:$OS_VERSION as base
 ---> 174c8c134b2a
Step 3/20 : ENV PYTHONBUFFERED=1
 ---> Using cache
 ---> 4c8587ef6c32
Step 4/20 : ENV DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> af19d19c2bbc
Step 5/20 : RUN DEBIAN_FRONTEND=noninteractive apt-get update     && apt install -y software-properties-common     && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> a373fab45024
Step 6/20 : RUN DEBIAN_FRONTEND=noninteractive apt-get update     && apt-get install -y     python3-pip     python3-setuptools     python3-wheel     ninja-build     build-essential     flex     bison     git     libsctp-dev     libgnutls28-dev     libgcrypt-dev     libssl-dev     libidn11-dev     libmongoc-dev     libbson-dev     libyaml-dev     libnghttp2-dev     libmicrohttpd-dev     libcurl4-gnutls-dev     libnghttp2-dev     libtins-dev     meson     curl     gettext     gdb     iproute2     iptables     iputils-ping     netcat     iperf     iperf3     libtalloc-dev     cmake     && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> f90c935e04f5
Step 7/20 : ARG MONGO_MAJOR_VERSION=6
 ---> Using cache
 ---> aaf71dc154d8
Step 8/20 : RUN DEBIAN_FRONTEND=noninteractive apt-get update && apt-get install -y --no-install-recommends wget gnupg     && wget -qO - https://www.mongodb.org/static/pgp/server-${MONGO_MAJOR_VERSION}.0.asc | apt-key add     && . /etc/os-release     && echo "deb [ arch=amd64,arm64 ] https://repo.mongodb.org/apt/ubuntu $UBUNTU_CODENAME/mongodb-org/${MONGO_MAJOR_VERSION}.0 multiverse" | tee /etc/apt/sources.list.d/mongodb-org-${MONGO_MAJOR_VERSION}.0.list     && DEBIAN_FRONTEND=noninteractive apt-get update && apt-get install -y --no-install-recommends mongodb-org     && apt-get autoremove && apt-get clean
 ---> Using cache
 ---> 0a31c5831b89
Step 9/20 : ARG OPEN5GS_VERSION=v2.6.1
 ---> Using cache
 ---> 9e24c5bc64f8
Step 10/20 : RUN echo $OPEN5GS_VERSION > ./open5gsversion
 ---> Using cache
 ---> 88e5f83f319c
Step 11/20 : RUN git clone --depth 1 --branch $(cat ./open5gsversion) https://github.com/open5gs/open5gs open5gs    && cd open5gs     && meson build --prefix=`pwd`/install     && ninja -C build     && cd build     && ninja install
 ---> Using cache
 ---> 9148fe67df99
Step 12/20 : ARG NODE_MAJOR=20
 ---> Using cache
 ---> bc9fccd50b5a
Step 13/20 : RUN curl -fsSL https://deb.nodesource.com/setup_${NODE_MAJOR}.x | bash -     && apt-get install -y nodejs     && cd open5gs/webui     && npm ci --no-optional
 ---> Using cache
 ---> 711527aa98e9
Step 14/20 : RUN python3 -m pip install pymongo click pyroute2 ipaddress python-iptables
 ---> Using cache
 ---> 3e24dd91e7cf
Step 15/20 : FROM base as open5gs
 ---> 3e24dd91e7cf
Step 16/20 : WORKDIR /open5gs
 ---> Using cache
 ---> f13efa9aedbe
Step 17/20 : COPY open5gs-5gc.yml open5gs-5gc.yml.in
 ---> Using cache
 ---> 843b74b85b10
Step 18/20 : COPY open5gs_entrypoint.sh add_users.py setup_tun.py subscriber_db.cs[v] ./
 ---> Using cache
 ---> 48da836f6852
Step 19/20 : ENV PATH="${PATH}:/open5gs/build/tests/app/"
 ---> Using cache
 ---> 17ab720a65ca
Step 20/20 : ENTRYPOINT [ "./open5gs_entrypoint.sh" ]
 ---> Using cache
 ---> 6449706cc82b
Successfully built 6449706cc82b
Successfully tagged docker_5gc:latest
Starting open5gs_5gc ... done
Attaching to open5gs_5gc
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | ./open5gs_entrypoint.sh: line 20:    71 Illegal instruction     (core dumped) mongod --logpath /tmp/mongodb.log
open5gs_5gc       | 
open5gs_5gc       | > open5gs@2.6.1 dev
open5gs_5gc       | > node server/index.js
open5gs_5gc       | 
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | > Using external babel configuration
open5gs_5gc       | > Location: "/open5gs/webui/.babelrc"
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       |  DONE  Compiled successfully in 2272ms5:58:04 PM
open5gs_5gc       | 
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | MongooseServerSelectionError: connect ECONNREFUSED 127.0.0.1:27017
open5gs_5gc       |     at NativeConnection.Connection.openUri (/open5gs/webui/node_modules/mongoose/lib/connection.js:824:32)
open5gs_5gc       |     at /open5gs/webui/node_modules/mongoose/lib/index.js:417:10
open5gs_5gc       |     at /open5gs/webui/node_modules/mongoose/lib/helpers/promiseOrCallback.js:41:5
open5gs_5gc       |     at new Promise (<anonymous>)
open5gs_5gc       |     at promiseOrCallback (/open5gs/webui/node_modules/mongoose/lib/helpers/promiseOrCallback.js:40:10)
open5gs_5gc       |     at Mongoose._promiseOrCallback (/open5gs/webui/node_modules/mongoose/lib/index.js:1270:10)
open5gs_5gc       |     at Mongoose.connect (/open5gs/webui/node_modules/mongoose/lib/index.js:416:20)
open5gs_5gc       |     at /open5gs/webui/server/index.js:39:29
open5gs_5gc       |     at Generator.next (<anonymous>)
open5gs_5gc       |     at onFulfilled (/open5gs/webui/node_modules/co/index.js:65:19)
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
open5gs_5gc       | nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
open5gs_5gc       | waiting for mongodb
]

How can I fix this issue?

WirelessRe, 2023-12-23:
I used another PC with a different CPU (intel i9) and the open5gs worked fine. I cloned the virtual box image that worked on the intel i9 cpu and installed it on my laptop with intel i7 cpu and I got the issue described above, where I get this message:
**open5gs_5gc | ./open5gs_entrypoint.sh: line 20: 71 Illegal instruction (core dumped) mongod --logpath /tmp/mongodb.log**

It seems that this issue is related to the hardware used. How can I fix this issue?

WirelessRe, 2023-12-28:
I found the solution where I enabled the avx in windows OS. 
Thank you to me.

closed 2023-12-28 by WirelessRe

raul-parada, 2024-03-25:
I'm trying to run Open5GS docker on an Ubuntu VM and I get this same error, I did enabled avx by executing this command bcdedit /set xsavedisable 1 (Windows 10), any clue?