srsran/srsRAN_Project #402: Help requested for UE bandwidth being throttled

rninan, 2024-01-12:
## Issue Description ##
[Describe the issue in detail] - We are trying to produce a simulation of 20 UEs on an eNb or 5G connected to an EPC or 5G core. In both these cases, we want to be able to get 142 Mbs on a 20 Mhz channel. However, we are not able to get more than single digit Mbps transfer. So we have some questions on where we can configure certain parameters, For example, where can we do PRBs, CQI, MCS, Bandwidth, Frequency (Band) and Power of the radio.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
20 UEs, 1 eNb, 1 EPC with iPerf running to test the transfer speed.

## Expected Behavior ##
[What you expect to happen]
Speed of close to 142 Mbps

## Actual Behaviour ##
[What happens instead e.g. error message]
4.6 Mpbs or lower.

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
Set up a 20 UEs connected to 1 eNb and 1 EPC.

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
When we turn of logging, speed increases marginally.

rninan, 2024-01-28:
Thanks for circling back. Here you go.
[configs.zip](https://github.com/srsran/srsRAN_Project/files/14074976/configs.zip)
[logs.zip](https://github.com/srsran/srsRAN_Project/files/14074977/logs.zip)
[configs.zip](https://github.com/srsran/srsRAN_Project/files/14074979/configs.zip)
[logs.zip](https://github.com/srsran/srsRAN_Project/files/14074980/logs.zip)

rninan, 2024-01-30:
@pgawlowicz Yes, we are running an LTE network. We were forced down that since we couldn't find a 5G core that worked and we needed a e2e 5G RAN/core for end to end services capability. So we had to default to LTE. Even with 5 Mhz, we should get ~31 Mbps and we are getting only 4-5 Mbps. Any other pointers to the problem? Much appreciated.

pgawlowicz, 2024-01-30:
Please follow this tutorial to set up an e2e 5g network: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html

rninan, 2024-02-12:
If we want to integrate srsRAN with emane (which doesn't support 5G yet, what choices do we have? Is there any consulting or technical service that can be available to us? Appreciate all the help.

frankist, 2024-02-12:
Hey @rninan,

>we couldn't find a 5G core that worked

What does this mean? There are several 5G open-source cores out there. We use open5gs often and can achieve >1.6Gbps with our gNB.

Regarding the logs above, you seem to not be pushing enough traffic to your 4G enb. Notice the DL buffer status of a UE seen by the scheduler:
>15:17:19.027278 [MAC    ] [I] [ 4067] SCHED: DL tx rnti=0x5a, cc=0, pid=0, mask=0x1000, dci=(0, 3), n_rtx=0, cfi=1, tbs=97, **buffer=94/0**, tti_tx_dl=4071
>15:17:19.027286 [MAC    ] [D] [ 4067] SCHED: DL MAC PDU payload cc=0:
>  \> rnti=0x5a, tbs=97, f=1, mcs=20: [SDU lcid=3, tb=0, len=95 B]

If not enough data is being pushed by the 4G core, you won't reach higher speeds.

rninan, 2024-02-13:
Thanks for the feedback. We want to use 5G srsRAN with emane for over the air interface. The open5G core doesn't provide the over the air interface. And emane only supports LTE.

pgawlowicz, 2024-02-14:
Sounds interesting. Note that you will need actually three different components, specifically Open5GS (core network), srsRAN_project (gNB) and srsRAN_4G (5G srsUE) to make it work. For the last one, you can probably reuse this repo: 
https://github.com/adjacentlink/srsRAN-emane

suttonpd, 2024-02-14:
> If we want to integrate srsRAN with emane (which doesn't support 5G yet, what choices do we have? Is there any consulting or technical service that can be available to us? Appreciate all the help.

Please reach out to info@srs.io for additional consulting or technical services.

closed 2024-02-29 by pgawlowicz