srsran/srsRAN_Project #417: Extract CSI information (both phase and amplitude)

HansJack1999, 2024-01-22:
Could someone please assist me in understanding if there's anyone familiar with the srsRAN project? I am seeking guidance on how to extract CSI information (both phase and amplitude) between a base station and a specified terminal during a 5G communication process from this project. Your help would be greatly appreciated.

frankist, 2024-02-06:
We already display as part of the console metrics the Rank Indicator (RI) and CQI. E.g.
```
==== gNodeB started ===
Type <t> to view trace
t

           -----------------DL-----------------------|------------------UL--------------------
 pci rnti  cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  mcs  brate   ok  nok  (%)    bsr
   1 4444   13   1   28    53M 1600    0   0%    10M |  99.9   28    12M  400    0   0%  81.5M
   1 4444   13   1   28    53M 1600    0   0%    10M |  99.9   28    12M  400    0   0%  81.5M
   1 4444   13   1   28    53M 1600    0   0%    10M |  99.9   28    12M  400    0   0%  81.5M
   1 4444   13   1   28    53M 1600    0   0%    10M |  99.9   28    12M  400    0   0%  81.5M
   1 4444   13   1   28    53M 1600    0   0%    10M |  99.9   28    12M  400    0   0%  81.5M
```

You can extend it to add the PMI. These metrics are also fed to external tools like grafana for GUI display.

closed 2024-02-29 by pgawlowicz

zhou-hao-ran521, 2025-07-19:
Hello, is there a specific guide or instructional method available?