srsran/srsRAN_Project #421: Srsran Crsh with Assertion `size() < MAX_N' failed - Cannot emplace back in full vector.

IbrahimBabiker87, 2024-01-22:
## Issue Description ##
==== gNodeB started ===                                                                               │
Type <t> to view trace                                                                                │
2024-01-22T18:53:37.329940 [GNB     ] [I] Starting Radio Unit...                                      │
2024-01-22T18:53:37.329944 [GNB     ] [I] Radio Unit started successfully                             │
/srsRAN_Project/include/srsran/adt/static_vector.h:281: constexpr T& srsran::static_ve│
ctor<T, MAX_N>::emplace_back(Args&& ...) [with Args = {srsran::ofh::uplane_section_params&}; T = srsra│
n::ofh::uplane_section_params; long unsigned int MAX_N = 2]:                                          │
Assertion `size() < MAX_N' failed - Cannot emplace back in full vector.                               │
Aborted                                                                                               │
admin@acc-bm-du1:~/srsRAN_Project/configs

## Setup Details ##
SrsRAN 23.10.1

## Expected Behavior ##
to work fine

## Actual Behaviour ##
Srsran Carshed with above error.

## Steps to reproduce the problem ##
was working fine, suddenly showed above error 

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

sauka, 2024-01-23:
Hi @IbrahimBabiker87 ,

Is it possible for you to send us a pcap showing uplink traffic?
Also could you attach your gnb config?

IbrahimBabiker87, 2024-01-23:
Hi @sauka ,

Its related to PRACH Compression enabling in RRU, used Benetel.
after enabled its working fine.

sauka, 2024-01-24:
Good that it is working! 

When it crashed, was the static compression configured in the gNB config?

IbrahimBabiker87, 2024-01-24:
it crash just after start up, 
Yes the compression was set on gnb config file BFP/9

sauka, 2024-01-25:
Hey @IbrahimBabiker87 

Thanks for details. It would be very helpful to have a `pcap` in order to correctly identify where the issue is. Could you capture traffic while using the configs  leading to the gNB crash?
Also the relevant configuration info (gNB and R550) would be helpful in debugging.
Thanks!

IbrahimBabiker87, 2024-01-25:
@sauka 

The crash was because of mismatch between config file and Benetel configuration:

in config file:
```
compr_method_prach: bfp              # Optional TEXT (none). Sets the PRACH compression method. Supported: [none, bfp, bfp selective, block scaling, mu law, modulation, modulation selective].
  compr_bitwidth_prach: 9              # Optional UINT (16). Sets the PRACH compression bit width. Supported [1 - 16]. 

```

while in Benetel configuration :
Prach Comperession was disabled:
`
Register value :  C0353 -x 0x0`
`

sauka, 2024-01-26:
Thanks for all the details @IbrahimBabiker87

closed 2024-01-26 by sauka