srsran/srsRAN_Project #423: srsGNB ERROR: CU-CP failed to connect to AMF

oliverxsch, 2024-01-23:
## Issue Description ##
Hi community,

I previously had a functional srsRAN setup, but when I attempted to use it again last week, the gNB failed to start. Upon investigation, I discovered that the open5gs-amf and mme daemons were not running, and I was unable to initiate them.

In an attempt to resolve the issue, I reinstalled open5gs, but unfortunately, this did not fix the problem. Subsequently, I installed open5gs from source. Currently, all daemons are running; however, I'm encountering difficulty starting the gNB.

I have attached the gnb config file. I followed the tutorial at the following link: https://open5gs.org/open5gs/docs/guide/02-building-open5gs-from-sources/. I followed all the steps outlined in the tutorial (excluding the addresses, which remained as 127.0.0.X).

Changes were made to tac, plm, etc.

Do you have any ideas on how to resolve this issue?

Thanks in advance.

Regards

This is the gnb console log:
-----------
sudo gnb -c /home/inets/Documents/gnbx410.yml

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in quad executor mode.

--== srsRAN gNB (commit 0b2702cca) ==--

Connecting to AMF on 127.0.0.5:38412
srsGNB ERROR: CU-CP failed to connect to AMF
-----------

Open5G:
-----------
01/23 14:21:37.839: [amf] INFO: gNB-N2 accepted[127.0.0.1]:45665 in ng-path module (../src/amf/ngap-sctp.c:113)
01/23 14:21:37.839: [amf] INFO: gNB-N2 accepted[127.0.0.1] in master_sm module (../src/amf/amf-sm.c:736)
01/23 14:21:37.842: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1235)
01/23 14:21:37.842: [amf] INFO: gNB-N2[127.0.0.1] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:775)
01/23 14:21:37.842: [amf] WARNING: NG-Setup failure: (../src/amf/ngap-handler.c:342)
01/23 14:21:37.842: [amf] WARNING:     Cannot find Served TAI. Check 'amf.tai' configuration (../src/amf/ngap-handler.c:343)
01/23 14:21:47.898: [amf] INFO: gNB-N2[127.0.0.1] connection refused!!! (../src/amf/amf-sm.c:788)
01/23 14:21:47.898: [amf] INFO: [Removed] Number of gNBs is now 0 (../src/amf/context.c:1262)
-----------
This is the whole gnb.log 
-----------
2024-01-23T13:21:37.837644 [GNB     ] [I] Built in Release mode using commit 0b2702cca on branch main
2024-01-23T13:21:37.838850 [ALL     ] [I] Task worker "phy_prach#0" started...
2024-01-23T13:21:37.839038 [ALL     ] [I] Task worker "ru_stats_worker" started...
2024-01-23T13:21:37.839064 [ALL     ] [I] Task worker "radio" started...
2024-01-23T13:21:37.839101 [ALL     ] [I] Task worker "lower_phy_rx#0" started...
2024-01-23T13:21:37.839120 [ALL     ] [I] Task worker "lower_phy_dl#0" started...
2024-01-23T13:21:37.839151 [ALL     ] [I] Task worker "lower_phy_ul#0" started...
2024-01-23T13:21:37.839162 [ALL     ] [I] Task worker "lower_phy_tx#0" started...
2024-01-23T13:21:37.839326 [GNB     ] [I] Connecting to AMF (127.0.0.5)..
2024-01-23T13:21:37.839412 [GNB     ] [I] AMF connection established
2024-01-23T13:21:37.839806 [GNB     ] [I] Starting CU-CP...
2024-01-23T13:21:37.839857 [NGAP    ] [I] Sending NgSetupRequest
2024-01-23T13:21:37.842512 [NGAP    ] [I] Reinitiating NG setup in 10000s (retry=1/5). Received NGSetupFailure with Time to Wait IE
2024-01-23T13:21:47.849918 [GNB     ] [I] CU-CP started successfully 
-----------

## Setup Details ##
First PC:
Specs: i9 13900, 64GB Ram
OS: Ubuntu 22.04.03 LTS
Newest srsRAN Project and newest Open5Gs.
Connected via a 10G-Link to a USRP X410
"gNB"

[gnbx410.yml.txt](https://github.com/srsran/srsRAN_Project/files/14024684/gnbx410.yml.txt)

hudouseko, 2024-01-23:
Hi, try to check on open5gs in amf configuration (/etc/openg5s/amf.yaml) if it contains appropriate TAI configuration corresponding to your gNB config:
amf:
    tai:
      - plmn_id:
          mcc: 001
          mnc: 01
        tac: 7

oliverxsch, 2024-01-23:
> Hi, try to check on open5gs in amf configuration (/etc/openg5s/amf.yaml) if it contains appropriate TAI configuration corresponding to your gNB config: amf: tai: - plmn_id: mcc: 001 mnc: 01 tac: 7

Thanks for your answer.
I already did this before. 
But I will atach all of my Open5Gs configs, if it helps.
[open5gsconfs.zip](https://github.com/srsran/srsRAN_Project/files/14026772/open5gsconfs.zip)

oliverxsch, 2024-01-24:
> did you restart the Open5GS demons whose configs were changed? Check here: https://open5gs.org/open5gs/docs/guide/01-quickstart/ Probably you need:
> 
> ```
> sudo systemctl restart open5gs-amfd
> ```

Hello,
i restarted all of them because its easy to do.
To start the daemons i simply used the third command:
_

> For convenience, you can execute all NFs at once by using the following command.
> 
> $ ./build/tests/app/5gc ## 5G Core Only with ./build/configs/sample.yaml
> $ ./build/tests/app/epc -c ./build/configs/srslte.yaml ## EPC Only with ./build/configs/srslte.yaml
> $ ./build/tests/app/app ## Both 5G Core and EPC with ./build/configs/sample.yaml

_
And to stop them i press STRG+C in that command window.

Your suggestion:

> sudo systemctl restart open5gs-amfd
Did not work for me (maybe because its installed from source?)

Using this command results in: 
Failed to restart open5gs-amfd.service: Unit open5gs-amfd.service is masked.

Anyways, restarting all daemons/or the host computer did not help.

pgawlowicz, 2024-01-24:
Could you change TAC to 1 in srsgnb config and see if it connects?

```
cell_cfg:
...
  tac: 1
```

oliverxsch, 2024-01-24:
> Could you change TAC to 1 in srsgnb config and see if it connects?
> 
> ```
> cell_cfg:
> ...
>   tac: 1
> ```

Problem and console output stays the same.

pgawlowicz, 2024-01-24:
which open5gs version do you use?
have you seen those answers:
https://github.com/srsran/srsRAN_Project/discussions/388#discussioncomment-8055894
https://github.com/open5gs/open5gs/discussions/2782#discussioncomment-8001181

oliverxsch, 2024-01-24:
> which open5gs version do you use? have you seen those answers: [#388 (comment)](https://github.com/srsran/srsRAN_Project/discussions/388#discussioncomment-8055894) [open5gs/open5gs#2782 (comment)](https://github.com/open5gs/open5gs/discussions/2782#discussioncomment-8001181)

I don't have an idea how to find out which version im using.
It looks like the usual commands do not work if i installed it from source.
But if it helps: I installed it yesterday from source, so it should be the newest version.

Unfortunaly the answers in the linked issue did not work for me.

pgawlowicz, 2024-01-24:
ok, then please check with our Open5gs docker image by following this tutorial:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup

please pay attention to this troubleshooting note:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#nd-open5gs-instance-installed-manually

If it works with Open5gs docker image, then there is an issue with your local setup/configs.

oliverxsch, 2024-01-24:
> ok, then please check with our Open5gs docker image by following this tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup
>
> please [...]

Okay, i will try.

But maybe this helps:

> X
> 01/24 14:14:31.248: [amf] INFO: gNB-N2 accepted[127.0.0.1]:52482 in ng-path module (../src/amf/ngap-sctp.c:113)
> 01/24 14:14:31.248: [amf] DEBUG: amf_state_operational(): AMF_EVENT_NGAP_LO_ACCEPT (../src/amf/amf-sm.c:78)
> 01/24 14:14:31.248: [amf] INFO: gNB-N2 accepted[127.0.0.1] in master_sm module (../src/amf/amf-sm.c:736)
> 01/24 14:14:31.252: [amf] DEBUG: ngap_state_initial(): ENTRY (../src/amf/ngap-sm.c:28)
> 01/24 14:14:31.252: [amf] DEBUG: ngap_state_operational(): ENTRY (../src/amf/ngap-sm.c:55)
> 01/24 14:14:31.252: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1235)
> 01/24 14:14:31.252: [amf] DEBUG: SCTP_ASSOC_CHANGE:[T:32769, F:0x0, S:0, I/O:10/30] (../src/amf/ngap-sctp.c:152)
> 01/24 14:14:31.252: [amf] DEBUG: SCTP_COMM_UP (../src/amf/ngap-sctp.c:161)
> 01/24 14:14:31.252: [amf] DEBUG: amf_state_operational(): AMF_EVENT_NGAP_LO_SCTP_COMM_UP (../src/amf/amf-sm.c:78)
> 01/24 14:14:31.252: [amf] INFO: gNB-N2[127.0.0.1] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:775)
> 01/24 14:14:31.252: [amf] DEBUG: amf_state_operational(): AMF_EVENT_NGAP_MESSAGE (../src/amf/amf-sm.c:78)
> 01/24 14:14:31.252: [amf] DEBUG: ngap_state_operational(): AMF_EVENT_NGAP_MESSAGE (../src/amf/ngap-sm.c:55)
> 01/24 14:14:31.252: [amf] DEBUG: NGSetupRequest (../src/amf/ngap-handler.c:144)
> 01/24 14:14:31.252: [amf] DEBUG:     IP[127.0.0.1] GNB_ID[0x19b] (../src/amf/ngap-handler.c:195)
> 01/24 14:14:31.252: [amf] DEBUG:     PLMN_ID[MCC:1 MNC:1] (../src/amf/ngap-handler.c:199)
> 01/24 14:14:31.252: [amf] DEBUG:     PagingDRX[3] (../src/amf/ngap-handler.c:203)
> 01/24 14:14:31.252: [amf] DEBUG:     TAC[7] (../src/amf/ngap-handler.c:227)
> 01/24 14:14:31.252: [amf] DEBUG:     PLMN_ID[MCC:1 MNC:1] (../src/amf/ngap-handler.c:256)
> 01/24 14:14:31.252: [amf] DEBUG:     S_NSSAI[SST:1 SD:0xffffff] (../src/amf/ngap-handler.c:296)
> 01/24 14:14:31.252: [amf] WARNING: NG-Setup failure: (../src/amf/ngap-handler.c:342)
> 01/24 14:14:31.252: [amf] WARNING:     Cannot find Served TAI. Check 'amf.tai' configuration (../src/amf/ngap-handler.c:343)
> 01/24 14:14:31.252: [amf] DEBUG: NG-Setup failure (../src/amf/ngap-path.c:285)
> 01/24 14:14:31.252: [amf] DEBUG:     Group[5] Cause[4] TimeToWait[3] (../src/amf/ngap-build.c:176)
> 01/24 14:14:31.252: [amf] DEBUG:     IP[127.0.0.1] RAN_ID[0] (../src/amf/ngap-path.c:69)
> 01/24 14:14:36.188: [pfcp] DEBUG: [1] REMOTE Holding Timeout for step 2 type 6 peer [127.0.0.4]:8805 (../lib/pfcp/xact.c:629)
> 01/24 14:14:36.188: [pfcp] DEBUG: [1] REMOTE Delete Transaction for step 2 type 6 peer [127.0.0.4]:8805 (../lib/pfcp/xact.c:642)
> 01/24 14:14:36.188: [pfcp] DEBUG: [1] REMOTE Delete  peer [127.0.0.4]:8805 (../lib/pfcp/xact.c:788)
> 01/24 14:14:36.226: [pfcp] DEBUG: [1] REMOTE Holding Timeout for step 2 type 6 peer [127.0.0.3]:8805 (../lib/pfcp/xact.c:629)
> 01/24 14:14:36.226: [pfcp] DEBUG: [1] REMOTE Delete Transaction for step 2 type 6 peer [127.0.0.3]:8805 (../lib/pfcp/xact.c:642)
> 01/24 14:14:36.226: [pfcp] DEBUG: [1] REMOTE Delete  peer [127.0.0.3]:8805 (../lib/pfcp/xact.c:788)
> 01/24 14:14:38.412: [scp] DEBUG: scp_state_operational(): OGS_EVENT_NAME_SBI_TIMER (../src/scp/scp-sm.c:51)
> 01/24 14:14:38.412: [sbi] DEBUG: ogs_sbi_nf_state_registered(): OGS_EVENT_NAME_SBI_TIMER (../lib/sbi/nf-sm.c:211)
> 01/24 14:14:38.413: [sbi] DEBUG: [PATCH] http://127.0.0.10:7777/nnrf-nfm/v1/nf-instances/80e1bdfc-baba-41ee-a7d2-d7d364bb35b4 (../lib/sbi/client.c:728)
> 01/24 14:14:38.413: [sbi] DEBUG: SENDING...[100] (../lib/sbi/client.c:497)
> 01/24 14:14:38.413: [sbi] DEBUG: [{"op":"replace","path":"/nfStatus","value":"REGISTERED"},{"op":"replace","path":"/load","value":0}] (../lib/sbi/client.c:499)
> 01/24 14:14:38.413: [sbi] DEBUG: STREAM added [81] (../lib/sbi/nghttp2-server.c:1542)
> 01/24 14:14:38.413: [sbi] DEBUG: [PATCH] /nnrf-nfm/v1/nf-instances/80e1bdfc-baba-41ee-a7d2-d7d364bb35b4 (../lib/sbi/nghttp2-server.c:1158)
> 01/24 14:14:38.413: [sbi] DEBUG: RECEIVED: 100 (../lib/sbi/nghttp2-server.c:1162)
> 01/24 14:14:38.413: [sbi] DEBUG: [{"op":"replace","path":"/nfStatus","value":"REGISTERED"},{"op":"replace","path":"/load","value":0}] (../lib/sbi/nghttp2-server.c:1164)
> 01/24 14:14:38.413: [nrf] DEBUG: nrf_state_operational(): OGS_EVENT_NAME_SBI_SERVER (../src/nrf/nrf-sm.c:52)
> 01/24 14:14:38.413: [nrf] DEBUG: nrf_nf_state_registered(): OGS_EVENT_NAME_SBI_SERVER (../src/nrf/nf-sm.c:183)
> 01/24 14:14:38.413: [sbi] DEBUG: STATUS [204] (../lib/sbi/nghttp2-server.c:640)
> 01/24 14:14:38.413: [sbi] DEBUG: STREAM closed [81] (../lib/sbi/nghttp2-server.c:1268)
> 0000: 00001a01 05000000 5189c161 96e4593e   ........Q..a..Y>
> 0010: 94134a65 1d4a0802 6940b370 2d5c65e5   ..Je.J..i@.p-\e.
> 0020: 3168df                                1h.
> 01/24 14:14:38.413: [sbi] DEBUG: [204:PATCH] http://127.0.0.10:7777/nnrf-nfm/v1/nf-instances/80e1bdfc-baba-41ee-a7d2-d7d364bb35b4 (../lib/sbi/client.c:675)
> 01/24 14:14:38.413: [sbi] DEBUG: RECEIVED[0] (../lib/sbi/client.c:686)
> 01/24 14:14:38.413: [scp] DEBUG: scp_state_operational(): OGS_EVENT_NAME_SBI_CLIENT (../src/scp/scp-sm.c:51)
> 01/24 14:14:38.413: [sbi] DEBUG: ogs_sbi_nf_state_registered(): OGS_EVENT_NAME_SBI_CLIENT (../lib/sbi/nf-sm.c:211)
[... cut ...]
> 01/24 14:14:41.288: [amf] DEBUG: amf_state_operational(): AMF_EVENT_NGAP_LO_CONNREFUSED (../src/amf/amf-sm.c:78)
> 01/24 14:14:41.288: [amf] INFO: gNB-N2[127.0.0.1] connection refused!!! (../src/amf/amf-sm.c:788)
> 01/24 14:14:41.288: [amf] DEBUG: ngap_state_operational(): EXIT (../src/amf/ngap-sm.c:55)
> 01/24 14:14:41.288: [amf] DEBUG: ngap_state_final(): EXIT (../src/amf/ngap-sm.c:37)
> 01/24 14:14:41.288: [amf] INFO: [Removed] Number of gNBs is now 0 (../src/amf/context.c:1262)
> 
> X

Its from the open5gs console running in debug mode.

oliverxsch, 2024-01-24:
> ok, then please check with our Open5gs docker image by following this tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup
>
> please [...]

I tried to follow the new tutorial. Weirdly im unable to start Open5Gs. This is the console output:
inets@inets-LAB:~/srsRAN_Project/docker$ sudo docker-compose up --build 5gc
[+] Building 0.2s (16/16) FINISHED                               docker:default
 => [5gc internal] load build definition from Dockerfile                   0.0s
 => => transferring dockerfile: 3.11kB                                     0.0s
 => [5gc internal] load .dockerignore                                      0.0s
 => => transferring context: 2B                                            0.0s
 => [5gc internal] load metadata for docker.io/library/ubuntu:22.04        0.2s
 => [5gc internal] load build context                                      0.0s
 => => transferring context: 14.70kB                                       0.0s
 => [5gc base 1/8] FROM docker.io/library/ubuntu:22.04@sha256:e6173d4dc55  0.0s
 => CACHED [5gc base 2/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 3/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 4/8] RUN DEBIAN_FRONTEND=noninteractive apt-get upda  0.0s
 => CACHED [5gc base 5/8] RUN echo v2.6.1 > ./open5gsversion               0.0s
 => CACHED [5gc base 6/8] RUN git clone --depth 1 --branch $(cat ./open5g  0.0s
 => CACHED [5gc base 7/8] RUN curl -fsSL https://deb.nodesource.com/setup  0.0s
 => CACHED [5gc base 8/8] RUN python3 -m pip install pymongo click pyrout  0.0s
 => CACHED [5gc open5gs 1/3] WORKDIR /open5gs                              0.0s
 => CACHED [5gc open5gs 2/3] COPY open5gs-5gc.yml open5gs-5gc.yml.in       0.0s
 => CACHED [5gc open5gs 3/3] COPY open5gs_entrypoint.sh add_users.py setu  0.0s
 => [5gc] exporting to image                                               0.0s
 => => exporting layers                                                    0.0s
 => => writing image sha256:783d786f8e7ba04b5f29cd51c1e7babbeba2119080775  0.0s
 => => naming to docker.io/library/docker-5gc                              0.0s
[+] Running 1/0
 ✘ Network docker_ran  Error                                               0.0s 
failed to create network docker_ran: Error response from daemon: Pool overlaps with other one on this address space

pgawlowicz, 2024-01-24:
probably, there is already a docker network that uses the same address space the open5gs wants to create.
[check here](https://github.com/srsran/srsRAN_Project/blob/main/docker/docker-compose.yml#L151-L156)

you need to remove the existing docker network or change the the config in docker-compose file. In the latter case, you need to update here: 
https://github.com/srsran/srsRAN_Project/blob/main/docker/open5gs/open5gs.env#L2

oliverxsch, 2024-01-24:
> ok, then please check with our Open5gs docker image by following this tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup
>
> please [...]

Thanks a lot!!!
First i was unable to install the docker container but after doing this:
https://github.com/maxking/docker-mailman/issues/85#issuecomment-349429246
I was able to get a step further.

Now im facing a new issue:
inets@inets-LAB:~/srsRAN_Project/build$ sudo gnb -c /home/inets/Documents/gnbx410.yml

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in quad executor mode.

--== srsRAN gNB (commit 0b2702cca) ==--

Connecting to AMF on 10.53.1.2:38412
Available radio types: uhd.
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; DPDK_21.11; UHD_4.5.0.HEAD-0-g471af98f
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=x4xx,addr=192.168.11.2,second_addr=192.168.10.2,master_clock_rate=245.76e6,send_frame_size=8000,recv_frame_size=8000'
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.11.2,type=x4xx,product=x410,serial=324B8F0,name=ni-x4xx-324B8F0,fpga=X4_200,claimed=False,addr=192.168.11.2,second_addr=192.168.10.2,master_clock_rate=245.76e6,send_frame_size=8000,recv_frame_size=8000
[INFO] [MPM.main] Launching USRP/MPM, version: 4.5.0.0-g471af98f
[INFO] [MPM.main] Spawning RPC process...
[INFO] [MPM.PeriphManager] Device serial number: 324524B
[INFO] [MPM.PeriphManager.ClkMgr] Using Clock Configuration:
DB0: Master Clock Rate: 245.76 MSps @Converter Rate 2.94912 GHz
DB1: Master Clock Rate: 245.76 MSps @Converter Rate 2.94912 GHz
[INFO] [MPM.PeriphManager] Initialized 2 daughterboard(s).
[INFO] [MPM.PeriphManager] init() called with device args `boot_init=True,clock_source=internal,time_source=internal,initializing=True'.
[INFO] [MPM.RPCServer] RPC server ready!
[INFO] [MPM.RPCServer] Spawning watchdog task...
[INFO] [MPM.PeriphManager] init() called with device args `fpga=X4_200,master_clock_rate=(245760000.0,),mgmt_addr=192.168.11.2,name=ni-x4xx-324B8F0,product=x410,recv_frame_size=8000,second_addr=192.168.10.2,send_frame_size=8000,clock_source=internal,time_source=internal,initializing=True'.
[INFO] [0/Radio#0] Clocking reconfigured, running ADC Self Cal on DB0...
[INFO] [0/Radio#0] Calibrating channel 0...
[INFO] [0/Radio#0] Calibrating channel 1...
[INFO] [0/Radio#1] Clocking reconfigured, running ADC Self Cal on DB1...
[INFO] [0/Radio#1] Calibrating channel 2...
[INFO] [0/Radio#1] Calibrating channel 3...
[WARNING] [MPM.PeriphManager.ClkMgr] Clock source internal is an invalid selection with time source external. Coercing clock source to external
[INFO] [MPM.PeriphManager.ClkMgr] Using Clock Configuration:
DB0: Master Clock Rate: 245.76 MSps @Converter Rate 2.94912 GHz
DB1: Master Clock Rate: 245.76 MSps @Converter Rate 2.94912 GHz
[INFO] [0/Radio#0] Clocking reconfigured, running ADC Self Cal on DB0...
[INFO] [0/Radio#0] Calibrating channel 0...
[INFO] [0/Radio#0] Calibrating channel 1...
[INFO] [0/Radio#1] Clocking reconfigured, running ADC Self Cal on DB1...
[INFO] [0/Radio#1] Calibrating channel 2...
[INFO] [0/Radio#1] Calibrating channel 3...
Error: setting Tx sampling rate. Tx Rate 23.04 MHz is invalid. The nearest valid value is 22.341818181818184.
srsGNB ERROR: Unable to create radio session.

Is the X410 not supported?

oliverxsch, 2024-01-24:
> you have to use:
> 
> ```
> ru_sdr:
> ...
>   srate: 30.72 
> ```

Thanks a lot for your answer.
It looks like im now in a situation simular to one of my old issues. 
However it looks like your fixes from back then do not solve my problems now. 
Im able to connect the UE to the gnb. Im also able to ping the gnb from the UE.
However i cannot start an iperf test or an speedtest using the ue. 
Also i cannot ping the UE from the gnb.
I did all the commands for the routing configuration from the tutorial.
Do you have an idea, why this does not work?
[image]
[image]
[gnbx410.yml (2).txt](https://github.com/srsran/srsRAN_Project/files/14039928/gnbx410.yml.2.txt)
[ue_x310.conf.txt](https://github.com/srsran/srsRAN_Project/files/14039929/ue_x310.conf.txt)

pgawlowicz, 2024-01-24:
could you check if setting `time_adv_nsamples = 600` in srsue conf helps?

oliverxsch, 2024-01-24:
> could you check if setting `time_adv_nsamples = 600` in srsue conf helps?

It does not help.
More to the contrary, it leads to the UE not connecting at all.
If I change it back to auto its like before.

Console Log:
inets@inets-LAB:~/srsRAN_4G/build$ sudo srsue /home/inets/Documents/ue_x310.conf
Active RF plugins: libsrsran_rf_uhd.so
Inactive RF plugins: 
Reading configuration file /home/inets/Documents/ue_x310.conf...

Built in Release mode using commit eea87b1d8 on branch master.

Opening 1 channels in RF device=uhd with args=type=x300,addr=192.168.43.2,second_addr=192.168.30.2,master_clock_rate=184.32e6,send_frame_size=8000,recv_frame_size=8000,clock=external,sync=external
Supported RF device list: UHD file
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.5.0.HEAD-0-g471af98f
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=x300,addr=192.168.43.2,second_addr=192.168.30.2,master_clock_rate=184.32e6,send_frame_size=8000,recv_frame_size=8000,sync=external
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.929a
[INFO] [X300] Radio 1x clock: 184.32 MHz
Setting manual TX/RX offset to 600 samples
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1134
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1294
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1454
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1614
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1774
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1934
RF status: O=0, U=0, L=6
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=2094
^CStopping ..
---  exiting  ---

pgawlowicz, 2024-01-24:
could you change the line in srsue config:
from  `device_args = clock=external`  to `device_args = clock_source=external`

and try with both `time_adv_nsamples=auto` and `time_adv_nsamples=300` ??

oliverxsch, 2024-01-24:
> could you change the line in srsue config: from `device_args = clock=external` to `device_args = clock_source=external`
> 
> and try with both `time_adv_nsamples=auto` and `time_adv_nsamples=300` ??

Both of these suggestion lead to this console output. Nothing happens, there aren't even any Random Access Transmission:
inets@inets-LAB:~/srsRAN_4G/build$ sudo srsue /home/inets/Documents/ue_x310.conf
Active RF plugins: libsrsran_rf_uhd.so
Inactive RF plugins: 
Reading configuration file /home/inets/Documents/ue_x310.conf...

Built in Release mode using commit eea87b1d8 on branch master.

Opening 1 channels in RF device=uhd with args=type=x300,addr=192.168.43.2,second_addr=192.168.30.2,master_clock_rate=184.32e6,send_frame_size=8000,recv_frame_size=8000,clock_source=external,sync=external
Supported RF device list: UHD file
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.5.0.HEAD-0-g471af98f
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=x300,addr=192.168.43.2,second_addr=192.168.30.2,master_clock_rate=184.32e6,send_frame_size=8000,recv_frame_size=8000,clock_source=external,sync=external
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.929a
[INFO] [X300] Radio 1x clock: 184.32 MHz
Waiting PHY to initialize ... done!
Attaching UE...

oliverxsch, 2024-01-24:
Might this be an issue because I use two different types of SDR for UE and GNB?
Because my UE is a X410 and my GNB is a X310.
I do not have access to another X410, but I could try another X310?

pgawlowicz, 2024-01-24:
no, it should not be an issue to run with two different USRPs

pgawlowicz, 2024-01-24:
could you go back to the last working configs (when the srsUE connects and gets IP address) and try with 10mhz bandwidth?

oliverxsch, 2024-01-25:
> could you go back to the last working configs (when the srsUE connects and gets IP address) and try with 10mhz bandwidth?

I tried, weirdly enough this doesn't work either.
After "Attaching UE" nothing happens.
I adjusted the values to 10 Mhz but i might not have done it correctly which is why i will attach the UE and gNB config.

[gnb_20Mhz_x410.yml.txt](https://github.com/srsran/srsRAN_Project/files/14050335/gnb_20Mhz_x410.yml.txt)
[ue_10Mhz_x310.conf.txt](https://github.com/srsran/srsRAN_Project/files/14050336/ue_10Mhz_x310.conf.txt)

pgawlowicz, 2024-01-25:
You need to run gnb with 10mhz bandwidth as well.

oliverxsch, 2024-01-25:
> You need to run gnb with 10mhz bandwidth as well.

Ups. I did. I just attached the wrong file ;)
I will be able to upload the file tomorrow.

mrouili, 2024-01-25:
> > you have to use:
> > ```
> > ru_sdr:
> > ...
> >   srate: 30.72 
> > ```
> 
> Thanks a lot for your answer. It looks like im now in a situation simular to one of my old issues. However it looks like your fixes from back then do not solve my problems now. Im able to connect the UE to the gnb. Im also able to ping the gnb from the UE. However i cannot start an iperf test or an speedtest using the ue. Also i cannot ping the UE from the gnb. I did all the commands for the routing configuration from the tutorial. Do you have an idea, why this does not work? ![image2](https://private-user-images.githubusercontent.com/29520143/299352510-36d5a6d6-7f1a-47e5-885f-097160047ce0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyMTY4MzUsIm5iZiI6MTcwNjIxNjUzNSwicGF0aCI6Ii8yOTUyMDE0My8yOTkzNTI1MTAtMzZkNWE2ZDYtN2YxYS00N2U1LTg4NWYtMDk3MTYwMDQ3Y2UwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI1VDIxMDIxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ1ZDgxMTYzOGI4MjIwZjkxNGZjZmVlYTM2MGY2OWI0OTY2NjhmNmVjYjM2NmFkNDI2YjZjODNjMWI1ZjkzZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tK0z665oYMJYo4pjVSUJsFVtt2GhbpRkoLvPgstDsEs) ![image](https://private-user-images.githubusercontent.com/29520143/299352520-366e4abe-17c3-400d-89ad-bdb4603d0243.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyMTY4MzUsIm5iZiI6MTcwNjIxNjUzNSwicGF0aCI6Ii8yOTUyMDE0My8yOTkzNTI1MjAtMzY2ZTRhYmUtMTdjMy00MDBkLTg5YWQtYmRiNDYwM2QwMjQzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI1VDIxMDIxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQxZTIxM2E4Nzg0MjgzZjUwMzQ1NWRlNDQ4ODJlY2MyZmI1MTdiMWYwNzg5OTY3N2YzMmEzMWFhNmFiY2Q2ODAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4k4mMHqEdDGBiXM52LXVfxdFHh5mbM-OtDLxMfWGR7Q) [gnbx410.yml (2).txt](https://github.com/srsran/srsRAN_Project/files/14039928/gnbx410.yml.2.txt) [ue_x310.conf.txt](https://github.com/srsran/srsRAN_Project/files/14039929/ue_x310.conf.txt)

Hi @oliverxsch and @pgawlowicz , I'd like to point out that a while back I have tried using the Ettus USRP x410 and I faced the exact same issues. The only difference is that I was using an actual UE (Pixel pro 7) instead of srsUE. With the  X310 everything works correctly but for some reason this happens when using an x410.

oliverxsch, 2024-01-26:
> > > you have to use:
> > > ```
> > > ru_sdr:
> > > ...
> > >   srate: 30.72 
> > > ```
> > 
> > 
> > Thanks a lot for your answer. It looks like im now in a situation simular to one of my old issues. However it looks like your fixes from back then do not solve my problems now. Im able to connect the UE to the gnb. Im also able to ping the gnb from the UE. However i cannot start an iperf test or an speedtest using the ue. Also i cannot ping the UE from the gnb. I did all the commands for the routing configuration from the tutorial. Do you have an idea, why this does not work? ![image2](https://private-user-images.githubusercontent.com/29520143/299352510-36d5a6d6-7f1a-47e5-885f-097160047ce0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyMTY4MzUsIm5iZiI6MTcwNjIxNjUzNSwicGF0aCI6Ii8yOTUyMDE0My8yOTkzNTI1MTAtMzZkNWE2ZDYtN2YxYS00N2U1LTg4NWYtMDk3MTYwMDQ3Y2UwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI1VDIxMDIxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ1ZDgxMTYzOGI4MjIwZjkxNGZjZmVlYTM2MGY2OWI0OTY2NjhmNmVjYjM2NmFkNDI2YjZjODNjMWI1ZjkzZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tK0z665oYMJYo4pjVSUJsFVtt2GhbpRkoLvPgstDsEs) ![image](https://private-user-images.githubusercontent.com/29520143/299352520-366e4abe-17c3-400d-89ad-bdb4603d0243.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyMTY4MzUsIm5iZiI6MTcwNjIxNjUzNSwicGF0aCI6Ii8yOTUyMDE0My8yOTkzNTI1MjAtMzY2ZTRhYmUtMTdjMy00MDBkLTg5YWQtYmRiNDYwM2QwMjQzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI1VDIxMDIxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQxZTIxM2E4Nzg0MjgzZjUwMzQ1NWRlNDQ4ODJlY2MyZmI1MTdiMWYwNzg5OTY3N2YzMmEzMWFhNmFiY2Q2ODAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4k4mMHqEdDGBiXM52LXVfxdFHh5mbM-OtDLxMfWGR7Q) [gnbx410.yml (2).txt](https://github.com/srsran/srsRAN_Project/files/14039928/gnbx410.yml.2.txt) [ue_x310.conf.txt](https://github.com/srsran/srsRAN_Project/files/14039929/ue_x310.conf.txt)
> 
> Hi @oliverxsch and @pgawlowicz , I'd like to point out that a while back I have tried using the Ettus USRP x410 and I faced the exact same issues. The only difference is that I was using an actual UE (Pixel 7 pro) instead of srsUE. With the X310 everything works correctly but for some reason this happens when using an x410.
> 
> @oliverxsch maybe try to use the X310 as the RU and x410 as UE ? this isn't a solution but just to see if the usage of the x410 as the RU is what causes this issue.

Okay, thanks for your suggestion.
Then I will try to use another X310 first to see if this solves the issue and if it does i will try the X410 as the UE.
I thought the X410 would lead to better results because with e.G. Open Air Interface it lead to better performance.

oliverxsch, 2024-01-26:
> You need to run gnb with 10mhz bandwidth as well.
[gnb_10Mhz_x410.yml.txt](https://github.com/srsran/srsRAN_Project/files/14064011/gnb_10Mhz_x410.yml.txt)

Here is the correct config :)

oliverxsch, 2024-01-26:
> Hi @oliverxsch and @pgawlowicz , I'd like to point out that a while back I have tried using the Ettus USRP x410 and I faced the exact same issues. The only difference is that I was using an actual [...]

Mh. 
The situation did not change.
Now both my gnb and ue both are X310s but the problem stays.
On 20MHz it connects but only ping from UE to gnb is possible.
On 10 Mhz it does not even connect.
These were my configs:
[ue and gnb confs.zip](https://github.com/srsran/srsRAN_Project/files/14064220/ue.and.gnb.confs.zip)
This is a screenshot from the unsuccesful ping from gnb to UE (on 20 MHz)
[image]
Also I tried to redo the network configuration from the tutorial but this error occured on the UE side:
[image]

Thanks in advance

pgawlowicz, 2024-01-26:
thanks for the configs, they look good except the gnb config for 10mhz, where you are missing:
```
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 6            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE
```

Please try to add it and see what happens, if there is any RACH attempt from the UE. I will try to reproduce your experiment in our local lab next week.

oliverxsch, 2024-01-26:
> thanks for the configs, they look good except the gnb config for 10mhz, where you are missing:
>
> ```
>   pdcch:
>     common:
>       ss0_index: 0                  # Set search space zero index to match [...]

This explains why 10 Mhz did not work at all. Now i have the same situation like with 20 Mhz. 
Only ping from UE to gnb is possible. If i ping the UE from the gnb, the same like in the screenshot above happens.

oliverxsch, 2024-01-30:
Today i tried to use a different x310 and different cables just in case anything was faulty.
But it still does not work.

pgawlowicz, 2024-01-31:
@oliverxsch I still have not managed to replicate your experiment here locally yet, I will do it this week. 
But could you quickly try to use the following values `{345, 800, 845}` for the `time_adv_nsamples` for your srsUE running on x310 with srate=30.72Mhz? And let me know the results.

oliverxsch, 2024-01-31:
> @oliverxsch I still have not managed to replicate your experiment here locally yet, I will do it this week. But could you quickly try to use the following values `{345, 800, 845}` for the `time_adv_nsamples` for your srsUE running on x310 with srate=30.72Mhz? And let me know the results.

Hi,
i tried to use your values.
These are the console output i got:
value 345 and 800:

Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=974
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1134
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1294
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1454
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1614
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1774
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1934 

value 845:

Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=3694
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=3854
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=4014
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=4174
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=4334
RF status: O=0, U=0, L=7
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=4494
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=4654

On auto it still connects but only ping from ue to gnb is possible but not the other way around.
By the way, i just realised that i never sended you the log and pcap files. I will attach them for the "auto" case.
[gnblog(1).txt](https://github.com/srsran/srsRAN_Project/files/14111213/gnblog.1.txt)
[pcapfiles.zip](https://github.com/srsran/srsRAN_Project/files/14111214/pcapfiles.zip)
[uelog(2).txt](https://github.com/srsran/srsRAN_Project/files/14111215/uelog.2.txt)

pgawlowicz, 2024-01-31:
ok, thank you! Then I need to check locally

oliverxsch, 2024-02-10:
Any new ideas?
I would very much appreciate it.

oliverxsch, 2024-02-19:
> ok, thank you! Then I need to check locally

> Hi @oliverxsch and @pgawlowicz , I'd like to point out that a while back I have tried using the Ettus USRP x410 and I faced the exact same issues. The only difference is that I was using an actual [...]

I closed this issue, because the current problem does not fit to the title.
https://github.com/srsran/srsRAN_Project/issues/480

closed 2024-02-19 by oliverxsch