srsran/srsRAN_Project #428: srsGNB not able to establish SCTP connection

meenaresh19, 2024-01-25:
## Issue Description ##
SCTP connection is timing out, COOKIE_ECHO is not reaching 5G core

Client (srsGNB) sent INIT to Core (AMF)   --> Detected in pcap of both client and core
Core responded with INIT_ACK(+cookie)  --> Detected in pcap of both client and core
Client sends COOKIE_ECHO                      --> Detected only in pcap of the client.

COOKIE_ECHO didnt reach AMF.

Same is tested with UERANSIM, SCTP connection is established successfully.                  

## Setup Details ##
srsRAN is installed on Ubuntu 22.04 (Core i7) [Client laptop]
5G Core: Open5GS deployed on EKS cluster in AWS cloud
Client laptop connected to AWS VPC using VPN (OpenVPN).
Able to ping IP of AMF node from client laptop.
SCTP port : 38412

## Expected Behavior ##
SCTP connection should be successful

## Actual Behaviour ##
SCTP connection timeout

## Steps to reproduce the problem ##
Connect srsGNB to open5GS core in AWS EKS cluster.

## Additional Information ##

meenaresh19, 2024-01-30:
If I connect to core in bare metal machine (separate laptop), its working fine. 
Not sure whats going wrong when I connect to core in AWS cloud

pgawlowicz, 2024-01-30:
Can you ping the AWS host from the gNB host? Maybe you need to add a routing entry.

meenaresh19, 2024-01-30:
Able to ping. Infact SCTP INIT reached core node, and core node responded with INIT ACK. After that SRS RAN node sent COOKIE_ECHO, which is not received by 5G core node.

pgawlowicz, 2024-01-31:
Do you bind the gnb to a proper IP address? here:
```
amf:
  addr: x.x.x.x                  # The address or hostname of the AMF.
  bind_addr: x.x.x.x
```

meenaresh19, 2024-01-31:
Yes, 
  addr : <AMF node IP address>
  bind_addr : Local machine IP address where srsRAN is running

pgawlowicz, 2024-02-01:
Is the `bind_addr :` an IP address of an interface connected to the internet?

dhiaboujebha, 2024-02-08:
> Is the `bind_addr :` an IP address of an interface connected to the internet?

Should it work, if we create a static ip address?

pgawlowicz, 2024-02-09:
Could you provide the NGAP PCAP file?

closed 2024-04-11 by pgawlowicz