srsran/srsRAN_Project #430: PRACH Short Sequence Gives Error: Assertion `symbol < nof_symbols' failed - Invalid symbol index.

IbrahimBabiker87, 2024-01-26:
## Issue Description ##
when using PRACH short codes below Error appeared after start up

==== gNodeB started ===                                                                               │
Type <t> to view trace                                                                                │
2024-01-26T12:50:45.049381 [GNB     ] [I] Starting Radio Unit...                                      │
2024-01-26T12:50:45.049386 [GNB     ] [I] Radio Unit started successfully                             │
/home/accelleran/srsRAN_Project/lib/ofh/receiver/../support/prach_context_repository.h:121: void srsran::ofh::prach_context::write_iq(unsigned int, unsigned int, unsigned int, srsran::span<const std::complex<float> >):                                                                                        
Assertion `symbol < nof_symbols' failed - Invalid symbol index.                                       
Aborted                                                                                               
accelleran@acc-bm-du1:~/srsRAN_Project/configs$       

## Setup Details ##
last compiled version of  Srsran 23.10.1
Benetel 1.0.4

Srsran Config:
```
is_prach_cp_enabled: 1               
compr_method_ul: bfp                  
compr_bitwidth_ul: 9                  
compr_method_dl: bfp                  
compr_bitwidth_dl: 9                  
compr_method_prach: bfp
compr_bitwidth_prach: 9             
```
Benetel Config:
short Sequence for Prach.
short Sequence Compression: both are trialed.
**static_compressed**
O-RU expects O-DU will send the C,U Plane packets without udCompHdr and
BFP9 compressed
**dynamic_compressed**
O-RU expects O-DU will send the C,U Plane packets with udCompHdr and
BFP9 compressed

## Actual Behaviour ##
Srsran Exited with below Error:
```
/home/accelleran/srsRAN_Project/lib/ofh/receiver/../support/prach_context_repository.h:121: void srsran::ofh::prach_context::write_iq(unsigned int, unsigned int, unsigned int, srsran::span<const std::complex<float> >):                                                                                        
Assertion `symbol < nof_symbols' failed - Invalid symbol index.                                       
Aborted                                                         

```

## Steps to reproduce the problem ##
setting Srsran and Benetel with above settings.

## Additional Information ##
Log:
```

2024-01-26T12:50:44.793823 [ALL     ] [I] Task worker "phy_prach#0" started...
2024-01-26T12:50:44.793914 [ALL     ] [I] Task worker "ru_timing" started...
2024-01-26T12:50:44.793990 [ALL     ] [I] Task worker "ru_tx_0" started...
2024-01-26T12:50:44.793996 [ALL     ] [I] Task worker "ru_rx_0" started...
2024-01-26T12:50:44.794283 [NGAP    ] [I] Sending NgSetupRequest
2024-01-26T12:50:44.894414 [UDP-GW  ] [I] UDP GW configured. rx_max_mmsg=256
2024-01-26T12:50:44.894795 [GTPU    ] [I] GTP-U demux. warn_on_drop=false
2024-01-26T12:50:44.894798 [GTPU    ] [I] Tunnel added. teid=0x000000
2024-01-26T12:50:44.894818 [CU-UP   ] [I] CU-UP starting...
2024-01-26T12:50:44.894838 [CU-CP   ] [I] Added CU-UP 0
2024-01-26T12:50:44.894851 [CU-UP-E1] [I] Sending E1SetupRequest
2024-01-26T12:50:44.894861 [CU-UP-E1] [I] "E1AP CU-UP Setup Procedure" finalized
2024-01-26T12:50:44.894866 [CU-UP   ] [I] CU-UP started successfully
2024-01-26T12:50:45.045212 [DU      ] [I] DU created successfully
2024-01-26T12:50:45.045218 [DU      ] [I] Starting DU...
2024-01-26T12:50:45.045219 [DU      ] [I] Starting DU-High...
2024-01-26T12:50:45.045220 [DU-MNG  ] [I] DU manager starting...
2024-01-26T12:50:45.045332 [CU-CP   ] [I] Added DU 0
[... 248 lines cut ...]
2024-01-26T12:50:45.049383 [OFH     ] [I] Starting the realtime timing worker
2024-01-26T12:50:45.049385 [OFH     ] [I] Starting the DPDK ethernet frame receiver
2024-01-26T12:50:45.054760 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054775 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054777 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054801 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054813 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054838 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054874 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054876 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054877 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054901 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054937 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054938 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054940 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054964 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.054999 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.055001 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.055009 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.055034 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.055060 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.055062 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size
2024-01-26T12:50:45.055076 [OFH     ] [I] Received Open Fronthaul message size is '2' bytes and is smaller than the section header size

```

sauka, 2024-03-19:
Hi @IbrahimBabiker87 ,

Could you try again your config and report whether you still see the issue?
Thanks

IbrahimBabiker87, 2024-03-19:
@sauka 
I couldn't regenerate the issue again

closed 2024-03-19 by sauka