srsran/srsRAN_Project #431: No PDU Session establishment between gNB (USRP n310) and UE (usrp x310), always show "received RRC Release"

peihaoY, 2024-01-26:
## Issue Description ##
When running the GNB and UE, an RRC connection is established, but no PDU session is established, therefore no ip address is assigned to the UE. 

## Setup Details ##
UE:
USRP X310
Ubuntu 22.04
UHD 4.4.0

GNB and Core:
USRP N310
Ubuntu 22.04
UHD 4.5.0

## Expected Behavior ##
[image]
PDU session establishment and ip address assignment like:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html

## Actual Behaviour ##
always receive "received RRC Release" 
[image]

## Steps to reproduce the problem ##
The config files can be found below:
[ue_rf.zip](https://github.com/srsran/srsRAN_Project/files/14067324/ue_rf.zip)
[gnb_rf_n310_fdd_n3_20mhz.zip](https://github.com/srsran/srsRAN_Project/files/14067325/gnb_rf_n310_fdd_n3_20mhz.zip)
[open5gs config.zip](https://github.com/srsran/srsRAN_Project/files/14067326/open5gs.config.zip)

The log files can be found below：
[ue.log](https://github.com/srsran/srsRAN_Project/files/14067306/ue.log)
[gnb.log](https://github.com/srsran/srsRAN_Project/files/14067315/gnb.log)
[open5gs log.zip](https://github.com/srsran/srsRAN_Project/files/14067320/open5gs.log.zip)

pgawlowicz, 2024-01-26:
Please check whether you populated UE database of Open5gc correctly as there is this issue inside `amf.log` file:

```
01/25 18:43:01.367: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
01/25 18:43:01.367: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2550)
01/25 18:43:01.367: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[9] TAC[7] CellID[0x19b0] (../src/amf/ngap-handler.c:562)
01/25 18:43:01.367: [amf] INFO: [suci-0-001-01-0000-0-0-0123456780] Unknown UE by SUCI (../src/amf/context.c:1835)
01/25 18:43:01.367: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1616)
01/25 18:43:01.367: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1165)
01/25 18:43:01.367: [gmm] INFO: [suci-0-001-01-0000-0-0-0123456780]    SUCI (../src/amf/gmm-handler.c:166)
01/25 18:43:01.368: [sbi] WARNING: [AUSF] (SCP-discover) NF has already been added [2db7c6ee-bbd2-41ee-abd7-a34361f57cbb:1] (../lib/sbi/path.c:216)
01/25 18:43:01.368: [gmm] WARNING: [suci-0-001-01-0000-0-0-0123456780] Cannot find SUCI [404] (../src/amf/gmm-sm.c:1672)
01/25 18:43:01.368: [amf] WARNING: [suci-0-001-01-0000-0-0-0123456780] Registration reject [11] (../src/amf/nas-path.c:219)
01/25 18:43:01.511: [amf] INFO: UE Context Release [Action:3] (../src/amf/ngap-handler.c:1698)
01/25 18:43:01.511: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[9] (../src/amf/ngap-handler.c:1699)
01/25 18:43:01.511: [amf] INFO:     SUCI[suci-0-001-01-0000-0-0-0123456780] (../src/amf/ngap-handler.c:1702)
01/25 18:43:01.511: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2557)
01/25 18:43:01.511: [amf] INFO: [Removed] Number of AMF-UEs is now 0 (../src/amf/context.c:1709)
```

peihaoY, 2024-01-26:
Could you tell me how to populated UE database of Open5gc. Is it MongodDB?
[mongod.log](https://github.com/srsran/srsRAN_Project/files/14068452/mongod.log)

pgawlowicz, 2024-01-26:
You can use the Open5gc webgui, check `Register Subscriber Information` section here: https://open5gs.org/open5gs/docs/guide/01-quickstart/

peihaoY, 2024-01-27:
There is no Register Subscriber in my webgui. And this is the UDR log details "Cannot find SUPI in DB".
[image]

pgawlowicz, 2024-01-27:
Please follow the steps presented here: https://www.sharetechnote.com/html/OpenRAN/OR_open5gs_webui.html

peihaoY, 2024-01-27:
I would like to know if my UE is USRP, do I need to manually add subscribe as well? Because when I try to use n310 as UE and x310 as gNB, subscribe is added automatically.

pgawlowicz, 2024-01-29:
If you follow our tutorial and run the Open5GC using the provided docker-compose.yml file, then the UE data is automatically added to the Open5GC database. If you installed Open5GC manually, then you need to add it manually as well.

closed 2024-01-30 by peihaoY

peihaoY, 2024-02-10:
> If you follow our tutorial and run the Open5GC using the provided docker-compose.yml file, then the UE data is automatically added to the Open5GC database. If you installed Open5GC manually, then you need to add it manually as well.

Hi [pgawlowicz](https://github.com/pgawlowicz), can I run the Open5GC using the provided docker-compose.yml file and use COTS UE? Would the data from COTS UE also be automatically added to the Open5GC database? Because I've noticed that everyone seems to be following a tutorial for manual addition.

pgawlowicz, 2024-02-10:
You can use the provided Open5gs docker, but you need to put the COTS UE (SIM card) data into its UE database. 

You can create a subscriber.csv file and put it there, check here https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#id17

peihaoY, 2024-02-12:
> You can use the provided Open5gs docker, but you need to put the COTS UE (SIM card) data into its UE database.
> 
> You can create a subscriber_db.csv file and put it there, check this tutorial on how to do it: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#id17
Hi [pgawlowicz](https://github.com/pgawlowicz), thanks for your assistance. I have completed communication between one UE and srsgNB, and they are working well. I am now setting up a system with multiple UEs to support my future research, and I am confident that using ZMQ and Amarisoft UE will not be a problem. My question is, can I add multiple COTS UEs or multiple USRP UEs, or both, to the open5gs docker database at the same time? I would like to know if this solution is supported by the srsRAN_Project, to avoid wasting time and ending up with failure.

pgawlowicz, 2024-02-12:
Yes, you can. Just add multiple entries in the subscriber_db.csv file