srsran/srsRAN_Project #435: Have trouble setup srsRAN gnb with COT UE (OnePlus Nord 5G)

qiqingh, 2024-01-28:
## Issue Description ##
Hello  

I followed the tutorial "srsRAN gNB with COTS UEs" in the link: https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html 

I am using a OnePluss Nord5G 8T. I observed that the UE is able to detect the RAN sometimes(very few), but it cannot attach.

## Setup Details ##
Machine: Intel(R) Core(TM) i7-1260P X16  Ubuntu 20.04.6 LTS 
srsRAN Project branch origin/main
B210 USRP
Open5GS 5G Core
OnePlus Nord 5G 8T KB2005
External GPSDO 
UHD 3.15.0

## Expected Behavior ##
I expect the UE to connect with the gNB

## Actual Behaviour ##
The UE doesn't start with the connection.

## Steps to reproduce the problem ##
I totally followed the tutorial, using the amf.yaml, upf.yaml and gnb_b210_20MHz_oneplus_8t.yml provided

## Additional Information ##
The screen I saw from the oneplus UE:
[image]
[image]
[image]

The gnb.log has following output:
```
2024-01-28T16:01:23.133105 [FAPI    ] [W] [   185.2] Real-time failure in FAPI: Received late DL_TTI.request from slot 185.1
2024-01-28T16:02:05.772048 [FAPI    ] [W] [  342.18] Real-time failure in FAPI: Received late UL_TTI.request from slot 342.17
2024-01-28T16:02:07.252069 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:02:17.518528 [FAPI    ] [W] [  493.10] Real-time failure in FAPI: Received late UL_TTI.request from slot 493.9
2024-01-28T16:02:18.429573 [FAPI    ] [W] [  584.11] Real-time failure in FAPI: Received late DL_TTI.request from slot 584.10
2024-01-28T16:02:18.429599 [FAPI    ] [W] [  584.11] Real-time failure in FAPI: Received TX_Data.request from slot 584.10
2024-01-28T16:02:18.429617 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:02:21.311991 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:02:22.254963 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:02:33.226006 [FAPI    ] [W] [    16.4] Real-time failure in FAPI: Received late DL_TTI.request from slot 16.3
2024-01-28T16:02:39.293195 [FAPI    ] [W] [  622.18] Real-time failure in FAPI: Received late UL_TTI.request from slot 622.17
2024-01-28T16:02:55.474450 [FAPI    ] [W] [  192.19] Real-time failure in FAPI: Received late UL_TTI.request from slot 192.18
2024-01-28T16:02:56.677285 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:02:57.254277 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:02:58.257362 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:02:58.257715 [FAPI    ] [W] [   471.4] Real-time failure in FAPI: Received late DL_TTI.request from slot 471.3
2024-01-28T16:02:58.257721 [FAPI    ] [W] [   471.5] Real-time failure in FAPI: Received late DL_TTI.request from slot 471.4
2024-01-28T16:02:58.483836 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:02:58.484402 [FAPI    ] [W] [  493.18] Real-time failure in FAPI: Received late DL_TTI.request from slot 493.16
2024-01-28T16:02:58.484415 [FAPI    ] [W] [  493.18] Real-time failure in FAPI: Received late UL_TTI.request from slot 493.17
2024-01-28T16:02:59.079863 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:02:59.080115 [FAPI    ] [W] [  553.10] Real-time failure in FAPI: Received late UL_TTI.request from slot 553.8
2024-01-28T16:02:59.080126 [FAPI    ] [W] [  553.10] Real-time failure in FAPI: Received late UL_TTI.request from slot 553.9
2024-01-28T16:02:59.255867 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:02:59.682467 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:03:00.008923 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:03:00.251962 [RF      ] [W] Real-time failure in RF: underflow
2024-01-28T16:03:32.707264 [Low-PHY#0] [W] Real-time failure in low-phy: PRACH request late for sector 0, slot 843.19 and start symbol 0.
```

yagoda, 2024-01-28:
Hi,

Have you ran the srsran_performance script?

qiqingh, 2024-01-28:
@yagoda yes, I do run this script.

meenaresh19, 2024-01-30:
Is 311490 configured by you in SRS RAN config file?
Doesn't look like. What is the PLMN (MCC and MNC) configured in SRS RAN config file?
Are you using configured ISIM?

qiqingh, 2024-01-31:
@meenaresh19 
Hello, may I know what is 311490? I use the config file: https://docs.srsran.com/projects/project/en/latest/_downloads/cc142413003e87e53c16e0b63203ac57/gnb_b210_20MHz_oneplus_8t.yml as mentioned in the tutorial 

The PLMN is 90170, And I use the 00101 for the sim card.

For the sim card, I used sysmoISIM-SJA2, I followed the instruction for the sim programming secrtion and also disabled the UST service 124 and 125.

May I know how to verify that the sim card is successfully programmed?

meenaresh19, 2024-02-01:
I believe 311 490 is T-Mobile network.
I think your mobile is not detecting the PLMN network of your SDR(srsRAN + B210), which is 90170.
I am also facing the same issue.

qiqingh, 2024-02-02:
@meenaresh19 Thanks for your illustration. I think maybe the "Real-time failure in RF: underflow" revealed in the gNB.log would be the issue.

pgawlowicz, 2024-02-29:
@qiqingh Hi, any progress/update on this issue?

ff187, 2024-03-01:
Dear @qiqingh ,

Please make sure you have configured the amf, upf with the following configuration: IMSI 00101, gNB PLMN 90170. According the COTS setup- OnePlus Nord 5G was tested in a shielded box. Moreover, few of the phones does not work well with TDD band. If you are working in a chamber then you can try some other FDD band instead just to make sure the network is visible to the phone. Thank you.

Moussa-Guemdani, 2024-05-10:
> @meenaresh19 Thanks for your illustration. I think maybe the "Real-time failure in RF: underflow" revealed in the gNB.log would be the issue.

Hello, Any solution for this issue?

Thank you