srsran/srsRAN_Project #439: The prb of pucch guardbands calculation is controversial.(build_pucch_guardbands_list)

lpg-delete, 2024-02-01:
Issue Description
build_pucch_guardbands_list(lib/scheduler/config/sched_cell_config_helpers.cpp);
The prb of pucch guardbands calculation is controversial;

Expected Behavior
prb range of pucch_f2_f3 is from pucch_res.starting_prb to pucch_res.starting_prb + variant_get<pucch_format_2_3_cfg>(pucch_res.format_params).nof_prbs;

Actual Behaviour
The quantity of all pucch res prb is 1;
res_0.prbs.set(pucch_res.starting_prb, pucch_res.starting_prb + 1);

carlo-gal, 2024-02-02:
Hey, could you please explain the issue a bit better, @lpg-delete ? 

Please, reference the code and indicate which part exactly you think is not correct.

lpg-delete, 2024-02-03:
In the function build_pucch_guardbands_list, line 54 res_0.prbs.set(pucch_res.starting_prb, pucch_res.starting_prb + 1); In my opinion, prb should be set according to circumstances. There is only one prb for pucch format1, but the prb occupied by pucch format2_3 should be not fixed. Here, the size of pucch resource in all cases is set to a fixed prb. On line 66, there should be a corresponding dispute about frequency hopping. @carlo-gal thank you for response.

carlo-gal, 2024-02-05:
Ok, I see the problem.

Basically, I need to update the PRBs for the case of the PUCCH format 2  (Format 3 is not supported), which should be taken from the configuration instead of being fixed to 1. We'll work on it and provide a fix in the next release.

As for line 66, I don't see anything wrong with how that is implemented. 
Each PUCCH resource maps to 1 or 2 PRBs_x_symbols elements on the resource grid, depending on whether frequency hopping is enabled.
Without frequency hopping, the PUCCH resource occupies 1 element
`[starting_PRB, starting_PRB + nof_PRBs) x [starting_symbol, starting_symbol + nof_symbols)` on the resource grid.

With frequency hopping enabled, the resource occupies 2 elements:
`[starting_PRB, starting_PRB + nof_PRBs) x [starting_symbol, starting_symbol + nof_symbols / 2)`
and 
`[starting_PRB, starting_PRB + nof_PRBs) x [starting_symbol + nof_symbols / 2, starting_symbol + nof_symbols)`

Each of these elements is added only once to the list.

Thank you for pointing out the problem.