srsran/srsRAN_Project #441: Error: UE context doesn't exist

aibtw, 2024-02-01:
## Issue Description ##
Getting an error: 
```
/home/ms/srsRAN_Project/lib/ngap/ngap_impl.cpp:564: void srsran::srs_cu_cp::ngap_impl::handle_ue_context_release_command(const ue_context_release_cmd_s&):
Assertion `ue != nullptr' failed - ue=1 ran_ue_id=0 amf_ue_id=1: UE for UE context doesn't exist.
```

## Setup Details ##
Software
- OS: Ubuntu 22.04
- srsRAN_Project commit 0b2702cca094a2a1d8f2d25d482172d2aa00d454
- open5gs commit e12b1be3137a3f94594b1be7ba5d51ca06ee2ab7
- MIMO (4x4 on N310), MIMO (4x2 RAN550)

Hardware
- USRP N310 / Benetel RAN550
- 32GB RAM
- Intel i9 13900K
- Multiple UEs (realme 9i 5G, Galaxy A14 5G, Galaxy A53 5G)

## Expected Behavior ##
When UE goes a little far, it should just normally disconnect from the network.

## Actual Behaviour ##
*specifically on 4x4 N310 and 4x2 RAN550*
The error shown above appears after few seconds to minutes of running the system and connecting a ue to it.

## Steps to reproduce the problem ##
on 4x4 and 4x2, this happens in my case no matter what. just run the system and it occurs.
The configurations we use are the defaults from srsRAN github, and the same configuration work fine when switching to 2x2 or SISO.

FabianEckermann, 2024-02-01:
Hi,

could you provide logs, PCAPs, and your configuration files for this?
You can check out https://docs.srsran.com/projects/project/en/latest/user_manuals/source/outputs.html for how to get logs and PCAPs.

FabianEckermann, 2024-02-19:
This should be solved by the latest release, closing this.

closed 2024-02-19 by FabianEckermann