srsran/srsRAN_Project #444: Cannot pin UE. The 5g core(n310) docker show "[upf] error: [drop] Source IP-4 Spoofing APN: srsapn" when I run"sudo ip route add default via 10.45.1.1 dev tun_srsue" on UE(x310))

peihaoY, 2024-02-01:
## Issue Description ##
I am trying this link, https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html.
The 5g core(n310) docker show 
"[upf] error: [drop] Source IP-4 Spoofing APN: srsapn SrcIF:0 DstIF:1 TEID:0xe (../src/upf/gtp-path.c:479)
[upf] error: SRC: 2309206A, UE: 0A2D0102 (../src/upf/gtp-path.c:481)" 
when I run"sudo ip route add default via 10.45.1.1 dev tun_srsue" on UE(x310))
And 5gcore cannot ping ue, ue can ping 5gcore.
## Setup Details ##
UE:
USRP X310
Ubuntu 22.04
UHD 4.4.0

GNB and Core:
USRP N310
Ubuntu 22.04
UHD 4.6.0

## Actual Behaviour ##
[image]

## Steps to reproduce the problem ##
UE iptables and route
[image]
[image]

5gcore iptables and route
[image]
[image]

## Additional Information ##
Actually, I already have 5gcore and gnb configured on the n310 and ue configured on the x310. But I wanted to try their function interchange, so I configured ue on n310 and 5gcore and gnb on x310 respectively, and there was a problem. I want to go back to the original Settings, and this problem occurs.

peihaoY, 2024-02-03:
[upf] error: 
SRC: 2309206A can translate to 35.9.32.106 which is the IP address of the UE computer host, UE: 0A2D0102  can translate to 10.45.1.2. That means the system received a packet with source IP 35.9.32.106, but was expecting the source IP to be 35.9.32.106?

pgawlowicz, 2024-02-03:
Could you check if you can ping from the Open5gs container? 

docker exec -it open5gs_5gc ping 10.45.1.2

peihaoY, 2024-02-03:
Do I need to add this rule "default via 10.45.1.1 dev tun_srsue" before check? Because as soon as I add, the open5g core starts spewing out "[upf] errors", instead of only throwing errors when pinging.

pgawlowicz, 2024-02-03:
This sets the default route, not needed if you only want to ping the core. But needed if you want to ping the host PC

peihaoY, 2024-02-03:
> Could you check if you can ping the UE from the Open5gs container?
> 
> docker exec -it open5gs_5gc ping 10.45.1.2
Should I go to cd srsRAN_Project/docker to run it? Sorry, I couldn't locate the position of the container.

pgawlowicz, 2024-02-03:
It should be possibile to run this command from anywhere. The Open5gs docker container must be running

peihaoY, 2024-02-03:
I have run "docker compose up --build 5gc", and it is running like this. However, "Error response from daemon: No such container open5gs_5g

pgawlowicz, 2024-02-03:
open5gs_5gc

Check the container name with 'docker ps'

peihaoY, 2024-02-03:
Now it works! sudo docker exec -it open5gs_5gc ping 10.45.1.2
[sudo] password for phyan:
PING 10.45.1.2 (10.45.1.2) 56(84) bytes of data.
64 bytes from 10.45.1.2: icmp_seq=1 ttl=64 time=39.2 ms
64 bytes from 10.45.1.2: icmp_seq=2 ttl=64 time=39.1 ms
64 bytes from 10.45.1.2: icmp_seq=3 ttl=64 time=38.0 ms
64 bytes from 10.45.1.2: icmp_seq=4 ttl=64 time=37.1 ms
64 bytes from 10.45.1.2: icmp_seq=5 ttl=64 time=35.0 ms
64 bytes from 10.45.1.2: icmp_seq=6 ttl=64 time=33.0 ms

Do I still need to address this issue? "[upf] error: [drop] Source IP-4 Spoofing APN. Do you have idea to solve this?

pgawlowicz, 2024-02-03:
Nice, so sth must be wrong with the routing on the host machine

pgawlowicz, 2024-02-03:
Btw, could you share your configs for x310 and n310?

peihaoY, 2024-02-03:
> Nice, so sth must be wrong with the routing on the host machine

Is it the routing of the host where I installed gNode and 5gcore?

pgawlowicz, 2024-02-03:
Yes, try to reboot the machine, remove docker container and setup all again

peihaoY, 2024-02-03:
> Btw, could you share your configs for x310 and n310?

Sure.
[gnb_ue_Config.zip](https://github.com/srsran/srsRAN_Project/files/14154285/gnb_ue_Config.zip)

peihaoY, 2024-02-03:
> Yes, try to reboot the machine, remove docker container and setup all again

Unfortunately, I tried doing this, but still cannot ping directly to the UE. It only works using "docker exec -it open5gs_5gc". Could you look at the routing Settings for me?
UE IP routing table
Destination     Gateway         Genmask         Flags  Metric   Ref    Use Iface
0.0.0.0            35.9.32.1           0.0.0.0              UG    600        0        0 wlp0s20f3
10.45.1.0         0.0.0.0             255.255.255.0   U       0            0        0 tun_srsue
35.9.32.0         0.0.0.0             255.255.255.0   U       600        0        0 wlp0s20f3
169.254.0.0     0.0.0.0             255.255.0.0       U       1000      0        0 enp1s0
172.17.0.0       0.0.0.0             255.255.0.0       U       0            0        0 docker0
172.19.1.0       0.0.0.0             255.255.255.0   U       0            0        0 br-6eb0ed8019ec
192.168.0.0     0.0.0.0             255.255.0.0       U       100        0        0 enp1s0
192.168.0.0     192.168.1.1     255.255.0.0       UG     200        0        0 enp1s0
gNB IP routing table
[image]

pgawlowicz, 2024-02-05:
The default route at UE goes over a WiFi interface. You need to remove first and add a new one going over `tun_srsue` interface.

peihaoY, 2024-02-08:
In fact, the same problem occurs when adding the tun_srsue interface after remove the first.

pgawlowicz, 2024-02-09:
did you add a default route going through `tun_srsue` interface?

peihaoY, 2024-02-09:
Yes.
[image]

pgawlowicz, 2024-02-09:
On srsue, there is a local route to 10.53.1.0 going to a bridge interface. Why is it so? The core Has IP 10.53.1.2, therefore if you ping from core the ping reply will be routed to the local bridge interface.

peihaoY, 2024-02-25:
Hi [pgawlowicz](https://github.com/pgawlowicz), thanks for your help! I don't know what happened to my system. I think my routing table is fine now. However now, the UE cannot ping the gNB (it used to be able to), and the gNB also cannot ping the UE (it used to work inside Docker). This is so strange; it was working fine last week. Could you help me take look? Their routing table is as follows:
gNB: 
[image]
UE:
[image]

peihaoY, 2024-02-26:
> the routing seems to be correct, do you see any traffic on gnb trace when pinging? could you post here the gnb console output after pressing `t` there?

Sure.
[image]

pgawlowicz, 2024-02-26:
hmm, so there is some traffic going over the wireless link. Could you reboot your machines and try to setup the routing again?

peihaoY, 2024-02-26:
Awesome! Thank you! I never thought reboot could solve this problem, even though my routing table configuration was correct.

closed 2024-02-26 by peihaoY