srsran/srsRAN_Project #449: UL TCP Resulted in UE Drop (RLF) with OFH (Benetel Radio) permenantly

IbrahimBabiker87, 2024-02-02:
## Issue Description ##
Initiating Uplink TCP will always result in  RLF and UE drop. 

## Setup Details ##
Last Build of Srsran 23.10.1
Benetel Radio 1.0.4

## Expected Behavior ##
To work fine

## Actual Behaviour ##
when initiating UDP traffic in UL it works fine, However when using TCP the UE drops,
this is permanent behavior with several UE models.

```
ad@intel-vm-amf:~$ sudo iperf3 -c 10.45.0.20  -p 5210  -t 86400  -b 100M -R -u
Connecting to host 10.45.0.20, port 5210
Reverse mode, remote host 10.45.0.20 is sending
[  5] local 10.45.0.1 port 54063 connected to 10.45.0.20 port 5210
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-1.00   sec  6.79 MBytes  57.0 Mbits/sec  0.139 ms  0/5285 (0%)  
[  5]   1.00-2.00   sec  7.37 MBytes  61.8 Mbits/sec  0.201 ms  0/5734 (0%)  
[  5]   2.00-3.00   sec  7.02 MBytes  58.9 Mbits/sec  0.172 ms  0/5463 (0%)  
[  5]   3.00-4.00   sec  7.21 MBytes  60.5 Mbits/sec  0.972 ms  0/5611 (0%)  
[  5]   4.00-5.00   sec  6.94 MBytes  58.2 Mbits/sec  0.199 ms  0/5398 (0%)  
[  5]   5.00-6.00   sec  7.00 MBytes  58.7 Mbits/sec  0.147 ms  0/5446 (0%)  
[  5]   6.00-7.00   sec  7.27 MBytes  61.0 Mbits/sec  0.235 ms  0/5655 (0%)  
[  5]   7.00-8.00   sec  7.09 MBytes  59.5 Mbits/sec  0.458 ms  0/5519 (0%)  
[  5]   8.00-9.00   sec  7.01 MBytes  58.8 Mbits/sec  0.166 ms  0/5453 (0%)  
[  5]   9.00-10.00  sec  7.04 MBytes  59.0 Mbits/sec  0.202 ms  0/5474 (0%)  
^C[  5]  10.00-10.93  sec  6.11 MBytes  55.3 Mbits/sec  0.203 ms  0/4749 (0%)  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.93  sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  sender
[  5]   0.00-10.93  sec  76.9 MBytes  59.0 Mbits/sec  0.203 ms  0/59787 (0%)  receiver
iperf3: interrupt - the client has terminated
ad@intel-vm-amf:~$ sudo iperf3 -c 10.45.0.20  -p 5210  -t 86400  -b 100M -R 
Connecting to host 10.45.0.20, port 5210
Reverse mode, remote host 10.45.0.20 is sending
[  5] local 10.45.0.1 port 35940 connected to 10.45.0.20 port 5210
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  13.2 KBytes   108 Kbits/sec                  
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec                  
^C[  5]   7.00-7.57   sec  0.00 Bytes  0.00 bits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-7.57   sec  0.00 Bytes  0.00 bits/sec                  sender
[  5]   0.00-7.57   sec  13.2 KBytes  14.2 Kbits/sec                  receiver
iperf3: interrupt - the client has terminated
```

## Steps to reproduce the problem ##
connect last build of Srsran 23.10.1 with benetel radio and initiate UL TCP.

## Additional Information ##

[gnb.zip](https://github.com/srsran/srsRAN_Project/files/14142976/gnb.zip)

ismagom, 2024-02-02:
It seems the UE stops receiving/transmisting as soon as the iperf starts. 

can you capture a PCAP? 

Can you start iperf from the UE? (instead of iperf3 in reverse mode)

thanks

IbrahimBabiker87, 2024-02-05:
@ismagom 
Here you go,
The behavior is the same when starting iperf from UE side,

[gnb_pcap.zip](https://github.com/srsran/srsRAN_Project/files/14162489/gnb_pcap.zip)

FredNys, 2024-02-05:
@ismagom This problem is seen using all radios Benetel, N310, B210.

alvasMan, 2024-02-06:
Hum... the very large SINR swing seems to happen around here:

```
2024-02-02T16:40:44.970602 [UL-PHY  ] [I] [  536.19] PUSCH: rnti=0x4601 h_id=0 prb=[4, 165) symb=[0, 14) mod=64QAM rv=0 tbs=14862 crc=OK iter=3.0 sinr=21.7dB t=425.7us uci_t=0.0us ret_t=199.8us
2024-02-02T16:40:44.980534 [UL-PHY  ] [I] [  537.18] PUSCH: rnti=0x4601 h_id=0 prb=[4, 165) symb=[0, 14) mod=64QAM rv=0 tbs=14862 crc=KO iter=10.0 ack=01 sinr=-29.2dB t=856.3us uci_t=122.1us ret_t=238.5us
```

I also see alot of these messages:
```
2024-02-02T16:40:44.969971 [OFH     ] [I] Dropped received Open Fronthaul User-Plane packet as decoded eAxC value '1' is not configured
2024-02-02T16:40:44.969979 [OFH     ] [I] Dropped received Open Fronthaul User-Plane packet as decoded eAxC value '1' is not configured
```

@IbrahimBabiker87 could you please share your full config? @AlaiaL Could you please advise @IbrahimBabiker87 with the dropped OFH packets?

AzeezEbrahim, 2024-02-06:
hey @IbrahimBabiker87  @ismagom 
i am facing same issue with Quectel UE (RM502Q-AE).
see it here #454

AlaiaL, 2024-02-06:
> Dropped received Open Fronthaul User-Plane packet as decoded eAxC value '1' is not configured

This usually happens when there are more `ul_port_id` in the OFH section than the number of rx antennas in the gNB configuration. If you want to get rid of these messages you could match these parameters in your configuration. For example for SISO:

```

ru_ofh:
  cells:
    - ul_port_id: [0]

cell_cfg:
  nof_antennas_dl: 1
  nof_antennas_ul: 1

```

IbrahimBabiker87, 2024-02-06:
Matching the ports in UL and DL didnt solve the issue ( below results with 2X2)
for SISO need to reconfigure the Benetel RU, but usually we use 2x2 at least
its good to know that its working fine with Nokia Xr20 and Samsung S22

```
ad@intel-vm-amf:~$ sudo iperf3 -c 10.45.0.20  -p 5210  -t 86400 -b 0.1G -u -R
Connecting to host 10.45.0.20, port 5210
Reverse mode, remote host 10.45.0.20 is sending
[  5] local 10.45.0.1 port 40003 connected to 10.45.0.20 port 5210
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-1.00   sec  5.92 MBytes  49.6 Mbits/sec  0.054 ms  0/4602 (0%)  
[  5]   1.00-2.00   sec  5.91 MBytes  49.6 Mbits/sec  0.141 ms  0/4598 (0%)  
[  5]   2.00-3.00   sec  5.93 MBytes  49.8 Mbits/sec  0.115 ms  0/4616 (0%)  
[  5]   3.00-4.00   sec  6.01 MBytes  50.4 Mbits/sec  0.212 ms  0/4675 (0%)  
[  5]   4.00-5.00   sec  6.06 MBytes  50.8 Mbits/sec  0.253 ms  0/4715 (0%)  
[  5]   5.00-6.00   sec  5.84 MBytes  49.0 Mbits/sec  0.284 ms  0/4544 (0%)  
[  5]   6.00-7.00   sec  6.16 MBytes  51.7 Mbits/sec  0.390 ms  0/4794 (0%)  
[  5]   7.00-8.00   sec  5.93 MBytes  49.8 Mbits/sec  0.270 ms  0/4614 (0%)  
[  5]   8.00-9.00   sec  6.07 MBytes  50.9 Mbits/sec  0.231 ms  0/4720 (0%)  
[  5]   9.00-10.00  sec  5.92 MBytes  49.7 Mbits/sec  0.463 ms  0/4607 (0%)  
[  5]  10.00-11.00  sec  6.02 MBytes  50.5 Mbits/sec  0.255 ms  0/4680 (0%)  
[  5]  11.00-12.00  sec  5.98 MBytes  50.2 Mbits/sec  0.223 ms  0/4651 (0%)  
[  5]  12.00-13.00  sec  6.05 MBytes  50.7 Mbits/sec  0.213 ms  0/4705 (0%)  
[  5]  13.00-14.00  sec  6.14 MBytes  51.5 Mbits/sec  1.596 ms  0/4775 (0%)  
[  5]  14.00-15.00  sec  6.18 MBytes  51.9 Mbits/sec  0.198 ms  0/4811 (0%)  
[  5]  15.00-16.00  sec  5.75 MBytes  48.2 Mbits/sec  0.243 ms  0/4473 (0%)  
[  5]  16.00-17.00  sec  6.15 MBytes  51.6 Mbits/sec  1.113 ms  0/4781 (0%)  
[  5]  17.00-18.00  sec  6.10 MBytes  51.1 Mbits/sec  0.605 ms  0/4742 (0%)  
[  5]  18.00-19.00  sec  5.98 MBytes  50.2 Mbits/sec  0.290 ms  0/4654 (0%)  
[  5]  19.00-20.00  sec  6.14 MBytes  51.5 Mbits/sec  1.027 ms  0/4779 (0%)  
[  5]  20.00-21.00  sec  6.20 MBytes  52.0 Mbits/sec  0.052 ms  0/4820 (0%)  
[  5]  21.00-22.00  sec  6.28 MBytes  52.6 Mbits/sec  0.510 ms  0/4882 (0%)  
[  5]  22.00-23.00  sec  5.96 MBytes  50.0 Mbits/sec  0.216 ms  0/4633 (0%)  
[  5]  23.00-24.00  sec  6.25 MBytes  52.5 Mbits/sec  0.186 ms  0/4865 (0%)  
[  5]  24.00-25.00  sec  5.93 MBytes  49.7 Mbits/sec  0.352 ms  0/4613 (0%)  
[  5]  25.00-26.00  sec  6.16 MBytes  51.7 Mbits/sec  0.383 ms  0/4793 (0%)  
[  5]  26.00-27.00  sec  6.23 MBytes  52.3 Mbits/sec  0.135 ms  0/4847 (0%)  
[  5]  27.00-28.00  sec  6.21 MBytes  52.1 Mbits/sec  0.186 ms  0/4828 (0%)  
[  5]  28.00-29.00  sec  6.09 MBytes  51.1 Mbits/sec  0.184 ms  0/4740 (0%)  
^C[  5]  29.00-29.66  sec  4.04 MBytes  51.1 Mbits/sec  0.090 ms  0/3143 (0%)  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-29.66  sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  sender
[  5]   0.00-29.66  sec   180 MBytes  50.8 Mbits/sec  0.090 ms  0/139700 (0%)  receiver
iperf3: interrupt - the client has terminated
ad@intel-vm-amf:~$ sudo iperf3 -c 10.45.0.20  -p 5210  -t 86400 -b 0.1G  -R
Connecting to host 10.45.0.20, port 5210
Reverse mode, remote host 10.45.0.20 is sending
[  5] local 10.45.0.1 port 57344 connected to 10.45.0.20 port 5210
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  13.2 KBytes   108 Kbits/sec                  
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec                  
^C[  5]   4.00-4.93   sec  0.00 Bytes  0.00 bits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-4.93   sec  0.00 Bytes  0.00 bits/sec                  sender
[  5]   0.00-4.93   sec  13.2 KBytes  21.9 Kbits/sec                  receiver
iperf3: interrupt - the client has terminated
```

config:

```
ru_ofh:
  ru_bandwidth_MHz: 100                                           # RU instantaneous bandwidth.
  t1a_max_cp_dl: 470                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 419                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 336                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
[... 16 lines cut ...]
  iq_scaling: 8                                      # IQ samples scaling factor applied before compression, should be a positive value smaller than 10.  
  cells:                                # Optional TEXT. Sets the hardware specific cell configuration on a per cell basis.
  - network_interface: 0000:01:00.0             # Optional TEXT (enp1s0f0). Sets the ethernet network interface name for the RU. Format: a string, e.g. [interface_name].
    ru_mac_addr: 70:b3:d5:e1:5d:3d       # Optional TEXT (70:b3:d5:e1:5b:06). Sets the RU MAC address. Format: a string, e.g. [AA:BB:CC:DD:11:22:33].
    du_mac_addr: 7c:c2:55:69:f5:88       # Optional TEXT (00:11:22:33:00:77). Sets the DU MAC address. Format: a string, e.g. [AA:BB:CC:DD:11:22:33].
[... cut ...]
  plmn: "00101"                                                   # PLMN broadcasted by the gNB.
  tac: 1                                                          # Tracking area code (needs to match the core configuration).
  pci: 1                                                          # Physical cell ID. 
  nof_antennas_dl: 2                                              # Downlink number of antennas.
  nof_antennas_ul: 2                                              # Uplink number of anntennas.

```

robertfalkenberg, 2024-03-15:
Hi @IbrahimBabiker87,

I have tried to reproduce locally using a USRP B210 and a COTS Telit FN990A28 module, but UL iperf3 with TCP didn't have any issues - here are the runs of two consecutive iperf3 runs, each for 10s:

50MHz 1x1:
```
-== srsRAN gNB (commit 59008378e6) ==--

Connecting to AMF on 10.53.1.2:38412
Available radio types: uhd and zmq.
[INFO] [UHD] linux; GNU C++ version 13.2.1 20230801; Boost_108300; UHD_4.6.0.0-0-unknown
Making USRP object with args 'type=b200'
[INFO] [LOGGING] Fastpath logging disabled at runtime.
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 61.440000 MHz... 
[INFO] [B200] Actually got clock rate 61.440000 MHz.
[WARNING] [AD936X] Selected Tx sample rate (61.44 MHz) is greater than
analog frontend filter bandwidth (56 MHz).
[WARNING] [AD936X] Selected Rx sample rate (61.44 MHz) is greater than
analog frontend filter bandwidth (56 MHz).
Cell pci=1, bw=50 MHz, dl_arfcn=710000 (n79), dl_freq=4650.0 MHz, dl_ssb_arfcn=709536, ul_freq=4650.0 MHz

==== gNodeB started ===
Type <t> to view trace
t

           -----------------DL-----------------------|------------------UL--------------------
 pci rnti  cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  mcs  brate   ok  nok  (%)    bsr
   1 4601   15   1   25    27k   25    7  21%      0 |  19.9   27   161k   17   10  37%      0
   1 4601   15   1   20   302k  196   67  25%      0 |  15.5   24    10M  171   36  17%  77.3k
   1 4601   15   1   19   422k  291   79  21%      0 |  14.8   24    17M  242   51  17%   300k
   1 4601   15   1   19   945k  612  105  14%    290 |  15.1   24    31M  425   67  13%   300k
   1 4601   15   1   19   1.2M  714  141  16%    348 |  15.2   24    38M  500   82  14%   300k
   1 4601   15   1   19   1.3M  769  131  14%      0 |  15.0   23    38M  513   86  14%   300k
   1 4601   15   1   19   1.1M  632  134  17%    525 |  14.8   21    35M  509   93  15%   300k
Late: 0; Underflow: 3; Overflow: 0;
   1 4601   15   1   19   1.2M  765  121  13%      0 |  15.9   24    40M  524   75  12%   300k
   1 4601   15   1   19   1.2M  732  140  16%    174 |  15.9   24    40M  527   73  12%   300k
   1 4601   15   1   19   1.2M  685  137  16%      0 |  14.8   24    39M  505   95  15%   300k
Late: 24; Underflow: 10; Overflow: 1;
   1 4601   15   1   19   1.3M  827  119  12%    640 |  15.4   24    40M  519   81  13%   300k

           -----------------DL-----------------------|------------------UL--------------------
 pci rnti  cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  mcs  brate   ok  nok  (%)    bsr
   1 4601   15   1   19   139k   70   15  17%      0 |  19.3   25   3.0M   42    2   4%      0
   1 4601   15   1    0      0    0    0   0%      0 |   n/a    0      0    0    0   0%      0
   1 4601   15   1    0      0    0    0   0%      0 |   n/a    0      0    0    0   0%      0
   1 4601   15   1    0      0    0    0   0%      0 |   n/a    0      0    0    0   0%      0
   1 4601   15   1    0      0    0    0   0%      0 |   n/a    0      0    0    0   0%      0
   1 4602   15   1   25   162k   95   17  15%      0 |  17.6   27   3.9M   64   14  17%  77.3k
   1 4602   15   1   19   707k  490   86  14%     16 |  16.6   26    25M  314   44  12%   150k
   1 4602   15   1   19   1.2M  781   84   9%    290 |  16.0   23    37M  501   74  12%   300k
Late: 0; Underflow: 3; Overflow: 0;
   1 4602   15   1   19   1.4M  894  116  11%    116 |  17.5   25    44M  549   52   8%   300k

           -----------------DL-----------------------|------------------UL--------------------
 pci rnti  cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  mcs  brate   ok  nok  (%)    bsr
   1 4602   15   1   19   1.5M  880  111  11%    174 |  20.2   26    48M  572   26   4%   300k
   1 4602   15   1   19   1.5M  969   96   9%    795 |  19.9   26    49M  576   26   4%   300k
[... 83 lines cut ...]
   1 4602    9   2    8    576    1    0   0%      0 |  30.8   28   4.4k    1    0   0%      0
   1 4602    8   2   10   1.3k    2    0   0%      0 |  30.1   28   8.7k    2    0   0%      0
   1 4602    7   2    0      0    0    0   0%      0 |   n/a    0      0    0    0   0%      0
   1 4602    8   2    9   543k  286    3   1%    406 |  26.7   28    15M  306    0   0%   108k
   1 4602    8   2   10   1.0M  557    6   1%    580 |  26.3   28    31M  600    0   0%   300k

           -----------------DL-----------------------|------------------UL--------------------
 pci rnti  cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  mcs  brate   ok  nok  (%)    bsr
   1 4602    8   2   10   1.1M  532    5   0%    522 |  26.4   28    31M  600    0   0%   300k
   1 4602    9   2   10   1.0M  542   13   2%    580 |  26.2   28    31M  600    0   0%   300k
   1 4602    8   2    9   973k  539    7   1%    232 |  26.3   28    30M  599    0   0%   300k
   1 4602    8   2    9   981k  542    6   1%    290 |  26.2   28    31M  601    0   0%   300k
[... cut ...]
   1 4602    8   2    9   952k  531    5   0%    406 |  26.3   28    31M  600    0   0%   300k
   1 4602    9   2    9   643k  371    3   0%      0 |  26.1   28    20M  400    0   0%      0
   1 4602    8   2    9    640    1    0   0%      0 |  30.6   28   4.4k    1    0   0%      0
   1 4602    9   2    9    640    1    0   0%      0 |  29.9   28   4.4k    1    0   0%      0
```

Here are both configs that I used:
[gnb_rf_b200_tdd.tar.gz](https://github.com/srsran/srsRAN_Project/files/14615708/gnb_rf_b200_tdd.tar.gz)

I'll follow-up with for Quectel RM500Q shortly.

IbrahimBabiker87, 2024-03-19:
Hello @robertfalkenberg ,

Thank you for your feedback, The problem appeared in Qectel 500Q , and One Pluse

ismagom, 2024-04-04:
Hi @IbrahimBabiker87  , we were able to reproduce the issue and provided a fix. Could you test if: 

https://github.com/srsran/srsRAN_Project/tree/fix_disconnect

fixes it for you? 

FYI, the problem seemed to be we were using the extended size bit in the MAC header to indicate the size of a PDU which didn't need it.

IbrahimBabiker87, 2024-04-08:
@ismagom 

Thank you its solved now in fix_disconnect branch

closed 2024-04-08 by IbrahimBabiker87