srsran/srsRAN_Project #451: srsUE establishes RRC connection and then gets released & clock synchronization doubt

sourya1712, 2024-02-03:
I am running the open5gs core and the gNB on the same machine and it connects properly.

I am running the srsUE on a different machine. Each side has a USRP b200 connected to it. So the gNB is running on an USRP b200 and the UE is running on another USRP b200. I am using internal clocks of the USRPs.

Everytime I run the UE, there is an RRC connection made and about 3 seconds later, the RRC connection is released. I have played around with the time_adv_samples and Tx and Rx gain but to no avail. The UE is unable to maintain the communication and is unstable.

I have attached the config files for the UE and gNB and the logs for the UE and gNB as well. Help will really be appreciated.

Another Question is: How are the clocks synchronized if no GPSDO is used? srsRAN_Project tutorial page suggests the use of an external LeoBodnar GPSDO. Is it necessary to use one?
[ue.log](https://github.com/srsran/srsRAN_Project/files/14152647/ue.log)
[gnb.log](https://github.com/srsran/srsRAN_Project/files/14152649/gnb.log)
[ue.txt](https://github.com/srsran/srsRAN_Project/files/14152655/ue.txt)
[gnb config.txt](https://github.com/srsran/srsRAN_Project/files/14152656/gnb.config.txt)

pgawlowicz, 2024-02-03:
Does your setup work if you replace the USRPs with the ZMQ-based RF emulation? Could you provide Open5gs logs?

sourya1712, 2024-02-03:
> Does your setup work if you replace the USRPs with the ZMQ-based RF emulation? Could you provide Open5gs logs?

Yes. my setup works with ZMQ. It even establishes a PDU session with ZMQ.
Please find attached all the logs of the modules of Open5gs core.
[amf.txt](https://github.com/srsran/srsRAN_Project/files/14152705/amf.txt)
[ausf.txt](https://github.com/srsran/srsRAN_Project/files/14152706/ausf.txt)
[bsf.txt](https://github.com/srsran/srsRAN_Project/files/14152707/bsf.txt)
[nrf.txt](https://github.com/srsran/srsRAN_Project/files/14152708/nrf.txt)
[nssf.txt](https://github.com/srsran/srsRAN_Project/files/14152709/nssf.txt)
[pcf.txt](https://github.com/srsran/srsRAN_Project/files/14152710/pcf.txt)
[pcrf.txt](https://github.com/srsran/srsRAN_Project/files/14152711/pcrf.txt)
[scp.txt](https://github.com/srsran/srsRAN_Project/files/14152712/scp.txt)
[smf.txt](https://github.com/srsran/srsRAN_Project/files/14152713/smf.txt)
[udm.txt](https://github.com/srsran/srsRAN_Project/files/14152714/udm.txt)
[udr.txt](https://github.com/srsran/srsRAN_Project/files/14152715/udr.txt)
[upf.txt](https://github.com/srsran/srsRAN_Project/files/14152716/upf.txt)

pgawlowicz, 2024-02-03:
In the amf logs you have: 

'''
02/03 12:22:47.435: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
02/03 12:22:47.435: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2550)
02/03 12:22:47.435: [amf] INFO:     RAN_UE_NGAP_ID[1] AMF_UE_NGAP_ID[1] TAC[7] CellID[0x19b0] (../src/amf/ngap-handler.c:562)
02/03 12:22:47.435: [amf] INFO: [suci-0-001-01-0000-0-0-0123456780] Unknown UE by SUCI (../src/amf/context.c:1835)
02/03 12:22:47.435: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1616)
02/03 12:22:47.435: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1165)
02/03 12:22:47.435: [gmm] INFO: [suci-0-001-01-0000-0-0-0123456780]    SUCI (../src/amf/gmm-handler.c:166)
02/03 12:22:48.375: [gmm] ERROR: [suci-0-001-01-0000-0-0-0123456780] HTTP response error [500] (../src/amf/gmm-sm.c:1675)
02/03 12:22:48.375: [amf] WARNING: [suci-0-001-01-0000-0-0-0123456780] Registration reject [9] (../src/amf/nas-path.c:219)
02/03 12:22:48.497: [amf] INFO: UE Context Release [Action:3] (../src/amf/ngap-handler.c:1698)
''' 

So the core sends the UE context release. Did you installed the Open5gs manualy or do you use the provided dockerized version?

pgawlowicz, 2024-02-03:
On the other hand, the RX and TX gains seems to be very high. Could you try tx 60 and RX 40 for both UE and gnb

pgawlowicz, 2024-02-03:
time_adv_nsamples between 300-320, or you can try auto

sourya1712, 2024-02-03:
> In the amf logs you have:
>
> ''' 02/03 12:22:47.435: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401) 02/03 12:22:47.435: [amf] INFO: [Added] Number of gNB-UEs is now 1 [...]

I installed it from Source using the instructions on open5gs page.

sourya1712, 2024-02-03:
> On the other hand, the RX and TX gains seems to be very high. Could you try tx 60 and RX 40 for both UE and gnb

I will try and get back to you.

sourya1712, 2024-02-03:
> time_adv_nsamples between 300-320, or you can try auto

Is there an option for auto as well?

sourya1712, 2024-02-03:
> In the amf logs you have:
>
> ''' 02/03 12:22:47.435: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401) 02/03 12:22:47.435: [amf] INFO: [Added] Number of gNB-UEs is now 1 [...]

Why would the server reject registration?
It is returning code 500. I did not make any changes to the UE data. Why would it reject registration?

pgawlowicz, 2024-02-03:
There is info 'Unknown UE by SUCI' so the UE is not present in the Open5gs subscriber database. You need to add it

sourya1712, 2024-02-04:
> There is info 'Unknown UE by SUCI' so the UE is not present in the Open5gs subscriber database. You need to add it

I will add it via the GUI and see if that works.

sourya1712, 2024-02-05:
> There is info 'Unknown UE by SUCI' so the UE is not present in the Open5gs subscriber database. You need to add it

Hi. I just noticed that the UE was already present in the database. I removed it and added it again. The following is the info for the UE:
[image]
 The UE still gets an RRC Release and the AMF still signals an unknown UE:
[image]

sourya1712, 2024-02-07:
@pgawlowicz do you have any solution for the above problem?

pgawlowicz, 2024-02-07:
gNB connects to the core, and the UE tries to connect, so the RAN part works correctly. It seems that there is some Open5gs related issue. Probably sth is misconfgured there.

sourya1712, 2024-02-07:
> gNB connects to the core, and the UE tries to connect, so the RAN part works correctly. It seems that there is some Open5gs related issue. Probably sth is misconfgured there.
> 
> I would recommend to use the provided docker version of the open5gs

where can I find the docker version?

sourya1712, 2024-02-07:
> See the tutorial https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html

So it is already provided in your srsRAN_Project package as far as I can make out.

pgawlowicz, 2024-02-09:
Yes, you need to start it from srsRAN_Project/docker directory  :
```
cd ./srsRAN_Project/docker
docker compose up --build 5gc
```

sourya1712, 2024-02-10:
> Yes, you need to start it from srsRAN_Project/docker directory :
> 
> ```
> cd ./srsRAN_Project/docker
> docker compose up --build 5gc
> ```

I did a fresh install of the entire project. The UE now connects to the gNB and maintains connection for sometime before getting an RRC Release again. I have attached the Core Log below:
[amflog.txt](https://github.com/srsran/srsRAN_Project/files/14230029/amflog.txt)

pgawlowicz, 2024-02-29:
@sourya1712 did you have any progress with this issue? Does the UE keep the connection when you start pinging it?

pgawlowicz, 2024-02-29:
you might also check if this solution works for you https://github.com/srsran/srsRAN_Project/discussions/135#discussioncomment-6521094

sourya1712, 2024-02-29:
> you might also check if this solution works for you [#135 (reply in thread)](https://github.com/srsran/srsRAN_Project/discussions/135#discussioncomment-6521094)

I will try changing the apn to srsapn on the UE cpnfig file as well as the UE record in the open5gs database. Let me see if that works.

achkonsa, 2024-03-19:
hello, i meet the same question in amf terminal: [gmm] ERROR: [suci-0-001-01-0000-0-0-0123456780] HTTP response error, but my error code is [400], not [500]. I wanna ask if you have any progress :)

sourya1712, 2024-03-20:
> hello, i meet the same question in amf terminal: [gmm] ERROR: [suci-0-001-01-0000-0-0-0123456780] HTTP response error, but my error code is [400], not [500]. I wanna ask if you have any progress :)

Hi,
I changed the apn and did everything as instructed. The connection is still unstable. It lasts for around 5 seconds before disconnecting.

pgawlowicz, 2024-03-20:
Could you attach the gnb logs?

achkonsa, 2024-03-20:
Hello @pgawlowicz,

First of all, thank you for sharing this project. I encountered a similar issue as @sourya1712 . I've been following the discussion between you and @sourya1712 regarding the issue we're facing with this project. I saw that you requested the gnb log to further diagnose the problem. I'm encountering a similar issue and thought it might be helpful to share my gnb log as well.

Attached you'll find my gnb log
[gnblog.txt](https://github.com/srsran/srsRAN_Project/files/14664845/gnblog.txt)
. I hope this can provide some additional insights into this problem. And I have noticed that the gnb log sometimes become too large to open. Now it is over 5M bytes.

Thank you again for your help, and I look forward to your reply.

Best regards,
achkonsa

pgawlowicz, 2024-03-20:
could you run srsran_performance script?

mohammadih, 2024-03-28:
I do have the same issue
[image]
[image]

achkonsa, 2024-03-29:
I have run the srsran_performance script, but it is still the same. @pgawlowicz

achkonsa, 2024-03-29:
I recommend that you carefully review the configuration files data or try using an older version of Open5GS, such as 2.6.6 or even earlier version. Yesterday, I rebuilt this project on another computer, and it was successful. However, it remains the same on my original server. @mohammadih

pgawlowicz, 2024-04-02:
@mohammadih could you share gnb logs and ngap pcap file?

achkonsa, 2024-04-03:
Hi, sorry to disturb. It still remains the same on my server. I have tried the docker-open5gs and open5gs. UE is always gets released after 5 lines gnb output. I doubt that it is related to the gnb or ue.
[image]

achkonsa, 2024-04-12:
OK!! Thank you soooooo much for caring and asking!!! It works!!

pgawlowicz, 2024-04-12:
great! Would you be willing to share your final configs and shortly describe the Tx/Tx gains tuning? It would be very helpful for other users.

closed 2024-05-14 by pgawlowicz