srsran/srsRAN_Project #458: SrsRAN gnb Will freeze if cpu isolation is used

IbrahimBabiker87, 2024-02-07:
## Issue Description ##
With latest commit a09b000 isolated_cpus has no effect,  the gnb will hang as soon as started , the passed isolated cpus will not be taken into account, only the first CPU will be picked up causing hanging 

## Setup Details ##
Srsran With latest commit a09b000 isolated_cpus 
Setup:
` isolated_cpus: 4,5,6,7,8,9,10,11,12,13,14,15,20,21,22,23,24,25,26,27,28,29,30,31`

The 4th cpu will be overwhelmed with 100%. 

## Expected Behavior ##
to work fine

## Actual Behaviour ##
gnb hanging

below behavior appeared just before hanging, seems the gnb will not recognize the isolated cpu at first place
ad@bm-super4:~/srsRAN_Project/configs$ sudo gnb -c cell_config.yml 
**srsGNB ERROR: Invalid CPU core selected '4'. Valid CPU ids: 0 1 2 3 16 17 18 19 in the 'isolated_cpus' property**

## Steps to reproduce the problem ##
start the gnb with the latest commit  a09b000

## Additional Information ##
[image]

ismagom, 2024-02-07:
Can you share you cell_config.yml so we can reproduce?

IbrahimBabiker87, 2024-02-07:
HYG
[cell_config.txt](https://github.com/srsran/srsRAN_Project/files/14192734/cell_config.txt)

sauka, 2024-02-12:
Hi @IbrahimBabiker87 ,

thanks for the details. Could you try again your cell_config.yml with the latest code?

closed 2024-02-13 by sauka