srsran/srsRAN_Project #464: Assertion `ue != nullptr' failed

improperaffo, 2024-02-09:
## Issue Description ##
When testing the project, we encountered multiple times a crash with the following error message:
```
/opt/srsRAN_Project/lib/ngap/ngap_impl.cpp:564: void srsran::srs_cu_cp::ngap_impl::handle_ue_context_release_command(const ue_context_release_cmd_s&): 
Assertion `ue != nullptr' failed - ue=1 ran_ue_id=7 amf_ue_id=14: UE for UE context doesn't exist.
Aborted (core dumped)
```

Different occurrence:
```
/opt/srsRAN_Project/lib/ngap/ngap_impl.cpp:564: void srsran::srs_cu_cp::ngap_impl::handle_ue_context_release_command(const ue_context_release_cmd_s&): 
Assertion `ue != nullptr' failed - ue=1 ran_ue_id=0 amf_ue_id=20: UE for UE context doesn't exist.
Aborted (core dumped)
```

## Setup Details ##
We have only encountered this problem when using a 60 MHz channel, other than that we could not identify any source of error.

## Steps to reproduce the problem ##
Run the gNB using the following configuration:
```
gnb_id: 531

slicing:
 - sst: 1
   sd: 0
 - sst: 1
   sd: 1

amf:
  addr: 10.43.81.162
  bind_addr: 10.42.2.7

ru_sdr:
  device_driver: bladerf
  device_args: "feature=oversample"
  # clock:
  # sync:
  srate: 69.12 #23.040
  otw_format: sc8
  tx_gain: 66
  rx_gain: 15

cell_cfg:
  dl_arfcn: 632628
  band: 78
  channel_bandwidth_MHz: 60
  common_scs: 30
  plmn: "00101"
  tac: 1
  pci: 1
  tdd_ul_dl_cfg:
    dl_ul_tx_period: 4
    nof_dl_slots: 2
    nof_dl_symbols: 7
    nof_ul_slots: 1
    nof_ul_symbols: 6
  nof_antennas_ul: 1
  nof_antennas_dl: 1
  pdsch:
    mcs_table: qam256
  pusch:
    mcs_table: qam256
    max_ue_mcs: 27
  pucch:
    sr_period_ms: 4

e2:
  enable_du_e2: false
  e2sm_kpm_enabled: true
  addr: 127.0.0.1
  bind_addr: 10.42.2.6
  port: 36421

log:
  filename: /tmp/gnb.log
  all_level: debug

pcap:
  mac_enable: false
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: false
  ngap_filename: /tmp/gnb_ngap.pcap
```

frankist, 2024-02-09:
Hey @improperaffo, thanks for the report. Can you send logs and pcaps?

improperaffo, 2024-02-12:
yes, here is the output on the console
```
/opt/srsRAN_Project/include/srsran/adt/circular_map.h:262: T& srsran::circular_map<K, T, N>::operator[](K) [with K = unsigned int; T = srsran::rlc_tx_am_sdu_info; long unsigned int N = 131072]: 
Assertion `contains(key)' failed - Accessing non-existent ID=41.
Aborted (core dumped)
```
And find attached the logs and pcaps. I had to extract the last 25MB of the logs and MAC pcaps because that is the limit that GitHub is allowing me to upload (the `.pcaps` are renamed `.txt` because pcap is not supported by GitHub).
[gnb_mac_tail.txt](https://github.com/srsran/srsRAN_Project/files/14236209/gnb_mac_tail.txt)
[gnb_ngap.txt](https://github.com/srsran/srsRAN_Project/files/14236211/gnb_ngap.txt)
[gnb_tail.log](https://github.com/srsran/srsRAN_Project/files/14236207/gnb_tail.log)

improperaffo, 2024-02-12:
Regarding the original issue, here is the console output:
```
/opt/srsRAN_Project/lib/ngap/ngap_impl.cpp:564: void srsran::srs_cu_cp::ngap_impl::handle_ue_context_release_command(const ue_context_release_cmd_s&): 
Assertion `ue != nullptr' failed - ue=1 ran_ue_id=23 amf_ue_id=164: UE for UE context doesn't exist.
Aborted (core dumped)
```
And find attached the logs and pcaps with the same setup as specified above.
[gnb_ngap.txt](https://github.com/srsran/srsRAN_Project/files/14236298/gnb_ngap.txt)
[gnb_tail.log](https://github.com/srsran/srsRAN_Project/files/14236299/gnb_tail.log)
[gnb_mac_tail.txt](https://github.com/srsran/srsRAN_Project/files/14236300/gnb_mac_tail.txt)

improperaffo, 2024-02-21:
Hi Fabian,
I could not trigger the crash with the current code.

FabianEckermann, 2024-02-22:
Thats good news! I'll close the issue then.

closed 2024-02-22 by FabianEckermann