srsran/srsRAN_Project #478: ul max_counter of rlf is incorrect(/lib/mac/mac_sched /rlf_detector.h)

lpg-delete, 2024-02-17:
## Issue Description ##
void handle_ack_common(du_ue_index_t ue_index, du_cell_index_t cell_index, bool ack, unsigned count_index) 
count_index : 0(DL) 1(UL)
135   const unsigned max_counter = count_index == 0 ? max_consecutive_kos[cell_index].max_consecutive_dl_kos
                                                    : max_consecutive_kos[cell_index].**max_consecutive_csi_dtx**;

## Expected Behavior ##
void handle_ack_common(du_ue_index_t ue_index, du_cell_index_t cell_index, bool ack, unsigned count_index)
count_index : 0(DL) 1(UL)
135   const unsigned max_counter = count_index == 0 ? max_consecutive_kos[cell_index].max_consecutive_dl_kos
                                                    : max_consecutive_kos[cell_index].**max_consecutive_ul_kos**;

carlo-gal, 2024-02-21:
Thank you for reporting this problem. We'll fix it and you will be able to see it in the next release

carlo-gal, 2024-02-21:
Closing this, as the issue will be taken care of.

closed 2024-02-21 by carlo-gal